MarlinFirmware/Marlin #12519: [Bugfix-2.0 ]Linadvance 1.5 and 1.0 and the different values of K

mr-miky, 2018-11-23:
It happened to me, by mistake, to print gcode from the uSD or octoprint that had been prepared for Marlin 1.1.x.
The printer seemed crazy, then analyzing I found that the value of K sent  was for linadvance 1.0.

I wonder if it is not the case to check the value of K received and if it is outside an acceptable range, report it as an error or warning.

thinkyhead, 2018-11-26:
`M900` already throws an error for out-of-range values, and won’t update the K value if it is outside the acceptable range, which is from 0 to 10.

mr-miky, 2018-11-26:
So, theoretically, the printer should not print it?

In any case, the gcode files were certainly 1.1.8 or even earlier.
I have never updated the fw to 1.1.9 because I'm dedicated to 2.0 ...

Tonight I try again and see if it reports specific errors / warnings ...

boelle, 2019-02-21:
@mr-miky still an issue?

boelle, 2019-03-13:
@thinkyhead thinkwe can close this one

closed 2019-03-13 by mr-miky