MarlinFirmware/Marlin #12521: STM32F4 Compile error with STM32Generic

bipsendk, 2018-11-23:
### Description

Trying to compile Marlin 2.0 with the recommended STM32Generic toolchain as described on the readme.md on https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin/src/HAL/HAL_STM32F4

Copying example config file from Marlin/src/config/examples/STM32F4

Running compile in Arduino IDE 1.9.0 returns:

```
C:\Users\xxx\AppData\Local\Temp\arduino_build_419999\sketch\src\HAL\HAL_STM32\watchdog_STM32.cpp:30:25: fatal error: IWatchdog.h: No such file or directory

   #include <IWatchdog.h>

                         ^
compilation terminated.

```

### Steps to Reproduce

1. Install Arduino IDE 1.9.0 beta
2. Download STM32Generic and install according to instructions in c:\users\xxx\Documents\Arduino\hardware\
3. Copy STM32F4 example config to over the standard example config in the same foilder as the ino file
4. Select "BLACK F407VG..." board
5. Select "Black F407VE (v2.0)" as specific board
6. Compile

**Expected behavior:** [What you expect to happen]

Code compile successfully

**Actual behavior:** [What actually happens]

Compile fails:

C:\Users\xxx\AppData\Local\Temp\arduino_build_419999\sketch\src\HAL\HAL_STM32\watchdog_STM32.cpp:30:25: fatal error: IWatchdog.h: No such file or directory

   #include <IWatchdog.h>

                         ^
compilation terminated.

#### Additional Information

mr-miky, 2018-11-23:
The real question is why it uses the HAL of the STM32 core instead of HAL_STM32F4 for the Generic ...

What BOARD did you "define"?

bipsendk, 2018-11-23:
From configuration.h ?

  #define MOTHERBOARD BOARD_STM32F4

bipsendk, 2018-11-23:
STM32Generic is a spinoff/variuant of STM32Core - unfortunately it looks like STM32Generic isn't updated as often as STM32Duino

bipsendk, 2018-11-23:
[image]

Selected board config

mr-miky, 2018-11-23:
I do not know what to tell you I'm using STM32 Generic with STM32F407 and compile without problems.
Probably missing some define and because of this it is used the code written for STM32Core (stm32duino).

bipsendk, 2018-11-23:
I removes the STM32Core stuff - but still get the issue with the wrong HAL...

Does the board selection look OK ?

mr-miky, 2018-11-23:
Check the log on the Arduino IDE and see that define (-D) does for compiling

bipsendk, 2018-11-23:
C:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DF_CPU=168000000L -mthumb -DSTM32GENERIC -DRAM_LENGTH=131072 -DFLASH_LENGTH=524288 -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -DSTM32F4 -DARDUINO=10900 -DARDUINO_BLACK_F407VE -DARDUINO_ARCH_STM32 -DSTM32F407VE -DHSE_VALUE=8000000 -DMENU_USB_SERIAL -DMENU_SERIAL_AUTO=SerialUSB "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino/stm32" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino/usb" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/CMSIS" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/CMSIS_Inc" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/CMSIS_Src" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/HAL_Inc" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/HAL_Src" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/stm32_chip" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\variants\\BLACK_F407VE" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\libraries\\SPI\\src" "C:\\Users\\BRIAN~1.IPS\\AppData\\Local\\Temp\\arduino_build_551349\\sketch\\src\\HAL\\HAL_STM32\\watchdog_STM32.cpp" -o nul

bipsendk, 2018-11-23:
Tried to edit boards.h - to add 
-DSTM32F4 
to compile parameter. Seems like the wrong HAL still is being selected.

"C:\\Users\\xxx\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DF_CPU=168000000L -mthumb -DSTM32GENERIC -DRAM_LENGTH=131072 -DFLASH_LENGTH=524288 -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -w -x c++ -E -CC -DSTM32F4 -DARDUINO=10900 -DARDUINO_BLACK_F407VE -DARDUINO_ARCH_STM32 -DSTM32F407VE -DHSE_VALUE=8000000 -DSTM32F4 -DMENU_USB_SERIAL -DMENU_SERIAL_AUTO=SerialUSB "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino/stm32" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino/usb" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/CMSIS" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/CMSIS_Inc" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/CMSIS_Src" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/HAL_Inc" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/HAL_Src" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\system/STM32F4/stm32_chip" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\cores\\arduino" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\variants\\BLACK_F407VE" "-IC:\\Users\\xxx\\Documents\\Arduino\\hardware\\STM32GENERIC-master\\STM32\\libraries\\SPI\\src" "C:\\Users\\BRIAN~1.IPS\\AppData\\Local\\Temp\\arduino_build_11173\\sketch\\src\\HAL\\HAL_STM32\\watchdog_STM32.cpp" -o nul

mr-miky, 2018-11-23:
mine: 

-DSTM32F4 -DARDUINO=10802 -DARDUINO_BLACK_F407ZG -DARDUINO_ARCH_STM32  -DSTM32F407ZG -DHSE_VALUE=8000000   -DMENU_SERIAL_AUTO=SerialUSB
-DARDUINO_ARCH_STM32 t

I do not have **-DARDUINO_ARCH_STM32** that almost certainly messes with

```
#elif defined (__ STM32F1__) || defined (TARGET_STM32F1)
   #define HAL_PLATFORM HAL_STM32F1
 >>>>>>> #elif defined (STM32GENERIC) &&defined (STM32F4)
 >>>>>>>>>  #define HAL_PLATFORM HAL_STM32F4**
#elif defined (STM32GENERIC) &&defined (STM32F7)
   #define HAL_PLATFORM HAL_STM32F7
>>>>>>>>>  #elif defined (ARDUINO_ARCH_STM32)
 >>>>>>>>>  #define HAL_PLATFORM HAL_STM32
#elif defined (ARDUINO_ARCH_ESP32)
   #define HAL_PLATFORM HAL_ESP32
#else
```

in platforms.h

First it defines HAL_PLATFORM HAL_STM32F4 and then override it
by defining HAL_PLATFORM HAL_STM32

mr-miky, 2018-11-23:
Ooppppsss ,

mr-miky, 2018-11-23:
-DARDUINO_ARCH_STM32 almost certainly messes with

#elif defined (__ STM32F1__) || defined (TARGET_STM32F1)
   #define HAL_PLATFORM HAL_STM32F1
#elif defined (STM32GENERIC) &&defined (STM32F4)
   #define HAL_PLATFORM HAL_STM32F4
#elif defined (STM32GENERIC) &&defined (STM32F7)
   #define HAL_PLATFORM HAL_STM32F7
#elif defined (ARDUINO_ARCH_STM32)
   #define HAL_PLATFORM HAL_STM32
#elif defined (ARDUINO_ARCH_ESP32)
   #define HAL_PLATFORM HAL_ESP32
#else

in platforms.h

First it defines HAL_PLATFORM HAL_STM32F4 and then override it
by defining HAL_PLATFORM HAL_STM32

try to comment out 
```
#elif defined(ARDUINO_ARCH_STM32)
  #define HAL_PLATFORM HAL_STM32
```

**Now the real question is why it works for me**

bipsendk, 2018-11-23:
Same result - it still picks the 
src\HAL\HAL_STM32\watchdog_STM32.cpp
file 

Also - the #ifdef / #elif - else if - so if HAL_STM32F4 was defined correctly, it should never consider checking for "if defined (ARDUINO_ARCH_STM32)" ...

Are you using IDE 1.9.0 beta ?

bipsendk, 2018-11-23:
Could you try to attach your boards.txt and platform.txt from the STM32 catalog? I would like to compare them...

mr-miky, 2018-11-23:
Stupid test, what happens if you move the folder HAL_STM32 out of project ?

bipsendk, 2018-11-23:
That gives:

```
In file included from C:\Users\BRIAN~1.IPS\AppData\Local\Temp\arduino_build_668081\sketch\src\HAL\HAL_STM32F4\HAL.cpp:30:0:

C:\Users\BRIAN~1.IPS\AppData\Local\Temp\arduino_build_668081\sketch\src\HAL\HAL_STM32F4\HAL.h:58:4: error: #error "Serial port 0 does not exist"

   #error "Serial port 0 does not exist"

```

mr-miky, 2018-11-23:
change it in configuration.h 

```
/**
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
//#define SERIAL_PORT -1
#define SERIAL_PORT 1

/**
```

bipsendk, 2018-11-23:
Will that put output on the USB port of the card ?

mr-miky, 2018-11-23:
It would be better to be able to compile everything then complicate it ;-P

You can use. I think you have to enable it when you choose the Black card etc etc

mr-miky, 2018-11-23:
Arduino IDE: menu tools/usb_interface , select Serial[virtual com port].

bipsendk, 2018-11-23:
In configuration.h:

#define SERIAL_PORT 1

And

In file included from C:\Users\xxx\AppData\Local\Temp\arduino_build_639443\sketch\src\HAL\HAL_STM32F4\HAL.cpp:30:0:

```
C:\Users\BRIAN~1.IPS\AppData\Local\Temp\arduino_build_639443\sketch\src\HAL\HAL_STM32F4\HAL.h:58:4: error: #error "Serial port 0 does not exist"

   #error "Serial port 0 does not exist"
    ^

```

And serial:

[image]

mr-miky, 2018-11-23:
Are you sure you have modified the right config.h, not the one between the examples?

bipsendk, 2018-11-23:
The file edited was the one in the same folder as Marlin.ino

mr-miky, 2018-11-23:
> in platforms.h
> 
> First it defines HAL_PLATFORM HAL_STM32F4 and then override it
> by defining HAL_PLATFORM HAL_STM32
> 
> try to comment out
> 
> ```
> #elif defined(ARDUINO_ARCH_STM32)
>   #define HAL_PLATFORM HAL_STM32
> ```
> 
> **Now the real question is why it works for me**

Wow,
it is logical that it works, the #elif following the one with the match are no longer considered.
Maybe I need to sleep a little ...

mr-miky, 2018-11-24:
Ok, 
I have compiled everything on another computer.
Arduino IDE 1.85, STM32Generic, BLACK407ZG demo.
The only things I had to do are:

Remove the servo library folder from the arduino suite.
Remove the HAL_STM32 from the sources.

For safety, if you edit files from outside the Arduino IDE, close and reopen the project

The build ends successfully

mr-miky, 2018-11-24:
the problem should be caused by

`#ifdef ARDUINO_ARCH_STM32`

that are in all the .cpp files in HAL_STM32

In both STM32Generic and STM32Core is  "true" so if you build using STM32Generic, the code for STM32Core is also build and then fails miserably.

The opposite should not happen because in HAL_STM32F4 is present

`#if defined (STM32GENERIC) && (defined (STM32F4) || defined (STM32F4xx))`

which is erroneous because STM32GENERIC does not define STM32F4xx but still imposes a condition defined(STM32GENERIC) that prevents to build the code with STM32 Core.

In HAL_STM32F7 it is even easier because it is only checked if STM32F7 is defined, which only STM32GENERIC does. STM32Core defines STM32F7xx and STM32F4xx not STM32F7 STM32F4

hasenbanck, 2018-11-24:
Oh, did we broke the old STM32F4 HAL by using "ARDUINO_ARCH_STM32" to check for the "official" STM32 core? Geez, I was under the impression that the STM32GNERIC only sets STM32GENERIC and not ARDUINO_ARCH_STM32.

You could try to fix the STM32F4 HAL so that it compiles again of course (and of course turn in a PR to fix the issue), but in the long run, the STM32F4 HAL will phase out in favor of the STM32 HAL. The only issue right now left for the "official" STM32 core is the missing VCOM/CDC support. There is a branch by me that fixes the issue temporarily, but in the next major release we will have official support for that feature (v1.5).

I would advise you to use the STM32 HAL instead and write your own pin definition for the F4 board you use since maybe the core already support your board as a variant inside the core: https://github.com/stm32duino/Arduino_Core_STM32

bipsendk, 2018-11-24:
Trying to compile using the official STM32 Core just gives a whole bunch of other errors. As far as I remember, the right board already exists in the STM32 Core, but there is probably some conflicts between the Marlin HALs and whatever is present in STM32 Core....

hasenbanck, 2018-11-24:
If you mean compiling the #define MOTHERBOARD BOARD_STM32F4 board definition with the "official" core, that would be to be expected, since it targets the STM32F4 HAL. You would need to update those pin definition and other parts for that board in Marlin for use the STM32 HAL and the "official" core. Or you create a brand new pin definition for your board.

bipsendk, 2018-11-24:
If the target of Marlin 2.0 is to be able to compile using the STM32 Core package for STM32 based packages, I might just wait until at least the basic compilation runs OK. Modifying an existing pins file is not the biggest challenge - introducing a new board type is a bit more complicated (for me, at least). I know how to handle the hardware, and modify the pins.h/configuration.h - but I am not a programmer, and my only experience with git is to clone stuff down to my own PC to compile it. Pull/merge operations (as I think some of them are called) are not something I have experience with :-)

mr-miky, 2018-11-24:
Unfortunately, the example configuration for STM32F4 in marlin, at present, is made with the pin definitions of stm32generic, PA0, PA1 etc etc ... therefore, in theory, it was easier to use STM32Generic to have something that at least turns on.

I do not understand why the developers of STM32 Core continue to ignore the need to have a generic board definition with the pins mapped 1 to 1 without that damned translation of arduino pins.

Personally I continue to use the GENERIC because, despite all the problems it has (and to understand certain it took me days to figure out what was wrong), I managed to make everything work but, above all, I do not want to define a board just for my pcb and redo all the definition of the pins since in all STM32 schematics is reported PA0 and not D0, A13, D44 ...

hasenbanck, 2018-11-24:
The "official" core allows the usage of generic GPIO port names (PA0 etc)
for quite some time now. So you are free to use those in your pin
definition file with the STM32 HAL. Have a look at pins_ARMED.h for example.

hasenbanck, 2018-11-24:
I don't think there is a plug-in for that yet. Recently I extended
stm32flash to be able to flash using SPI, which will be helpful for my next
board. Sooner or later I want to write an octoprint plug-in for stm32flash.

mr-miky, 2018-11-24:
I can get the STM32F4 into bootloader mode via software through a specific menu / command of marlin but then I do not know how to do the upload by octoprint since it lacks the specific plugin for the stm32 firmware. It should be simple but is need stm32flash for ARM
  I was hoping you did it ;-)

mr-miky, 2018-11-24:
I took a look at pins_ARMED.h and they use PA0 PA1 et etc.

This works because they wrote a variant.cpp / varriant.h that maps the PAx pins in PA_x, which in my view is ridiculous.

hasenbanck, 2018-11-24:
It's not ridiculous at all. At least not in the context of the STM32
arduino core. Variant definitions are needed for STM32 systems, since they
are highly flexible and need a lot of configuration.

You are free to write / propose a better solution and bring it upstream
into the core. Frederic (maintainer of the core) is always open for
improvements.

Phr3d13, 2018-11-24:
Just my two cents, but with all the board schematics and chip data sheets that I've looked at for stm32 variants, most (if not all) of them default to PA0, etc. You have to remember that, with these boards, you're stepping away from the generic avr structure to a generic stm32 structure.

mr-miky, 2018-11-24:
What I mean to say is that it should be the opposite, that is, map the arduino pins on the standard stm32 pin.

If I do not care to use the names of the arduino pins, I access directly without going through the mapping, PA0 is PA0 and D44 will be translated(slowly) through PinName digitalPin [];

boelle, 2019-02-21:
@bipsendk problem gone?

bipsendk, 2019-02-21:
@boelle - I will just check with VSCode and Arduino IDE to verify...

boelle, 2019-02-21:
thanks, we need to get the list of issue as small we can, so only real and active issues are open :-D

bipsendk, 2019-02-21:
@boelle  - Seems like my compile attempt in VScode still fails - not sure whether it is caused by issues in my configuration.h file....

In file included from Marlin\src\HAL\HAL_STM32F4\HAL.cpp:30:0:
Marlin\src\HAL\HAL_STM32F4\HAL.h:58:4: error: #error "Serial port 0 does not exist"
#error "Serial port 0 does not exist"

But in configuration.h I have serial port set to -1 ....

I will try with Arduino IDE 1.9.0 beta next...

[stm32f4-config.zip](https://github.com/MarlinFirmware/Marlin/files/2891201/stm32f4-config.zip)

boelle, 2019-02-21:
and you use bugfix 2.0? sorry i had to ask, a lot use old versions still

bipsendk, 2019-02-22:
@boelle Just tried in Arduino IDE 1.8.5 - with a git pull from February 21st.
I get a compile error with:

```
In file included from C:\Users\BRIAN\AppData\Local\Temp\arduino_build_912204\sketch\src\lcd\dogm\status_screen_DOGM.cpp:32:0:

C:\Users\BRIAN\AppData\Local\Temp\arduino_build_912204\sketch\src\lcd\dogm\dogm_Statusscreen.h:30:20: fatal error: binary.h: No such file or directory

 #include <binary.h>
                    ^
compilation terminated.

```

This is with MKS_MINI_12864 defined as display.

bipsendk, 2019-02-22:
With 

#define SERIAL_PORT 1

and no LCD defined, I get:

```
In file included from C:\Users\BRIAN\AppData\Local\Temp\arduino_build_912204\sketch\src\HAL\HAL_STM32F4\HAL.cpp:30:0:

C:\Users\BRIAN\AppData\Local\Temp\arduino_build_912204\sketch\src\HAL\HAL_STM32F4\HAL.h:58:4: error: #error "Serial port 0 does not exist"

   #error "Serial port 0 does not exist"

    ^

```

boelle, 2019-02-22:
same with ide 1.9 ?

bipsendk, 2019-02-22:
Same error with IDE 1.9 beta - serial port 0 does not exist.

Both IDE´s uses the same STM32Generic toolchain installed in %AppData%\Local\Arduino15

bipsendk, 2019-02-22:
I discovered I had multiple STM32-generic's floating around (in docuemnts folder, appdata and so on). I will just clean up and try from scratch...

boelle, 2019-02-22:
we have all been there :-D

boelle, 2019-02-22:
maybe look in to vs code and platform io ? it has helped me in the way that i dont need to think about having the right lib's etc

but of course its a new tool to learn

bipsendk, 2019-02-22:
Not having great success with VScode and PlatformIO,...

Here I also see:

In file included from Marlin\src\HAL\HAL_STM32F4\HAL.cpp:30:0:
Marlin\src\HAL\HAL_STM32F4\HAL.h:58:4: error: #error "Serial port 0 does not exist"
#error "Serial port 0 does not exist"
^~~~~

And this is with:

#define SERIAL_PORT -1

in configuration.h 

Build-log attached.

[stm32f4-vscode-buildlog.txt](https://github.com/MarlinFirmware/Marlin/files/2893719/stm32f4-vscode-buildlog.txt)

boelle, 2019-02-22:
@thinkyhead you are the almighty and fearless code god, does it tell you anything?

AnHardt, 2019-02-22:
Are you sure Marlin is still ought to be compiled with STM32Generic?
I mean to remember on rumor this was switched to the `official` core. But i'm quite unsure about that.

mr-miky, 2019-02-23:
IMHO, platformio loses some define -D on the command line at compile time.
If I remember well there is something to be defined to enable the serial but I should investigate. For sure with STM32Generic I compile marlin 2.0x but not with platformIO. I currently use the serial 1 connected to the usual FT232 converter and it works ...

mr-miky, 2019-02-23:
`"/home/michele/Arduino/gcc-arm-none-eabi-7-2018-q2-update/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DF_CPU=168000000L -mthumb -DSTM32GENERIC -DRAM_LENGTH=131072 -DFLASH_LENGTH=1232896 -c -g -MMD -O1 -Wall -Wextra -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -Dprintf=iprintf -MMD -DSTM32F4 -DARDUINO=10802 -DARDUINO_BLACK_F407ZG -DARDUINO_ARCH_STM32  -DSTM32F407ZG -DHSE_VALUE=8000000   -DMENU_SERIAL_AUTO=SerialUSB  "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/cores/arduino/stm32" "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/cores/arduino/usb" "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/system/CMSIS" "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/system/STM32F4/CMSIS_Inc" "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/system/STM32F4/CMSIS_Src" "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/system/STM32F4/HAL_Inc" "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/system/STM32F4/HAL_Src"  "-I/home/michele/Arduino/hardware/STM32GENERIC/STM32/system/STM32F4/stm32_chip"   -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/cores/arduino" -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/variants/BLACK_F407ZG" -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/libraries/Servo/src" -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/libraries/SPI/src" -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/libraries/stm32_dma/src" -I"/home/michele/Arduino/libraries/U8glib/src" -I"/home/michele/Arduino/libraries/TMCStepper/src" -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/libraries/Wire/src" -I"/home/michele/Arduino/hardware/STM32GENERIC/STM32/libraries/SDIO/src" -MMD -MP -MF"core/core/usb/msc/usbd_msc_scsi.c.d" -MT"core/core/usb/msc/usbd_msc_scsi.c.o" -D__IN_ECLIPSE__=1 "/home/michele/Arduino/hardware/STM32GENERIC/STM32/cores/arduino/usb/msc/usbd_msc_scsi.c"  -o  "core/core/usb/msc/usbd_msc_scsi.c.o"  -Wno-expansion-to-defined -DU8G_HAL_LINKS -DBOOTLOADER_MENU -DTARGET_STM32F4 -DUSB_OTG_POWER
`
@bipsendk  -DMENU_SERIAL_AUTO=SerialUSB    exists on your configuration ??  

It should be platformio that adds it based on your selections ...
Check if you can see the complete command that is given to compile ..

mr-miky, 2019-02-23:
@bipsendk 
Ha ha ha...   "il diavolo fa le pentole ma non i coperchi" ( the devil makes the pots but not the lids ). 
I found out why I do not have that damn mistake.
in    `src/HAL/HAL_STM32F4/HAL.h`

`//Serial override`
`//extern HalSerial usb_serial;`
`#if SERIAL_PORT == 0`
`//  #error "Serial port 0 does not exist"`
`#endif`

`#if !WITHIN(SERIAL_PORT, -1, 6)`
`  #error "SERIAL_PORT must be from -1 to 6"`
`#endif`

if I uncomment it, the build fails with the same error messages
If you do not want to melt the fuse, substitute it with a nail ;-)

mr-miky, 2019-02-23:
Very interesting,

SERIAL_PORT is clearly 1 and error is still issued ...

[image]

mr-miky, 2019-02-23:
Is it not the case that the control on the serials goes to sanitycheck.h rather than here since the header is also included in other parts such as spi and timers?

mr-miky, 2019-02-23:
May be an acceptable fix for HAL.h ??

`#ifdef SERIAL_PORT`
`..#if SERIAL_PORT == 0`
`....#error "Serial port 0 does not exist"`
`..#endif`
`..#if !WITHIN(SERIAL_PORT, -1, 6)`
`....#error "SERIAL_PORT must be from -1 to 6"`
`..#endif`
`#endif`

bipsendk, 2019-02-23:
@mr-miky - The settings in my Arduino IDE regarding the USB part is shown in the attached image.

Seems like the extra define comes from boards.txt ....

[image]

From boards.txt:

BLACK_F407XX.menu.upload_method.STLinkMethod.build.extra_flags_serial_auto=-DMENU_SERIAL_AUTO=SerialUSB

[12 more replies]

closed 2019-03-13 by thinkyhead