MarlinFirmware/Marlin #12526: [Bugfix 2.0] RADDS Board with I2C OLED display

kwalters3, 2018-11-24:
### Description

#elif used for checking for an I2C display will never return true because SSD1306_OLED_I2C_CONTROLLER is never define.  This does not effect output to the display I am currently using which uses U8GLIB_SH1106, but my encoder is not enabled.  Encoder is connected to the BTN_EN1, BTN_EN2, and BTN_ENC

### Steps to Reproduce

1. Configure Marlin to use SH1109 controlled OLED
2. Compile and upload
3. start printer

**Expected behavior:** Encoder and encoder button should go to menu and select menu options

**Actual behavior:** Encoder does nothing

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

Ludy87, 2018-11-24:
edit this file `Marlin\src\inc\Conditionals_LCD.h`

change
`#define HAS_ENCODER_WHEEL   (!HAS_ADC_BUTTONS && ENABLED(NEWPANEL))`
to
`#define HAS_ENCODER_WHEEL   (!HAS_ADC_BUTTONS && ENABLED(NEWPANEL) && !HAS_DIGITAL_BUTTONS)`

and try again

kwalters3, 2018-11-24:
I think that I was a little unclear about the type of display being used.
It is just the i2c OLED and the encoder is a seperate unit.  The
conditional that I stated in the post is checked in the pins_RADDS.h file
and controls defining the pins to be used for BTN_ENC1, BTN_ENC2, BTN_ENC,
BEEP, LCD_SDSS, and SD_DETECT_PIN.

The issue is that the pins are not being assigned since the #elif
statement, line 250, will never be true because SSD1306_OLED_I2C_CONTROLLER
isn't being defined anywhere.

I have fixed this in my pins_RADDS.h as well as changing some of the pin
assignments to accommodate my not wanting to move some of my wiring coming
from Repetier Firmware.

The change that I made was to:
#elif ENABLED(U8GLIB_SSD1306) || ENABLED(U8GLIB_SSD1309) ||
ENABLED(U8GLIB_SH1106)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Ludy87, 2018-11-24:
that's what happens when you can not with posts the configuration files. ;-)

In theory, this must be OLED_PANEL_TINYBOY2

boelle, 2019-02-21:
@kwalters3 problem solved?

kwalters3, 2019-02-21:
The responses from Ludy87 does nothing to address the issue that I was reporting.  The problem was not in the LCD conditionals file, but in the file pins_RADDS.h

The statement #elif ENABLED(SSD1306_OLED_I2C_CONTROLLER) at line 250 of the file pins_RADDS.h is the issue that I was trying to report.  Because of this conditional not ever being defined the RADDS board will not intialize pin definitions for the rotary encoder.

boelle, 2019-02-21:
submit a PR to bugfix 2.0 with the correct file

kwalters3, 2019-02-21:
I'm very new to git and don't know how to do that.

boelle, 2019-02-21:
google is your friend :-D

but you can click the green close button, dev people will find you anyway

closed 2019-02-21 by kwalters3