MarlinFirmware/Marlin #12528: [Bugfix 2.0] BL Touch - intermittent malfunction - resolved

mmx01, 2018-11-24:
Hi All,

I have resolved my issue but maybe relevant to let others know and have it somewhat addressed in the next releases of Marlin (Servo 0 pin default assignment). This is specific to MKS SBASE 1.3 (pin out) and a situation when one would like to use a BL Touch Z probe.

According to information on the web you wire servo part to P1.23 (Servo 0) +5V/GND and then Black/White to Z_MIN. Basic Marlin setup with probe offset and activation, delay set to 100 etc.
 
#define Z_SAFE_HOMING
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

#define Z_PROBE_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {10,90}  // Z Servo Deploy and Stow angles

#define BLTOUCH
#if ENABLED(BLTOUCH)
  #define BLTOUCH_DELAY 100   // (ms) Enable and increase if needed
#endif

#define PROBING_HEATERS_OFF       // Turn heaters off when probing
#if ENABLED(PROBING_HEATERS_OFF)
  #define WAIT_FOR_BED_HEATER     // Wait for bed to heat back up between probes (to improve accuracy)
#endif

#define PROBING_FANS_OFF       
#define DELAY_BEFORE_PROBING 100  // (ms) To prevent vibrations from triggering piezo sensors

#define X_PROBE_OFFSET_FROM_EXTRUDER 52  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER -17 // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -2.2   // Z offset: -below +above  [the nozzle]

My issue was the probe seemed to be working okay but randomly failing to deploy causing bed contact with the extruder. Sometimes it also paused or deployed into bed without any reason.

It became evident upon UBL activation G29 P1 when multiple samples are taken. Sometimes mesh was completed with 3 sometimes with 6 probing errors reported. When this happened bed was moving towards extruder but the probe was not deploying causing bed to stop and on the next probing attempt everything was okay in the same spot.

I rewired everything related to probe with shielded cables, provided an external 5V 3A dedicated PS, checked again all contacts (assuming jitter or EMI issues) - no improvement. 3 days later pulling hair out I decided to change servo 0 pin and use servo 1 pin for servo 0 instead, moved the cable on J8 connector to 4. 

//#define SERVO0_PIN         P1_23   // J8-3 (low jitter)
#define SERVO0_PIN         P2_12
//#define SERVO1_PIN         P2_12   // J8-4

This solved the issue once for all. I doubt this is marlin issue, more of the board wiring. I was not able to find anything colliding with original servo 0 pin but none the less it is working now.  If you then have intermittent issues with BL Touch on MKS SBASE do this first...#

So if you follow wiring diagrams on the web they all instruct you to use 1.23. I would suggest maybe to have this changed in marlin and have default servo 0 on 2.12 with a commentary to wire yellow cable there.

Br,
M

p3p, 2018-11-24:
This was a framework issue (not Marlin) but it was patched (worked around until a real fix) about 5 days ago, make sure you have PlatformIO frameworks and libraries up to date (framework-arduino-lpc176x should be version 0.0.6), and try with P1_23 again it should be fixed. 

If not then the work around wasn't enough and I need to look harder.

mmx01, 2018-11-25:
I was pulling the marlin 2.0 branch in the recent days but not much looking into platformio part indeed. Probably two weeks old since the last download. Will revert servo pin 0 to 1.23 and come back tomorrow with updates. After platformio update I see
nxplpc lpc176x 0.0.1
framework arduino lpc176x 0.0.6
toolchain 1.70201.0

Out of pure curiousity what is / was causing this issue? When I wanted to have a separate fan for the controller - USE_CONTROLLER_FAN and yet again played with J8 connector the board was freezing randomly. It only worked with pin 4.28, other pins made it not bootable on occassion with a FET gate wire connected. The minute I removed the wire the board was acting normal.

Br
M

p3p, 2018-11-25:
> It only worked with pin 4.28, other pins made it not bootable on occassion with a FET gate wire connected. 

The PWM shouldn't be able to stall the board, the mcu has no idea if the pin is actually connected to anything, if it was only when the gate was connected it could have been a power draw issue.. I'm unsure. The known issue is with the hardware PWM being updated multiple times in the same PWM period, so I've disabled it for now so only Software PWM is available, by moving to a secondary Servo port you fixed it my moving off a Hardware PWM pin onto one that only supports Software PWM.

p3p, 2018-11-25:
Although pin 2.12 would be an issue if it was high during boot, it puts the MCU into the bootloader rather than start Marlin.

mmx01, 2018-11-25:
It was not my intention to get much off topic from BL Touch but with 2.12 pin it seems a likely scenario with the bootloader. All leds are on and nothing happens when FET gate (IRF520) is connected during boot. When I disconnect it and plug it back after the booting sequence things are okay until the next power cycle. On pin 4.28 things are normal all the time.

Power wise I feed the printer from 3 PSes - 2x 12V 30A in series with two sets of terminals one for 12v serving the board and 24v for the bed (via 30A FET) + 1x 5V 3A for all external sensors and LCDs connected via common GND to the board. This is saving some strain on poor onboard 5V VR and also 12V rail. Good quality thick cables for higher loads. 

M

mmx01, 2018-11-25:
So having updated the libraries and servo 0 pin back to 1.23 etc. looks like the issue is resolved. G29 P1 returned 0 probing errors on the original pin and no other anomalies were observed.

Interesting one would expect a hardware PWM to be much more reliable and less prone to issues like frequency/width stability not depending on software stack and multitasking (or lack of) capabilities of an MCU.

Thanks for the tip - I push it to closed (for now until no workaround is required 👍 )

closed 2018-11-25 by mmx01