MarlinFirmware/Marlin #12531: TMC2208 with weird noise when near the hybrid threshold value

italocjs, 2018-11-24:
Steps to reproduce:
Set the hybrid threshold to 60

>Move the head in a speed of 50 super silent
>Move the head in a speed of 59  (still in stealthchop) but shows some weird noises (kinda looks like it is changing from stealthchop and spreadcyle many times in the same move)
>Move the head in a speed of 70 loud, but normal for spreadcycle

Marlin version:  1.1.9
Drivers: TMC2208

italocjs, 2018-11-24:
Also, not sure if its a problem with my driver, but the spreadcycle is way louder than any a4988!

pipersw, 2018-12-04:
I have both of them, and no noise @24V. 12V is to lower for these drivers.

boelle, 2019-02-21:
@italocjs problem gone?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

italocjs, 2019-03-06:
Hi boelle, sorry i didnt reply, the issue is not solved at least for the last time i updated the firmware (on january this year), it stopped bothering me because i changed the driver to run on stealthchop all the time.  You can close this, but im not sure if it was fixed

boelle, 2019-03-06:
@italocjs have you tried with new copy of bugfix 2.0?

boelle, 2019-03-06:
i mean tried with 24V, trinamic do need above 18V to become noiseless

italocjs, 2019-03-07:
Not with the latest version, i tried with mid-january latest bugfix on 2.0

Ill try testing with a 24v psu and post the result here.   If you want, you can close the issue

boelle, 2019-03-07:
only you and @thinkyhead can close

there should be a close button  below but only you can see it

boelle, 2019-03-13:
@italocjs please click close button below

closed 2019-03-13 by italocjs