MarlinFirmware/Marlin #12533: "Probing failed" error after tuning min. Z travel

Ivomola, 2018-11-24:
Hi. Today I upgraded my Tevo Tarantula to marlin 1.1.9 in the MKS Base Gen 1.3, and after a few errors here and there I got it working, but when I got to print, even if I put the Z offset of the probe, the nozzle was too far from the bed, so after a bit of research I found a line (can't put it now because I don't have my computer at the moment) that specifies the minimum travel of any axis (obviously there were all set to 0 to match the endstop), so I changed z to a number close to the offset I use, -3 mm to be precise, and it finally lowered z more than 0, but now when I try to do the bed leveling procedure, it raises z a bit, it goes to the first point, it lowers z a bit and it gives me the "probing failed" error because the probe doesn't even trigger, and the thing is that it doesn't trigger because it's coincidentally 3 mm above of what it should be, and I can't find a way to solve it, any ideas? Thanks.

thinkyhead, 2018-11-26:
Does `M502`, `M500` help?

The issue template asked you to supply your configuration files. Where are they?

thinkyhead, 2018-11-26:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

Ivomola, 2018-11-26:
Hi. I reported it not as an user question but as a possible bug. Nevertheless, I managed to solve it myself undoing some changes, I will investigate if some of them are glitched or not, but for now the problem is solved, I will comment here if I find something useful.

closed 2018-11-26 by Ivomola