MarlinFirmware/Marlin #12534: [Bugfix 2.0] M106 ignored for Tools >0

akurz42, 2018-11-25:
### Description

In the latest Bugfix 2.0 versions the commands controlling the part cooling fans are ignored when a tool other than 0 is set. This used to work before (I have a version from 9th September that works fine),

### Steps to Reproduce

1. Connect to printer. Default Tool 0 is set.
2. M106 S255
3. M107
4. T1
5. M106 S255
6. M107
7. T0
8. M106 S255
9. M107

**Expected behavior:** [What you expect to happen]
After M106 S255 the part cooling fan ist started at full speed, no matter which tool is set.

**Actual behavior:** [What actually happens]
The fan only works when Tool 0 is set.
Steps 1-4 are ok. Steps 5-6 show no effect at all. Steps 7 to 9 work fine again. All GCODEs are confirmed with "ok" in the terminal.

#### Additional Information

I am using a RADDS board and a THC-01 (3 materials) with a single hotend (SINGLENOZZLE is set). Settings in recent versions have always been the same as with the version from 9th September. The older version works, recent versions don't.

I've tried to address the fan directly (M106 P0 S255), used all kinds of combinations with FANMUXx_PIN to no avail.

(https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

InsanityAutomation, 2018-11-26:
Give this a shot and see if it does what youre looking for

https://github.com/InsanityAutomation/Marlin/tree/SingleNozzleM106Tweak

I modified multi extruder to default to active_extruder if P is not set, then set single nozzle to always set p to 0 before writing the fan speed, since the current command was still using the P value for the fan output setting. Also moved extra fan speeds above a branch which could return, causing it to never update a value.

akurz42, 2018-11-26:
Ah, bad timing - I've just started a 13 hour print job, but I'll test it first thing tomorrow morning. Thanks for the effort!

InsanityAutomation, 2018-11-26:
No problem. While youre at it, try the new single nozzle retract / purge and park at tool change with differing temps :)

akurz42, 2018-11-26:
Sounds cool! On my testing list now :-)

InsanityAutomation, 2018-11-26:
Adding these as well as storage for alternative extruder temps and fan speeds made the change in behavior you referenced, so at least there will be a positive reason (and no more tool change scripts in the slicer!)

akurz42, 2018-11-26:
Bad news, then good news, then bad news again:
My print failed relatively early (bad), so I could test the tweak right earlier than expected (good). Unfortunately the behaviour is exactly the same as before (bad):

T0
M106 S255 - works

T1
M106 S255 - doesn't

T1
M106 P1 S255 - doesn't work either.

Pity :(

InsanityAutomation, 2018-11-26:
And you used the branch I linked, not current head right?

akurz42, 2018-11-26:
Absolutely. "Marlin-SingleNozzleM106Tweak.zip" downloaded from "InsanityAutomation/Marlin - forked from MarlinFirmware/Marlin"

thinkyhead, 2018-11-28:
Please ZIP your `Configuration.h` and `Configuration_adv.h` files so that we can check them over.

akurz42, 2018-12-16:
Hi,

just checking in to see if the requested configs helped in any way?

Alex

InsanityAutomation, 2018-12-16:
Holidays make for chaos. I'm struggling on automotive shutdown work and will hopefully be back to this soon. I have a creality crx set aside to test with.

akurz42, 2018-12-16:
Hey, you do this for free in your free time, so please don't feel pushed. Enjoy the holidays and have a look next year if you want. No pressure. I just wanted to make sure the thread isn't forgotten.

For now the old, working version is fine.

Thanks!
Alex

InsanityAutomation, 2018-12-16:
I'd much rather be working on this than the FANUC robot over profisafe on an s7 plc that's been consuming my time! You know the OEM spec is overly complicated when the Rockwell guys stop saying should've used AB and start saying they're glad it's you not them.... Lol

akurz42, 2018-12-17:
Had to google most of what you've just said, but I sure want one of these robots :)
Maybe create a Marlin fork to control them? ;)

InsanityAutomation, 2018-12-31:
@akurz42 please try Pull #12777 and confirm

closed 2019-02-04 by thinkyhead

akurz42, 2019-03-27:
Sorry - got sidetracked by life...

Tested now and found all working. Big thanks and sorry for the late reply.

Alex