MarlinFirmware/Marlin #12536: boards.h documentation missing

Louis-Cypher, 2018-11-25:
### Description

I want to contribute a pin file for BIQU Ramps 1.6 plus. Reading boards.h I am already lost. To my understanding boards.h is undocumented, I find a 

`define BOARD_WHATEVER  123` and I don't have the roughest ideas what happens in the chain from Configuration.h boards.h through pins.h to the final PIN file. The MOTHERBOARD variable in Configurations.h has no `BOARD_` infront (which very likely is stripped off somewhere in the code) and I do not have the slightest idea what the number after is and what it is good for. The steps given in the devolpment documentation are incomplete as well. This keeps newbies like me from contributing pin files.
### Steps to Reproduce

read the code of boards.h

**Expected behavior:** [What you expect to happen]
At least one line of explanation what happens in the board definition. What is the convention for `BOARD_xxx` and what is the number (I guess an ID? How fo find a unique one? What are the conventions to choose one?

**Actual behavior:** [What actually happens]
Undocumented code

Louis-Cypher, 2018-11-25:
The BOARD_ is stripped in the last line of boards.h (undocumented) but what happens to the number/ID still stays a miracle to me

FanDjango, 2018-11-25:
I've got a nice pin file too for Anet A8 modded with a Geetech GT256 Board, but don't really know what to do with it...

Ludy87, 2018-11-25:
i read this files without documentation tags and understand this function and implementation. I am not a Freak. The code is very simple.

The Board number be random, respectively +1 if it a same manufactor a new board created

Louis-Cypher, 2018-11-25:
That's good for you but not very helpful. The question is about undocumented codemaybe preventing people from contributing to Marlin. What is the random number good for? If it is good for nothing, why is it there? Should it match something? Since random is pretty likely to collide we will end up with numbers of 20 digits soon if there is no rule how to generate numbers.   

Furthermore, the documentation of how to add a board is not helping much regardless whether you managed to solve the problem or not. If there is documentation it should be of use in my eyes.

Ludy87, 2018-11-25:
the number defination is for compare

example:

`#define MOTHERBOARD BOARD_RAMPS_14_EFB`

```c++
boards.h

BOARD_RAMPS_14_EFB equal 43
```

```c++
pins.h

#define MB(board) (defined(BOARD_ ## board) && MOTHERBOARD==BOARD_ ## board)

MB(RAMPS_14_EFB)

```
just shorter like

`defined(BOARD_RAMPS_14_EFB)  && MOTHERBOARD==BOARD_RAMPS_14_EFB)`

Louis-Cypher, 2018-11-25:
Great, Thanks a lot for clarifying. I agree on documentation in source code being better than external documents. From my point of vie boards.h could be documented better. Maybe I find time to add some comments (as soon as I fully understood what's happening there; not a C programmer at all.)

AnHardt, 2018-11-25:
The C PreProcessor (CPP) can't compare strings - just integer numbers. `boards.h` gives numbers a human memorisable names - or if you want, the other way around.

AnHardt, 2018-11-25:
```cpp
// The following define selects which electronics board you have.
// Please choose the name from boards.h that matches your setup
#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_RAMPS_14_EFB
#endif
``` 
If you want you can fill in the number (43) here instead of the boardname.
If `MOTHERBOARD` is already defined here the value comes from the MAKEFILE where the compiler is called with a that -D parameter. 
Later on this number is just and only used to pick the right pin file

thinkyhead, 2018-11-26:
We can always use help with the MarlinFirmware/MarlinDocumentation project to expand the website documentation on various subjects. We haven’t written up a very lengthy development guide beyond coding conventions and making PRs, so that is one area where contributions would be helpful. I’m aiming to make more videos and tutorials on Marlin features and on general Marlin development. I’ll plan to make a short video on the subject of adding new boards, new features, etc., as video production gets going.

thinkyhead, 2019-01-23:
请把它做得更好！

closed 2019-01-23 by thinkyhead