MarlinFirmware/Marlin #12537: Ender-3 Pro updated _Statusscreen.h

Lord-Quake, 2018-11-25:
Here is a very mynute contribution from myself to the Marlin community.

Having an Ender-3 Pro I'd like to have the name shown in the status display.
After having stumbled on a YT video posted by Scott in January '18 I finally figured out how to make the changes.

[image]

I created a folder `..\example_configurations\Creality\Ender-3 Pro`  and copied the files from the Ender-3 folder. You will then need to swap the file _Statusscreen.h with this one:

[_Statusscreen.zip](https://github.com/MarlinFirmware/Marlin/files/2612952/_Statusscreen.zip)

Maybe it's worth adding... ?

Ludy87, 2018-11-25:
this isn't for Marlin 2.0

thinkyhead, 2018-11-26:
Thanks! I’ll adapt your bitmap for Marlin 2.0 and stick it in the folder with the Ender 3 configuration, which already has the Creality version of the Ender logo. Is there any difference between the Ender 3 and the Ender 3 Pro that would require us to also have a separate `Configuration.h` file for it?

thinkyhead, 2018-11-26:
Ok, thanks! I’ll just put the extra status logo into a sub-folder, then. With the new scheme, the logo part is all that’s needed. Everything else comes from our core bitmaps file now.

boelle, 2019-02-21:
@Lord-Quake problem solved?

Lord-Quake, 2019-02-21:
Yes sir :-)

closed 2019-02-21 by Lord-Quake

ETE-Design, 2019-09-20:
Where to find the Ender-3 Pro Status Screen? Can only find the one for the Standard Ender-3

J0hnMatrix, 2019-10-02:
Would like to have the Ender-3 Pro status screen too :)

xtianpaiva, 2019-11-22:
Here @Lord-Quake 's `Ender-e Pro` icon ported to Marlin 2.0.x https://raw.githubusercontent.com/xtianpaiva/Marlin/ender-3-pro/Marlin/_Statusscreen.h