MarlinFirmware/Marlin #12542: [2.0.x] Four lines on RepRapDiscount Full Graphics Smart Controller vs five lines in 1.1.9

marcio-ao, 2018-11-26:
When I flash 1.1.9 on my printer, the RepRapDiscount Full Graphics Smart Controller shows five lines of text on the LCD menus. On 2.0.x, only four lines show up. There is space for a fifth line, but it is always blank and unused.

What changed?

gloomyandy, 2018-11-26:
Five lines on mine built from 2.0.x a few days ago. Perhaps you should post a picture?

InsanityAutomation, 2018-11-26:
Seeing the same thing here with 4 lines on a build. Its also chopped to 14chars cutting off some menu text. Need to play with config options tonight and find what messes with it.

marcio-ao, 2018-11-26:
@InsanityAutomation: Are you using `LIGHTWEIGHT_UI`? Turning that feature on seems to be what is limiting the LCD display to four lines for me.

marcio-ao, 2018-11-26:
Here is the problem. In `Conditionals_post.h` there is the following code:

```
#ifndef LCD_WIDTH
  #if ENABLED(LIGHTWEIGHT_UI)
    #define LCD_WIDTH 16
  #elif HAS_GRAPHICAL_LCD
    #define LCD_WIDTH 22
  #elif ENABLED(ULTIPANEL)
    #define LCD_WIDTH 20
  #elif HAS_SPI_LCD
    #define LCD_WIDTH 16
  #endif
#endif
#ifndef LCD_HEIGHT
  #if ENABLED(LIGHTWEIGHT_UI)
    #define LCD_HEIGHT 4
  #elif HAS_GRAPHICAL_LCD
    #define LCD_HEIGHT 5
  #elif ENABLED(ULTIPANEL)
    #define LCD_HEIGHT 4
  #elif HAS_SPI_LCD
    #define LCD_HEIGHT 2
  #endif
#endif
```

I believe this was done as a misunderstanding of how the `LIGHTWEIGHT_UI` operates. When in the lightweight status screen, the text mode of the display is used, which indeed limits the display to 16x4 characters. However, when in the menu, the old U8G draw code is still used, so the limits should be the same as when `LIGHTWEIGHT_UI` is disabled.

I think the simplest fix is to remove the `#if ENABLED(LIGHTWEIGHT_UI)` from `Conditionals_post.h`, those should not be there.

marcio-ao, 2018-11-26:
@thinkyhead: It looks like it was you who added this code in commit a3923424c7, do you agree that removing that if statement is the best way forward?

silentninja1, 2018-11-26:
@marcio-ao We have it as a toggle option, but yeah I believe this was on a machine with it. Looks like you nailed it before I even found time to sit back down!