MarlinFirmware/Marlin #12545: How can I set the coordinates in Marlin for homing the printhead 0,0,0 in stead of center to the bed ?

garaguido, 2018-11-27:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AnHardt, 2018-11-27:
Experiment with:
```
// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Move the Z probe (or nozzle) to a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
//
//#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif
```

garaguido, 2018-11-28:
Sorry , my mistake  , I understand

garaguido, 2018-11-28:
Is working fine now .

Thanks

garaguido, 2018-12-23:
Hi 

Sorry , perhaps not the wright Forum .

I have posted a question on the marlin Forum but there is no answer for my problem.

Perhaps you can help me ?

I have a magnetique printbed , with remouvable surface and the printhead is to close to the bed after HOME or G28 the themp. from printbed can max bear 150 gr C.

What I wound is that the print head at least 5mm from the bed away is  ,  X0 Y0 and Z5

Where and how can I configure this in Marlin ?

Regards

Guido

From: AnHardt 
Sent: Tuesday, November 27, 2018 10:13 PM
To: MarlinFirmware/Marlin 
Cc: garaguido ; Author 
Subject: Re: [MarlinFirmware/Marlin] How can I set the coordinates in Marlin for homing the printhead 0,0,0 in stead of center to the bed ? (#12545)

Experiment with:

// Use "Z Safe Homing" to avoid homing with a Z probe outside the bed area.
//
// With this feature enabled:
//
// - Allow Z homing only after X and Y homing AND stepper drivers still enabled.
// - If stepper drivers time out, it will need X and Y homing again before Z homing.
// - Move the Z probe (or nozzle) to a defined XY point before Z Homing when homing all axes (G28).
// - Prevent Z homing when the Z probe is outside bed area.
//
//#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axes (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axes (G28).
#endif

Bob-the-Kuhn, 2018-12-23:
Use the Z_PROBE_END_SCRIPT option in configuration.h
`#define Z_PROBE_END_SCRIPT "G0 X0 Y0 Z5"`

garaguido, 2018-12-24:
Hi

At last a solution , thank you verry much !!!!

A Verry Happy Christmas to you and all my wishes for 2019

Regards

Guido

From: Bob Kuhn 
Sent: Sunday, December 23, 2018 10:16 PM
To: MarlinFirmware/Marlin 
Cc: garaguido ; Author 
Subject: Re: [MarlinFirmware/Marlin] How can I set the coordinates in Marlin for homing the printhead 0,0,0 in stead of center to the bed ? (#12545)

Use the Z_PROBE_END_SCRIPT option in configuration.h
#define Z_PROBE_END_SCRIPT "G0 X0 Y0 Z5"

garaguido, 2018-12-24:
HI

I have tried your settings but NO 5mm from the bed .

I use the Mesh bed leveling

Here are my settings for it

//===========================================================================
  //=================================== Mesh ==================================
  //===========================================================================

  #define MESH_INSET 30          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 3    // Don't use more than 7 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X

  //#define MESH_G28_REST_ORIGIN // After homing all axes ('G28' or 'G28 XYZ') rest Z at Z_MIN_POS

#endif // BED_LEVELING

/**
* Points to probe for all 3-point Leveling procedures.
* Override if the automatically selected points are inadequate.
*/
#if ENABLED(AUTO_BED_LEVELING_3POINT) || ENABLED(AUTO_BED_LEVELING_UBL)
  //#define PROBE_PT_1_X 15
  //#define PROBE_PT_1_Y 180
  //#define PROBE_PT_2_X 15
  //#define PROBE_PT_2_Y 20
  //#define PROBE_PT_3_X 170
  //#define PROBE_PT_3_Y 20
#endif

/**
* Add a bed leveling sub-menu for ABL or MBL.
* Include a guided procedure if manual probing is enabled.
*/
#define LCD_BED_LEVELING

#if ENABLED(LCD_BED_LEVELING)
  #define MBL_Z_STEP 0.025    // Step size while manually probing Z axis.
  #define LCD_PROBE_Z_RANGE 4 // Z Range centered on Z_MIN_POS for LCD Z adjustment
#endif

// Add a menu item to move between bed corners for manual bed adjustment
#define LEVEL_BED_CORNERS

#if ENABLED(LEVEL_BED_CORNERS)
  #define LEVEL_CORNERS_INSET 30    // (mm) An inset for corner leveling
  #define LEVEL_CENTER_TOO        // Move to the center after the last corner
#endif

/**
* Commands to execute at the end of G29 probing.
* Useful to retract or move the Z probe out of the way.
*/

//#define Z_PROBE_END_SCRIPT "G1 Z10 F12000\nG1 X15 Y330\nG1 Z0.5\nG1 Z10"

#define Z_PROBE_END_SCRIPT "G0 X0 Y0 Z5 " // Z  5 mm from bed with G28 and HOME

// @section homing

From: Bob Kuhn 
Sent: Sunday, December 23, 2018 10:16 PM
To: MarlinFirmware/Marlin 
Cc: garaguido ; Author 
Subject: Re: [MarlinFirmware/Marlin] How can I set the coordinates in Marlin for homing the printhead 0,0,0 in stead of center to the bed ? (#12545)

Use the Z_PROBE_END_SCRIPT option in configuration.h
#define Z_PROBE_END_SCRIPT "G0 X0 Y0 Z5"

Bob-the-Kuhn, 2018-12-24:
Try commenting out the following line:
`#define LEVEL_CENTER_TOO  `

garaguido, 2018-12-24:
Hi Bob

When I use the MBL procedure , the head (on pos 9) go up to 5mm and stay there , this is a parcial of the problem solved and not touching ( or to close)  the printbed at the center .

What I still wanted  is by a AUTO HOME position or G28 the head go to pos. X0 Y0 Z5 , is that possible ?

Regards

Guido

From: Bob Kuhn 
Sent: Monday, December 24, 2018 2:18 PM
To: MarlinFirmware/Marlin 
Cc: garaguido ; Author 
Subject: Re: [MarlinFirmware/Marlin] How can I set the coordinates in Marlin for homing the printhead 0,0,0 in stead of center to the bed ? (#12545)

Try commenting out the following line:
#define LEVEL_CENTER_TOO

Bob-the-Kuhn, 2018-12-28:
Please post your configuration.h & configuration_adv.h files.

Please do a **G28** followed by a **M114** and report the results.

garaguido, 2018-12-28:
Hi Bob

Here is my G28 followed by M114 , it is akword that the Z –20 look why ?

SENDING:G28
echo:busy: processing
echo:busy: processing
>>> M114
SENDING:M114
X:0.00 Y:0.00 Z:-20.00 E:0.00 Count X:0 Y:0 Z:0

Here is my configuration.h file:

* Based on Sprinter and grbl.
* Copyright (C) 2011 Camiel Gubbels / Erik van der Zalm
*
* This program is free software: you can redistribute it and/or modify
* it under the terms of the GNU General Public License as published by
* the Free Software Foundation, either version 3 of the License, or
* (at your option) any later version.
*
* This program is distributed in the hope that it will be useful,
* but WITHOUT ANY WARRANTY; without even the implied warranty of
* MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
* GNU General Public License for more details.
*
* You should have received a copy of the GNU General Public License
* along with this program.  If not, see <http://www.gnu.org/licenses/>.
*
*/

/**
* Configuration.h
*
* Basic settings such as:
*
* - Type of electronics
* - Type of temperature sensor
* - Printer geometry
* - Endstop configuration
* - LCD controller
* - Extra features
*
* Advanced settings can be found in Configuration_adv.h
*
*/
#ifndef CONFIGURATION_H
#define CONFIGURATION_H
#define CONFIGURATION_H_VERSION 010109

//===========================================================================
//============================= Getting Started =============================
//===========================================================================

/**
* Here are some standard links for getting your machine calibrated:
*
* http://reprap.org/wiki/Calibration
* http://youtu.be/wAL9d7FgInk
* http://calculator.josefprusa.cz
* http://reprap.org/wiki/Triffid_Hunter%27s_Calibration_Guide
* http://www.thingiverse.com/thing:5573
* https://sites.google.com/site/repraplogphase/calibration-of-your-reprap
* http://www.thingiverse.com/thing:298812
*/

//===========================================================================
//============================= DELTA Printer ===============================
//===========================================================================
// For a Delta printer start with one of the configuration files in the
[... 1675 lines cut ...]

//
// Factory display for Creality CR-10
// https://www.aliexpress.com/item/Universal-LCD-12864-3D-Printer-Display-Screen-With-Encoder-For-CR-10-CR-7-Model/32833148327.html
//
// This is RAMPS-compatible using a single 10-pin connector.
// (For CR-10 owners who want to replace the Melzi Creality board but retain the display)
//
//#define CR10_STOCKDISPLAY

//
// ANET and Tronxy Graphical Controller
//
//#define ANET_FULL_GRAPHICS_LCD  // Anet 128x64 full graphics lcd with rotary encoder as used on Anet A6
                                  // A clone of the RepRapDiscount full graphics display but with
                                  // different pins/wiring (see pins_ANET_10.h).

//
// MKS OLED 1.3" 128 × 64 FULL GRAPHICS CONTROLLER
// http://reprap.org/wiki/MKS_12864OLED
//
// Tiny, but very sharp OLED display
//
//#define MKS_12864OLED          // Uses the SH1106 controller (default)
//#define MKS_12864OLED_SSD1306  // Uses the

garaguido, 2018-12-29:
I have solved the problem with downloading the controller and doing a M502 followed by M500

The Z=0 now .

Is there no possible way , that the Marlin programmers , write som code for adjusting  the Z hieght after a HOME or  G28 for Z=5mm ?

Perhaps there are many peoples how are using a magnetic bed  ex. all the Ender3 users etc.....

Many thanks  for your support.

Have a great 2019 !!!!

Regards

Guido

From: Bob Kuhn 
Sent: Friday, December 28, 2018 5:24 PM
To: MarlinFirmware/Marlin 
Cc: garaguido ; Author 
Subject: Re: [MarlinFirmware/Marlin] How can I set the coordinates in Marlin for homing the printhead 0,0,0 in stead of center to the bed ? (#12545)

I'm sorry but I gave you bad information. Z_PROBE_END_SCRIPT is only executed after a G29 command.

The only ways I know to not have the head at Z = 0 after G28 are:

  a.. Write custom code 
  b.. Home to the Z_MAX endstop instead of the Z_MIN endstop. That means you'll need to play with the Z_MAX_POS setting in order to get the Z = 0 position correct. 

--------------------------------------------------------------------------------

Yes, the Z –20 is strange. From the configuration I would expect that it would be 0,0,0 after a G28.

I've downloaded your configuration into a controller and I get:

10:01:16.343 : N21 G28*32

10:09:41.561 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0

10:09:51.401 : N22 M114*23

10:09:51.401 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0

I have no idea where the Z:-20.00 is coming from.

I suggest downloading the controller again and then doing a M502 followed by a M500.