MarlinFirmware/Marlin #12548: M402 command not working

Rudgas, 2018-11-28:
### Description

Issuing M402 does nothing.

### Steps to Reproduce

1. M280 S0 P15 --> manually stow Probe
2. M401 --> deploy probe via Gcode, works
3. M402 --> no response

#### Additional Information

The servo can be controlled manually, issuing the commands M280 S0 P15 and M280 S0 P50 the probe will be deployed/stored correctly. Servo angles are entered as:

#define Z_PROBE_SERVO_NR 0   // Defaults to SERVO 0 connector.
#define Z_SERVO_ANGLES {15,50}  // Z Servo Deploy and Stow angles

Since the servo can be controlled via M280, wiring and code set-up should be ok. Further, since M401 works, at least the deployed angle is correctly communicated. I also tried to set the servo angle with M280 S0 P35 to see if there is perhaps a duplication of the stored angle, but still no reaction from M402.

[conf_confadv.zip](https://github.com/MarlinFirmware/Marlin/files/2623801/conf_confadv.zip)

thinkyhead, 2018-11-28:
I can’t see any obvious reason why `M402` would fail. See if it helps to adjust the servo delay and servo disable options. And also try enabling `DEBUG_LEVELING_FEATURE` just to make sure `M402` is invoking the probe handler.

Rudgas, 2018-11-28:
Servo delay is 400ms, what baffles me is that M280 and M401 works... 

Did as adviced, this is my output from: M111 S247 | M401 | M402
Probe was manually stored before, again M401 works as intended.

23:00:51.657 : echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
23:01:13.857 : echo:N23 M401*23
23:01:13.863 : current_position=(0.00, 0.00, 3.35) : set_probe_deployed
23:01:13.864 : deploy: 1
23:01:13.866 : do_probe_raise(2.00)
23:01:14.273 : >>> do_blocking_move_to(0.00, 0.00, 3.35)
23:01:14.275 : <<< do_blocking_move_to
23:01:14.282 : X:0.00 Y:0.00 Z:3.35 E:0.00 Count A:0 B:0 Z:1340
23:01:19.425 : echo:N24 M402*19
23:01:19.429 : current_position=(0.00, 0.00, 3.35) : set_probe_deployed
23:01:19.430 : deploy: 0
23:01:19.432 : do_probe_raise(2.00)
23:01:19.839 : >>> do_blocking_move_to(0.00, 0.00, 3.35)
23:01:19.844 : <<< do_blocking_move_to
23:01:19.846 : X:0.00 Y:0.00 Z:3.35 E:0.00 Count A:0 B:0 Z:1340

HA! Got the bugger! If
#define EDITABLE_SERVO_ANGLES
is not enabled, M402 doesn't work. :-)

thinkyhead, 2018-11-30:
I'l have a patch soon!

closed 2018-11-30 by thinkyhead