MarlinFirmware/Marlin #12549: [bugfix 2.0] Printer Event LEDs troubles

mr-miky, 2018-11-28:
I'm having some problems with RGBW lighting that should, in theory, indicate the heating / printing / job status.
From what I understand it should start from blue and, as the bed and the extruder heats up, it changes to red. In reality it happens that it starts in blue, it becomes light purple at the end of the bed heating and stays there until it reaches the temperature set for the extruder.
When the set temperature for the extruder is reached, it flashes red (3 or 4 if I remember correctly) and the lighting turns white and stays there forever. At the end of the print it should turn green but it does not. From the green I think it should go to blue because it is cooling but does not do anything.
I tried to look at the code but I can not understand the operating logic. Does anyone work ???
Who wrote / is following this piece of code?

thinkyhead, 2018-11-28:
> Who wrote / is following this piece of code?

From the GitHub web interface you can use the "Blame" button when viewing a file to see who has modified each line of the file.

Moyster, 2018-12-04:
There's a bit of confusion (maybe add some comments near the #define for RGB and RGBW to indicate RGB has 3+gnd legs and RGBW has 4+gnd legs) regarding leds.  
I have a ramps board and added a single LED to my extruder ( I made a quick video showing the color changes here : https://www.youtube.com/watch?v=aU5FLqL7tzs )  
It's an RGB led (1 led per R/G/B and when you light the 3, you get white) which is different from rgbW (which has 4 led, one per color and a real white led).  
All I had to do was enable `#define RGB_LED` and define my pins below ( `RGB_LED_x_PIN` ), I used the servo pins on my RAMPS shield, pin 4 / 5 / 6 and GND, and kept `RGB_LED_W_PIN -1` (unless you're using a rgbW, you shouldn't define a pin for white).  
When you enable `#define RGB_LED` , the default configuration enables `#define PRINTER_EVENT_LEDS` which is what you see in the youtube video I linked above :  
-> starts blue to purple (bed heating)  
-> purple to red (extruder heating)  
-> white during the print  
-> green when print succeed  
(note: canceling a print will keep the led lit and on the last color it was set to)  
(note2: purple is really washed off because most RGB led because red takes over blue, a "proper" purple would be : red=170, blue=255, green=0 )  

>> From the green I think it should go to blue  

That's not the default behavior, what you refer to is called `TEMP_STAT_LEDS` and you need to uncomment the #define to enable it.  
But overall it works fine, so I would suggest you to check your wirings / solderings / if your led is not broken (if you can plug your led to an arduino, just upload a sketch to test each led and lighting the 3 colors to get white, this way you can rule a burnt led in your RGB led).  

>> it flashes red  

afaik, it shouldn't flash, when bed is warm (led is purple), it should start heating your extruder (led is purple (blue + red) and blue is decreasing at extruder temp rises, transitioning to red over time).  
What kind of resistors or transistors did you put before your led ?

mr-miky, 2018-12-04:
I can assure you that it is not a hardware problem due to the simple fact that from the lcd menu I control the 4 mosfets that control a strip of RGBW LEDs quietly in pwm. I can adjust every single color without problems.
I designed and built the control board and I have some experience in designing and debugging hardware (ok I admit I started with the ZX81 like someone else here) so I can exclude a hardware problem (and I had already checked ;-))

I had actually missed the TEMP_STAT_LEDS option but I do not think it depends on this, anyway I try with the option activated tonight.

The thing that puzzles me, and that's why I asked, is that the heating of the BED is displayed correctly but not that of the extruder.
The only idea that came to me is that the bed is heated with the command M190 (Wait for Bed Temperature) at the end of which the temperature of the extruder M104 (Set Hotend Temperature) is set without waiting for the temperature to be reached and the real extruder heating occurs during G29 auto-bed-leveling ....
The doubt was that the transition from purple to red is only initiated by M109 and M104 had been forgotten.

I have looked at the source code and can not understand the logic behind the color change management based on the hotend and bed temperature.
Let's assume that the bed must reach 50 ° and the extruder  200 °, the transition from blue to red must take place in the arc of 250 °, but how do you calculate the blue / red transition correctly if the temperature information of the hotend is set only after having reached the bed's terget?

Another question, from what is it understood that the printing is finished?

Moyster, 2018-12-04:
It's always a good thing to double check, it felt like something was wrong when you said "red is flashing" and you omitted being a sinclair dinosaur in your first post :^) 

>> The thing that puzzles me, and that's why I asked, is that the heating of the BED is displayed correctly but not that of the extruder.  

It might be a dumb idea, but have you tried controlling your led strip with the RGB_LED feature (ommitting the White leds), to see if that's not a bug regarding the RGB**W**_LED feature ?

The commit introducing RGBW_LED is here if you want to look at it : https://github.com/MarlinFirmware/Marlin/commit/dac21ec680b348db8106e13cf202698e6c0dab12  

I don't know much about marlin yet so I won't assume anything without reading some more code, but I can at least confirm RGB_LED behave like it should here, so maybe RGBW_LED has an issue, good luck troubleshooting your setup :)  
flying over marlin_main.cpp, M109 will trigger the "purple to red" transition ( https://github.com/MarlinFirmware/Marlin/blob/dac21ec680b348db8106e13cf202698e6c0dab12/Marlin/Marlin_main.cpp#L6174 ) and the printer will wait for `wait_for_heatup` to light the white led (or all rgb colors at once).  
(same goes for M190 (bed heating i guess)

>> but how do you calculate the blue / red transition correctly if the temperature information of the hotend is set only after having reached the bed's terget?  

If M190 tracks bed temperature and wait for the desired one, it's pretty simple to add a starting temp (blue=255,r=0,g=0), then as the loop checks for bed temp, interpolate this value to 0-255 and update the red color to get the "blue to purple" effect.  
Same for M109, when M190 ends (bed is hot), M109 can start with blue=255,red=255,g=0 then turn the temperature of the extruder and decrease blue from 255 to 0 to have a "purple to red" transition, actually :  
```
const uint8_t blue = map(constrain(temp, start_temp, target_temp), start_temp, target_temp, 255, 0);
if (blue != old_blue) set_led_color(255, 0, (old_blue = blue));
```
is pretty clear (red is fixed to 255, and only blue is calculated and updated in M109's loop).  

Green is here : https://github.com/MarlinFirmware/Marlin/blob/dac21ec680b348db8106e13cf202698e6c0dab12/Marlin/Marlin_main.cpp#L1169  
and seem to rely on card_eof, i guess that's some kinda of function to know when they reached the file.gcode end of file, and it triggers green and a "press a button to resume" or a fixed delay (in case no screen, i assume).

mr-miky, 2018-12-05:
I believe I have found what causes the problem of green.
It simply can not work because only the print from uSD invokes onPrintCompleted ().
I thought it was based on the fact the temperature of the extruder and bed is set to 0 °;
Obviously, since I print using octoprint, it is never invoked onPrintCompleted ().

For the purple / red transaction problem I'm still trying to figure out what causes the anomaly but I suspect it is a further defect of the STM32Generic PWM.

mr-miky, 2018-12-06:
I did a bit of testing and I came to the conclusion that the transition from blue to purple and then purple to red works but on condition that:

use M190 for the bed
use M109 for the hotend

in short, the printer must be stopped waiting to reach the temperature. If you set the temperature of the hotend and in the meantime you do something else, the transition to red does not happen.

Similarly if you start printing with preheated bed and hotend the RGB remains completely blue while it should immediately become red ...

mr-miky, 2018-12-06:
Yes, the strip passes correctly from blue to red as long as it starts with a cold extruder and bed ... and I use M190 and M109.

edn247, 2018-12-15:
Perhaps it should be stated clearly that the GREEN on complete feature only work when printing from SD.

I was really confused as to what I had done to make it not show green when it was done since everything else was being shown properly.  I thought for sure I made some kind of issue by enabling case light and the LED menus...  but ended up with what sounds like the correct solution to make the printer (kind of) do what I was initially expecting it to do, since I don't print from SD.

I just added M150 codes, both in Octoprint and Cura, so when the printer pauses for filament change, Octoprint runs a macro with a color to let me know it needs attention (wish it could be 2 alternating colors - even more eye catching) and the end gcode in cura just has M150 U255 to set it to green.

I couldn't figure out how to replicate the click to turn off or back to white or similar...  just have to go into LED menu and choose white or off.

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-13 by mr-miky