MarlinFirmware/Marlin #12550: M303 parsing E value bug in M303.cpp

dayaan82, 2018-11-28:
The original code has E value parsing by parser.byteval('E'); It does not work for me.. so i couldn't tune the BED.. maybe a sign problem

Switching to parser.intval('E') solved the problem for me.

void GcodeSuite::M303() {

---->  const int8_t e = parser.intval('E');

  if (!WITHIN(e, 0
    #if ENABLED(PIDTEMPBED)
      -1
    #endif
    ,
    #if ENABLED(PIDTEMP)
      HOTENDS
    #endif
    -1
  )) {
    SERIAL_ECHOLNPGM(MSG_PID_BAD_EXTRUDER_NUM);
    return;
  }
.
.
.

dayaan82, 2018-11-28:
yep:

parser.h

static inline uint8_t value_byte() { return (uint8_t)constrain(value_long(), ---> 0, 255  <----)

thinkyhead, 2018-11-28:
The current code uses `parser.intval('E')`. What version of Marlin are you looking at?

closed 2018-11-28 by thinkyhead

dayaan82, 2018-11-28:
OMG... amateur hour!
I'm sorry.. i have MarlinTarantula-EasyConfig-2.0.x.

But still... byte is by default signed, so the bug in parser still remains.

thinkyhead, 2018-11-28:
The way `parser.value_byte()` works is intentional. It is meant to be unsigned, but to reject incoming negative values. Parameters interpreted by `value_byte` are legal when positive between 0-255.

dayaan82, 2018-11-28:
Ok.
Sorry for the confusion.

You can mark this as a mistake.