MarlinFirmware/Marlin #12554: pins_RADDS.h due+radds+full graphic display

Abysslabs, 2018-11-28:
Hello

Marlin 2.0.x (downloaded today) + Due + Radds + Full graphic display installed according (https://www.thingiverse.com/thing:1740725)

After update, i had no image on display with same settings used with Marlin 2.0.x ( downloaded first week of august) and working ok.

After comparing settings and files, I found "pins_RADDS.h" with differences between versions:
NEW version:
#define LCD_PINS_RS 42
#define LCD_PINS_ENABLE 43
OLD version:
#define LCD_PINS_RS 46
#define LCD_PINS_ENABLE 47

I changed the pin number and tested, display and encoder are working ok, but have follow problems:

It detects the SD card inserted, but no files are showen, tested with 2nd SD card same thing.
It seems that beep also is not working, need to do more tests.
I don't know if this is a bug, but acording to instructions the pins are wrong.
I don't know yet if the SD card and Beeper problem are also related with this pins problem, meanwhile i will keep doing more tests, but my programming skills are basic and hope get your feedback, and point me some solution.

Also attached Configuration and pins files.
[pins_RADDS.zip](https://github.com/MarlinFirmware/Marlin/files/2626816/pins_RADDS.zip)
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2626817/Configuration.zip)

thinkyhead, 2018-11-28:
This was changed by @Kevin in ae1271b2ae so let's find out why. Are you using the adapter mentioned in the comments?
```cpp
// The REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER requires
// an adapter such as https://www.thingiverse.com/thing:1740725
```

thinkyhead, 2018-11-28:
This was changed by @smoki3 in ae1271b2ae so let's find out why. Are you using the adapter mentioned in the comments?
```cpp
// The REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER requires
// an adapter such as https://www.thingiverse.com/thing:1740725
```

Abysslabs, 2018-11-29:
I made my self that adapter, as far as I can check, I have two schematics for that adapter, the most recent is according to the pinouts presents on Marlin at moment, and I have made the other version, so i need to rewire those two pins, and I could also check that it matchs with RADDS schematics, so new Marlin is correct.
The other pins are matching, so I need to investigate further why no SD-card and no beep.

Thanks for your time.

smoki3, 2018-11-29:
Yes I use the adapter which is in the comment. I think it’s the most common for the radds

Abysslabs, 2018-11-29:
@thinkyhead 
Rectified the connections, problem solved.
I will close this case.
Thanks for your time, and sorry anything.

@smoki3 
In fact I was using the GLCD connections, and now the standard as you say is RADDS2LCD.
everything is working ok.

LETS PRINT ;)

closed 2018-11-29 by Abysslabs