MarlinFirmware/Marlin #12558: Due Compile Error "undefined reference to `setup'" and "`loop'"

bwgeez02, 2018-11-29:
I need to change my steps for some printer diagnostics. Printer keeps shutting down, so I'm trying 1/64 microstepping instead of 1/128. I keep getting a compile error now. The only change I made is  default steps per axis in Config.h. I tried to compile with an earlier 1.9.0 build that has always worked and also 1.8.7 with the same results. I uninstalled Arduino and installed the latest beta 1.9.0 and also deleted the Arduino15 file in Appdata first just to be sure.

Arduino: 1.9.0-beta (Windows 7), Board: "Arduino Due (Programming Port)"

C:\Users\Brian\AppData\Local\Temp\arduino_build_114438/core\core.a(main.cpp.o): In function `main':

C:\Users\Brian\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/main.cpp:57: undefined reference to `setup'

C:\Users\Brian\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.11\cores\arduino/main.cpp:61: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Due (Programming Port).

bwgeez02, 2018-11-29:
Here is the sketch for main.cpp

/*
  main.cpp - Main loop for Arduino sketches
  Copyright (c) 2005-2013 Arduino Team.  All right reserved.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
*/

#define ARDUINO_MAIN
#include "Arduino.h"

/*
 * Cortex-M3 Systick IT handler
 */
/*
extern void SysTick_Handler( void )
{
  // Increment tick count each ms
  TimeTick_Increment() ;
}
*/

// Weak empty variant initialization function.
// May be redefined by variant files.
void initVariant() __attribute__((weak));
void initVariant() { }

/*
 * \brief Main entry point of Arduino application
 */
int main( void )
{
	// Initialize watchdog
	watchdogSetup();

	init();

	initVariant();

	delay(1);

#if defined(USBCON)
	USBDevice.attach();
#endif

	setup();

	for (;;)
	{
		loop();
		if (serialEventRun) serialEventRun();
	}

	return 0;
}

thinkyhead, 2018-11-29:
You have no `setup()` or `loop()` defined in your sketch.

bwgeez02, 2018-11-29:
It's the sketch straight from Arduino IDE.

thinkyhead, 2018-11-29:
I'm confused by your inclusion of this sketch. What does it have to do with Marlin?

bwgeez02, 2018-11-29:
Just that it is the sketch the compiler error refers to. Should I post in Arduino instead?

thinkyhead, 2018-11-29:
Yes. This is the issue queue for Marlin development. It is not a support forum.

If you remove the `setup()` and `loop()` lines your sketch will compile.

closed 2018-11-29 by thinkyhead