MarlinFirmware/Marlin #12560: UBL trouble shooting G29 P6 C going in wrong direction

kwc280, 2018-11-30:
### Description

When I run through the UBL and set my G29 P6 C3.40 for instance the gantry moves up when I tell it to go to Z0 after homing with G28 and running a G1 Z10 and G1 X150 Y150.  When it does take the measurement in the right direction it crashes into the bed and does not adjust from the bed to go higher when inputing and setting a G29 P6 C-3.4

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

gloomyandy, 2018-11-30:
That is a big adjustment in the mesh. What exactly are you trying to do? Why do you need to make such a large shift? 

You also need to post your configuration files (as we need to see how your machine is set up). It would also help if you provided a step by step description of exactly what you have done to create the problem (so how did you create your mesh, have you activated levelling etc.). You should also post the output from G29 T.

As a general point you will be better off creating your mesh with the correct values (using a well adjusted Z offset) rather than trying to adjust it with G26 P6

kwc280, 2018-12-02:
I will have to figure out how to get all that on here since the computer is in an area without internet.  The printer was shipped to me with the probe preset and the bed was all the way down along with the z end stop and was hitting the z stop at first before the gantry hit 0.  I left the end stop in the same position and raised the bed to get past that issue.  I had the printer working and then the bed got hit when something fell off a shelf so I reset the level after I tightened the gantry wheels back up in contact with the rails.  The G29 T was in the 8.75 range on all the corners. I used the following process to get to where I was:
G28
G29 P1
G29 T
Adjusted the level and repeated until level on all 4 corners.
G29 A
M500
M501
G28 
G1 Z10
G1 X150 Y150
lowered the gantry to the bed to set to where it had tension using the "move axis z" under the prepare menu and recorded that value and inserted it using the G29 P6 Cn.nn which in this case was 3.40mm.

I have since adjusted the end stop up about half way and my values on my mesh G29 T are now around 5.59 and the value found with the control box knob lowering is at 0.70mm to get resistance on the sheet of paper on the bed

gloomyandy, 2018-12-02:
You really need to post your configuration files. At the moment we have no idea what sort of printer you are using, what control board, what sort of probe, what version of Marlin. Without that information no one will be able to give you accurate help.

It looks like you have not correctly set your zoffset value to describe the location of the probe. It really is essential that this and the corresponding x and y offsets are set correctly, are they? Those mesh values are still much larger than normal (they are typically +/- 1 or less).

Roxy-3D, 2018-12-02:
> It looks like you have not correctly set your zoffset value to describe the location of the probe. 

This is very true.   In fact, part of the setup instruction to bring up UBL are to verify you can print a small object (for example a calibration cube) in the center of the bed with no bed leveling active.   Usually when we see these types of problems, this step hasn't been performed and the user is trying to use more complicated features of the bed leveling system to compensate.    

Can you home the machine and print a small object with no bed leveling active?

boelle, 2019-02-21:
@kwc280 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead