MarlinFirmware/Marlin #12563: [2.0.x] SWITCHING_NOZZLE doesn't raise hotend before switching nozzle

scott0122, 2018-11-30:
Manually switching nozzle doesn't raise hotend before switching nozzle.
And after nozzle switched, hotend lower TOOLCHANGE_ZRAISE.
If hotend switches 5 times, hotend lower TOOLCHANGE_ZRAISE*5 but no raise performed. 

Why "no_move" at line 645 in tool_change.cpp?

See attahed video for detail.
[MOV_1080.zip](https://github.com/MarlinFirmware/Marlin/files/2633224/MOV_1080.zip)

scott0122, 2018-12-02:
@GMagician Yes, SWITCHING_NOZZLE is enabled.

Look at the video for detail.
https://drive.google.com/file/d/1VrWl4w-bKRG25sbjCk7RTzCCj40wxEXp/view?usp=sharing

After deleteing [if (!no_move)] at line 645 in tool_change.cpp, problem is gone.

scott0122, 2018-12-02:
@GMagician herer~ https://www.facebook.com/scott0122/videos/10217433481006970/

scott0122, 2018-12-02:
Here is my configuration 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2637269/Configuration.zip)

GMagician, 2018-12-02:
You may try to change line 634 (current 2.0):
`current_position[Z_AXIS] += MAX(-zdiff, 0.0) + toolchange_settings.z_raise;`
with
`current_position[Z_AXIS] += MAX(-zdiff, 0.0) + (no_move ? 0 : toolchange_settings.z_raise);`
but logic has been changed since I used Dondolo and someone should take a look at this because the
`NOMORE(current_position[Z_AXIS], soft_endstop_max[Z_AXIS]);`
may create some bad behaviour when it limit position destroying real Z position

GMagician, 2018-12-02:
@InsanityAutomation / @thinkyhead what do you think about above post? I think you changed last these lines

GMagician, 2018-12-02:
@silentninja1 the problem here, with this cnf, is that above code will change Z position but Z is not moved. Here is summary of involded code:
`current_position[Z_AXIS] += MAX(-zdiff, 0.0) + toolchange_settings.z_raise;` raised new tool position
`if (!no_move)planner.buffer_line(current_position, planner.settings.max_feedrate_mm_s[Z_AXIS], active_extruder);` axis is not really moved
`current_position[Z_AXIS] += zdiff;` current position is adjusted (raise is still there)
`sync_plan_position();` now system thinks z is raised with adjusted pos for new tool
`do_blocking_move_to(destination[X_AXIS], destination[Y_AXIS], destination[Z_AXIS]);` move to original pos (but raise has never been done so axis move down)

GMagician, 2018-12-03:
@scott0122 have you tried my temporary solution?

thinkyhead, 2018-12-03:
> `current_position[Z_AXIS] += zdiff;` … current position is adjusted (raise is still there)

The adding of `zdiff` is to deal with the configured difference in height of the two nozzles. The new nozzle may be at a slightly different height, even without any movement having been done.

> `current_position[Z_AXIS] += MAX(-zdiff, 0.0) + toolchange_settings.z_raise;` raised new tool position.

The `zdiff` is also used in deciding how far to (actually) raise the nozzle, at minimum, to ensure clearance when the newly-selected nozzle could be at a slightly lower height.

GMagician, 2018-12-03:
@thinkyhead in second line I agree about zdiff but since you also add `z_raise` but you don't really move the axis because of `no_move` .. next in code you sync planner and again you move to destination hence Z will go move z_diff (and this is ok) but also z_raise (and this is wrong)

scott0122, 2018-12-03:
@GMagician I just delete "if (!no_move)"
My opinion is that if Z HOME is not Z_MAX, Z will rarely reach Z_MAX. So if Z HOME is Z MIN, regardless of whether there is autohome or not, it is better to raise hotend before switching nozzles.

GMagician, 2018-12-03:
I agree that nozzle must be always raised, my solution just adjust to new nozzle. Don't know what is better and I leave judgement to guys most smart than me

boelle, 2019-02-21:
@scott0122 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

scott0122, 2019-04-17:
> @scott0122 problem solved?

What I need is always raise hotend, no matter what it is homed or not

thinkyhead, 2019-05-16:
I think we have an option for this now. Remind me to check. I vaguely recall we added it for beds that fall when not powered.

closed 2019-07-11 by shitcreek