MarlinFirmware/Marlin #12564: Linear Advance actually make your printer slower

Dragondepierre, 2018-11-30:
I tested Linear advance 1.5 at 100mm/s and a small print took me almost twice the time then the same print without linear advance on at 60mm/s.

Is it normally ?

stefan85, 2018-11-30:
I agree, when I enable Linear advance the ramp up and down is much slower. I expected more of an pressure advance that controls the extruder and not the acceleration of the toolhead.

A while ago I used Klipper as a test. With Klipper I can print much faster with pressure advance turned on. This doesn'n change the acceleration of the toolhead, only the pressure in the nozzle.

gloomyandy, 2018-11-30:
Not an expert, but I suspect that your extruder acceleration and jerk settings may be limiting things.

closed 2018-11-30 by thinkyhead

Dragondepierre, 2018-11-30:
Well, I don't think so, my accelerations and jerk settings are very high ... What would you recommend ?

thinkyhead, 2018-11-30:
See if it helps to increase your overall print speed when using `LIN_ADVANCE`.

gloomyandy, 2018-11-30:
Perhaps if you had posted your configuration files someone may be able to suggest something.

SimonSolar2C, 2019-04-05:
I have also observed this on my Tevo Tornado, Bowden setup, using Linear Advance setting of 0.88
I am printing speed test cubes of 10mm/s up to 170mm/s to find maximum.  I set accelerations, and jerk and maximums all very high and could not acheive high speeds while printing.  ( I could get high speeds using S3Ds jog controls, and by typing G1 X300 F20000 ).
With all the settings very high, I could not get the 20mm cube to print super fast.
I then paused print, changed just the Linear Advance to zero, and resumed the print.  Then it printed super fast.
Whilst I agree LA should only be changing the extrusion, it is not.
What configurations are you looking for?
Here is my M503:
G21    ; (mm)
M149 C ; Units in Celsius
Filament settings: Disabled
M200 D1.75
M200 D0
Steps per unit:
M92 X79.76 Y79.76 Z399.29 E416.00
Maximum feedrates (units/s):
M203 X500.00 Y500.00 Z30.00 E55.00
Maximum Acceleration (units/s2):
M201 X50000 Y2500 Z120 E10000
Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
M204 P5000.00 R10000.00 T5000.00
Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> X<max_x_jerk> Y<max_y_jerk> Z<max_z_jerk> E<max_e_jerk>
M205 B20000 S0.00 T0.00 X15.00 Y10.00 Z0.40 E5.00
Home offset:
M206 X0.00 Y0.00 Z0.00
Auto Bed Leveling:
M420 S1 Z0.00
G29 W I0 J0 Z0.16654
G29 W I1 J0 Z-0.00126
G29 W I2 J0 Z-0.62486
G29 W I0 J1 Z0.28675
G29 W I1 J1 Z0.10393
G29 W I2 J1 Z-0.48712
G29 W I0 J2 Z0.30178
G29 W I1 J2 Z0.06636
G29 W I2 J2 Z-0.55474
Material heatup parameters:
M145 S0 H220 B80 F0
M145 S1 H240 B110 F0
PID settings:
M301 P29.26 I3.34 D64.11
M304 P128.82 I16.91 D245.36
Z-Probe Offset (mm):
M851 Z-2.25
Linear Advance:
M900 K0.00
Filament load/unload lengths:
M603 L0.00 U100.00

thinkyhead, 2019-04-11:
@Sebastianv650 — Can you provide any insight into the effect Linear Advance has on XY motion?

Sebastianv650, 2019-04-12:
I never had a core machine to test on, but i think it should work at least since we have made some changes for kinematik printers like deltas.
I'm quite sure the slowdown is caused by the low extruder jerk value of 5mm/s. As described in the documentation, this has to lower the printing acceleration as the needed extruder speed offset would lead to lost steps on the extruder otherwise.

If you want to print real fast, you have to make sure all of the included components are fast. Not only x and y ;-)

AckmaniacSoftware, 2019-04-19:
Tested Linear Advance today and also experienced a very slow print. Especially the first layers surprised me as they only filled all the area. Print speed was set at 100 mm/s but felt like 25mm/s. DEFAULT_EJERK is at standard 5.0

Sebastianv650, 2019-04-20:
Then please provide your print settings, i would guess you ate using a thick and / or wide first layer?

AckmaniacSoftware, 2019-04-20:
Dis some more tests and activated JUNCTION_DEVIATION which seems to make it quicker. Maybe i am totally wrong but now the speeds seem to be fine again. Also lowered retraction from 5 to 4 millimeters. LIN_ADVANCE_K is now at 0.7 for my bowden setup and the test cube came out really nice. With the linear advance calibration print you can also nicely test the retraction by looking at the stringing it creates when it shifts to the next testline.

vk2gib, 2019-06-29:
I have activated LA on marlin 1.1.9 yesterday.
Works great and noticeable difference on extrusion and corners.

Some complain about the extruder motion but thats the point! Pressure control to much x y movements instead of constant E push.

Yep..i too am experiencing slow prints. Twice as long as notmal estimate times!
I am about to fully lose retraction in cura to test speed..
And maybe bump jerks up heaps.
I usually printed at 45mm/s on a Tevo Terantula.
I have now pushed it to 70mm/s in cura and it prints well and seems faster.
Definately LA is over riding sone controls somewhere.
I have turned coasting off.

Im not sure what else to eliminate.
Ideas!!??

Sebastianv650, 2019-06-29:
@vk2gib it's most likely your extruder jerk setting. Please read the documentation:
http://marlinfw.org/docs/features/lin_advance.html#lin_advance-can-reduce-your-print-acceleration

vk2gib, 2019-06-29:
Yes..i have just pumped it to 7.0 .. it was 3.0
Printing much faster..will know quality in 20mins!

Previous i had junction deviation set in marlin also which loses or has no control over jerk settings (it doesnt look at them)..
Print was lousy with deviation on at 0.020.

Commented out deviation for this print and have jerk settings active again.
Seems to be good at 80mms right now.
We ate closer to getting lin_adv with speed active.
I also am NOT using any retraction as i feel we dont need it..
Keep you posted.

vk2gib, 2019-07-01:
@Sebastianv650..
Ok after extruder pump up , speed is back to printer.

I still have print estimates way off from slicer estimations though. Some upto 50% longer than stated.

Retraction was re-added to eliminate excessive stringing which is now a 3.0mm retraction in the slicer.
Retraction disbaled had ALOT of stringing.

Quality is good with LA active.
I had , at one point, junction deviation activated in Marlin as well which gave funky results so i disabled it again with better results.
Speed DID NOT increase with junction dev ON.

At the moment i am happy , although my advance seems high at K1.6 per test results.
Other group (FBook) users claim they are at K0.1 / 0.2!!
Not sure the deal there.

Thanks for your help