MarlinFirmware/Marlin #12565: [bugfix-2.0.x]Build for stm32f103

Laczen, 2018-11-30:
Hi, I am trying to build and test Marlin 2.0 on a stm32f103 device using HAL_STM32. I installed Arduino 1.8.7 and STM32Core. After adding some defines for STM32F1xx I was able to build Marlin and also to upload it.

The problem I'm having now it that all the device does is reset after 4s due to the watchdog. The serial communication works ok as I receive the startup communication with names, settings,...

What could be wrong?

tna11, 2018-11-30:
Hi,

Cant help you on the watchdog, sorry, but you maybe able to help me compile with the HAL_STM32 and arduino. 

Could you possibly say what defines you made? Having no success in compiling via the arduino ide.

Thanks

Laczen, 2018-11-30:
Hi @tna11, I didn't really make any define. I have noticed that arduino defines stm32f1xx, while in Marlin there are checks for defines of the form __stm32f1__. This was the case for the boards file. Then in HAL/shared/backtrace/unwmemaccess.cpp I had to make a modification to make sure that the correct start and end of flash and ram where used.
I hope that this information can help you

AnHardt, 2018-11-30:
To reset the watchdog timer it needs two components working together.
The temperature-isr must run to get the ADC values and
the thermal manager, called in idle(), must read them and adjust the heaters accordingly (even when not heating).

If for example the temperature-ISR takes longer than a ms (called about 1k/s) there is no time left for idle.

Look up if the used Marlin-HAL uses analogRead(). And look up adc_read_value(), what's called from that, in the core  - and get a crysis (adc setup for every read including calibration plus busy waitin for the conversions). 

Make a simple sketch to measure the duration of a analogRead(). If more or close to a ms you found the evil.

Laczen, 2018-11-30:
Hi @AnHardt, thanks for the reply.
 I have been looking at the HAL I used and I am doubting if I selected the best HAL. I used the HAL_STM32, and maybe I should use HAL_STM32F1. I think the HAL_STM32 has no specific adc routines, which could make adc slow. HAL_STM32F1 seems to have specific adc routines, but I am not sure they are used by Marlin.

tna11, 2018-12-01:
> Hi @tna11, I didn't really make any define. I have noticed that arduino defines `stm32f1xx`,while in Marlin there are checks for defines of the form `__stm32f1__`. This was the case for the boards [...]

Hi Laczen, thanks for replying. I'll look into that.

xC0000005, 2018-12-04:
The ST official core uses hardware PWM for everything and does not throw an error if the timer is already in use. The result is that if your board definition uses hardware PWM on a pin serviced by the same timer as the temperature ISR, the ISR will stop running as soon as the oversample value is set and the first pwmWrite() happens. 4 seconds later, reset. Just check your timer definitions versus pins used for pwmWrite to rule this out.

thinkyhead, 2018-12-07:
>  is there a way to specify that all outputs should use software pwm?

That would be easy if there was a way to switch between hardware and software PWM for all pins. (e.g., if we could elect to include a software-only version of `analogWrite`).  Right now we use whatever PWM is provided by the platform core, or we use a custom routine in `Temperature.cpp`. That is, we use `analogWrite` to set PWM in various cases (`M42`), and we don't have an option to choose software PWM within Marlin (AVR) except for the fans and heaters, which all run from within our own custom PWM timer (the temperature ISR at ~1kHz).

The only way we could alter this situation to make every pin PWM-able would be to make a PWM wrapper class and do all our PWM through the class —using hardware or software as available— or we could look for a library that already does this for us (which could potentially bloat the binary). On AVR especially, available timers is limited, so it's best to use as few as possible and a relatively low frequency (e.g., <=25kHz). A smart PWM class could manage the details and just pick the best timers and frequencies according to the needs.

Laczen, 2018-12-16:
Hi @thinkyhead, thanks for the reply. Since timers are limited on AVR this should set the standard. The best solution imo would be to use only one timer for all gpio (steppers, heaters, fans, servo...), and to provide two kind of changes: a gpio toggle and a gpio pulse. The gpio pulse would only be required on AVR where a two toggles cannot be scheduled fast enough.
Anyhow, I will close this issue as it has been answered.

closed 2018-12-16 by Laczen