MarlinFirmware/Marlin #12566: print pause does not recover.

ruggb, 2018-11-30:
### Description
print pause does not recover. 
when pause is clicked printer stops and goes to park position
when continue is clicked printer jams stops and is totally lost
this occurred on an October build so I updated to bugfix-2.0.x.c986239
I don't remember if I ever tried to use pause in the past.
using Repetier Host V2.1.3
windows 10-1803 17134.407
all my modifications to the files are flagged with //**

### Steps to Reproduce
1. start print
2. pause print
3. continue print

**Expected behavior:** [What you expect to happen]
when I click continue it should restart printing where it left off
**Actual behavior:** [What actually happens]
when continue is clicked printer jams stops and is totally lost

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
[bugfix-2.0.x.c986239 changed files.zip](https://github.com/MarlinFirmware/Marlin/files/2634790/bugfix-2.0.x.c986239.changed.files.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.

ruggb, 2018-12-01:
no joy. this is the print log. Don't know if there is any useful info here.
[print log.txt](https://github.com/MarlinFirmware/Marlin/files/2635868/print.log.txt)
What is a EEPROM version mismatch?

ruggb, 2018-12-01:
no joy. here is the print log from Repetier. Maybe some info there.
[print log.txt](https://github.com/MarlinFirmware/Marlin/files/2635875/print.log.txt)
What is the EEPROM version mismatch?

GMagician, 2018-12-01:
> What is the EEPROM version mismatch?

M502 and M500 to resync eeprom to firmware defaults

ruggb, 2018-12-01:
still no joy. When I do a hard reset to stop it from banging the stops, it resets then bangs the stops more. I must reset it again. Now that is strange.

ruggb, 2018-12-01:
did not mean to close this
when it starts banging the stops, I do a hard reset and it then starts banging the stops again. I must reset it again . Now that is very strange. It has done that every time.

ruggb, 2018-12-01:
manually entering m125 indicates it is an unknown command.

ruggb, 2018-12-01:
there appears to be a couple of inter-related issues here. Should I open issues for each??
1. pause won't restart
2. m125 command not recognized
3. setting Advanced pause to implement pause retract is not compatible with filament runout prevention, however the text says that FILAMENT_RUNOUT_SCRIPT is required??????

GMagician, 2018-12-01:
Well it's a good choice to split issues because otherwise there may be a misunderstanding
1) Have you done M502 and M500 to resync eeprom to firmware defaults?
2) M125 needs ADVANCED_PAUSE_FEATURE and PARK_HEAD_ON_PAUSE to be enabled
3) I can't see such in code, please give details with real error text

ruggb, 2018-12-02:
1. I did resync - no change in problem
2. & 3. (I guess) In order to pause printing and have M125, I need to have PARK_HEAD_ON_PAUSE enabled. In order to have PARK_HEAD_ON_PAUSE enabled I need ADVANCED_PAUSE_FEATURE enabled. In order to enable ADVANCED_PAUSE_FEATURE I need a filament runout sensor. I do not have a filament runout sensor. 

Does that mean I can no longer pause printing - which is why it doesn't work?????
Does not seem logical to not be able to pause printing just because I do not have a filament runout sensor. I do have EXTRUDER_RUNOUT_PREVENT enabled.

Is this design intent???

ruggb, 2018-12-02:
1. I did resync - no change in problem
2. & 3. (I guess) In order to pause printing and have M125, I need to have PARK_HEAD_ON_PAUSE enabled. In order to have PARK_HEAD_ON_PAUSE enabled I need ADVANCED_PAUSE_FEATURE enabled. In order to enable ADVANCED_PAUSE_FEATURE I need a filament runout sensor. I do not have a filament runout sensor. 
I do have EXTRUDER_RUNOUT_PREVENT enabled, which is incompatible with ADVANCED_PAUSE_FEATURE.
So I can't pause my print because I can't enable pause because I don't have a filament runout sensor.
AND if I did have a filament runout sensor and enabled pause I can't have EXTRUDER_RUNOUT_PREVENT because it is incompatible with pause.

Something does not seem logical here.
I should be able to pause my print, with or without a filament runout sensor.
I should also be able to prevent my extruder from jamming up by having EXTRUDER_RUNOUT_PREVENT (BTW, the name doesn't seem logical either)
When the extruder is parked/paused the EXTRUDER_RUNOUT_PREVENT should still function, but it might create a mess on CONTINUE - maybe shutting down the heater immediately on pause would be good, but then would you need to define something else for filament change??
I hope this is clearer to you than it is to me.

GMagician, 2018-12-02:
> In order to pause printing and have M125, I need to have PARK_HEAD_ON_PAUSE enabled. In order to have PARK_HEAD_ON_PAUSE enabled I need ADVANCED_PAUSE_FEATURE enabled. 

yes

> In order to enable ADVANCED_PAUSE_FEATURE I need a filament runout sensor

Not required

> I do have EXTRUDER_RUNOUT_PREVENT enabled, which is incompatible with ADVANCED_PAUSE_FEATURE.

that's true but I think this is due to the fact that ADVANCED_PAUSE_FEATURE add the filament change procedure and this may trigger extruder runout error since machine is idle and hotend is hot.
Also in pause it may be triggered during hotend re-heating after pause...

ruggb, 2018-12-03:
I will try it without EXTRUDER_RUNOUT_PREVENT enabled.
I would still like to have  EXTRUDER_RUNOUT_PREVENT. Is it possible to set up the the filament change procedure to handle having EXTRUDER_RUNOUT_PREVENT enabled??

ruggb, 2018-12-04:
OK. Maybe the machine is not lost, but I am.......
EXTRUDER_RUNOUT_PREVENT disabled
advanced pause disabled 
No joy. upon continue from pause, it is hopelessly lost, though it did not bang the stops.
M125 still does not work.
It is pausing because I have a pause script = G27 P2. If I take this out it pauses and restarts but does not move the nozzle.

new compile:
EXTRUDER_RUNOUT_PREVENT disabled
advanced pause enabled
park head on pause enabled.
no pause script
click pause in Repetier gives nothing. M125 pauses but does not park head. I get
echo: Press button to resume. >> which button? I pressed pause and get
echo: Press button to heat nozzle. >> again which button = pressing pause gives me continue screen Continue pressed but no heating occurs.

where do I go from here? Is this me or Marlin?

ruggb, 2018-12-05:
I set M125 as a park script in Repetier and it works.
It also works if I enter M125 as a command. Not sure why it started working now.
I do not have EMERGENCY_PARSER enabled.
If I click Pause in Repetier I must click continue before pushing the display button. Apparently, that is all pause does is stop sending (unless there is a script which is sent first).

NOW, can you make EXTRUDER_RUNOUT_PREVENT compatible with ADVANCED_PAUSE?
If for some reason I have turned on the hotend and do not start immediately, without it the nozzle will jam.

closed 2019-02-04 by thinkyhead