MarlinFirmware/Marlin #12569: LCDs backlight control support ? (brightness/blink/on-off/animation)

yucelll, 2018-12-01:
Hello. I have an idea:

128x64 GLCD, 
2004 4x20 Character LCD
or other backlight LCDs

I think;
**1. Brightness**: Users adjust change their LCD panel backlight (0 - 255 PWM control)
Three Option for this: 

- Day Light(%100)
- Night Mode (%50 or etc..)
- Custom Mode (Users option %0 -%100)

**2.Blink**: If printer error(max temp,min temp,axis problem.. etc) and kill button click, backlight blink functions (1sec. on 1 sec. off)

**3.On-off**: Users can on or off own backlight

**4.Animation**: If Printer power on; LCD backlight %0 between (users option or %100) in 4 sec.
example: Printer power on. 0.sec %0, 1.sec. %30 2.sec. %60 4.sec(users option or %100)

**But how?**
Using 1 pin for this. Then marhin configiration.h file define this property and add lcd menu line for control this.

Phr3d13, 2018-12-01:
> 3.On-off: Users can on or off own backlight
If you turn off the LCD backlight via the LCD, how are you going to be able to see the menus to turn the backlight back on?

yucelll, 2018-12-01:
> 
> 
> > 3.On-off: Users can on or off own backlight
> 
> If you turn off the LCD backlight via the LCD, how are you going to be able to see the menus to turn the backlight back on?

not completely closed 10% ..etc

gloomyandy, 2018-12-01:
Do most of these displays actually provide access to the LCD backlight control? Just had a quick look at the full graphic smart controller and that does not seem to have access?

mr-miky, 2018-12-05:
With 100k limiting the backlight current, the pwm control is not needed because there will never be enough current to illuminate the leds.
Maybe 100 Ohm?

yucelll, 2018-12-05:
Yes. lcd board standart connected with 100k.
But we can remove 100k.

mr-miky, 2018-12-06:
100k in series at the back light ??? If we consider 2.5 v of voltage drop on the backlight, assuming that the whole is powered at 5V, remain 2.5V that divided by 100k make 25 uA. With 25 micro amps the backlight does not make light or makes it so little that it looks dark.
Check the schematic ...

boelle, 2019-03-06:
what lcd and controller combo will even make this possible?

closed 2019-05-16 by thinkyhead