MarlinFirmware/Marlin #12570: Possible Bug: negative steps/mm not working. Give compile error in case of negative steps/mm

Louis-Cypher, 2018-12-02:
### Description

When setting negative steps/mm (in my case for a geared extruder) Marlin does compile correctly but the motors don't move properly.

### Steps to Reproduce

1.Set negative steps per minute in Configurarion.h
2. Upload
3. M502
4. M500

**Expected behavior:** 
Marlin should complain about a negative steps/mm value and ask user to invert motor direction instead or motors should run correctly.

**Actual behavior:** 
Marlin compiles without an error message. Motor moves very roughly if at all.

#### Additional Information

ManuelMcLure, 2018-12-03:
Instead of setting a negative value for steps/mm, you should set `INVERT_E0_DIR` (or `INVERT_E1_DIR`, etc.) to specify that the extruder motor should rotate in reverse.

Louis-Cypher, 2018-12-04:
If you read what I wrote it would have been clear that I am not searching for a solution (I have it, otherwise I wouldn't have been able to file this issue, right?), I am saying that Marlin either shouldn't compile in case negative values are set or it should be able to handle negative values (e.g. by changing the respective INVERT_DIR value and using positive values).

If somebody would point me to the correct peace of code to insert such functionality I'll try to change the code

thinkyhead, 2018-12-05:
All the steps/mm values are required to be positive. I can add a sanity check for that now that we have `static_assert` at our disposal.

Louis-Cypher, 2018-12-06:
Thanks thinkyhead, that is exactly what I would Marlin expect to do. Check for valid values.

thinkyhead, 2018-12-07:
Indeed. Previously this was not possible, because `static_assert` was not available, and preprocessor directives like `#if` don't recognize decimal values.

Louis-Cypher, 2018-12-09:
I wasn't aware of that. Could one tell Marlin to change negative steps/mm to a positive value and invert the STEP_DIR value? Of course such behaviour might make debugging more cumbersome

Louis-Cypher, 2019-01-19:
Then, from my point of view marlin should throw a compile error when negative steps-per-mm are defined

closed 2019-01-20 by thinkyhead