MarlinFirmware/Marlin #12571: [1.1.9 & 2.0.x-bugfix] G28 to X- Y- with unknown starting position sometimes tries to move X+ Y+ to safe homing point first

Maeyanie, 2018-12-02:
### Description

Sometimes, but not always, after a power-cycle, when I run a G28 (or G28 X0 Y0, or G28 X0 followed by G28 Y0) Marlin will first try to move to my safe homing point of X110 Y110 before starting to home to the X-min and Y-min end-stops.
However, it assumes the location at power-on is 0,0 so if it is actually more than halfway across the bed, it will slam the X and/or Y axis into the side of the bed until it figures it has reached the right spot.

### Steps to Reproduce

I have not been able to get this to happen reliably, sometimes everything works fine. I suspect there is some specific set of conditions which triggers it, but I haven't yet been able to determine what that is.

**Expected behavior:**
1. Do a homing move towards the X- and Y- endstops until they trigger
2. Move to X110 Y110 to probe Z.

**Actual behavior:**
1. Move X+110 and Y+110 from wherever it happens to be resting.
2. Do a homing move towards the X- and Y- endstops until they trigger
3. Move to X110 Y110 to probe Z.

#### Additional Information

I compiled with levelling debugging on, and got logs from both successful and error homing moves.
When everything works correctly, it gives this:
```
04:55:58.572 : >>> G28
04:55:58.572 : Machine Type: Cartesian
04:55:58.575 : Probe: FIX_MOUNTED_PROBE
04:55:58.580 : Probe Offset X:0 Y:-26 Z:-0.55 (Middle-Front & Below Nozzle)
04:55:58.583 : Auto Bed Leveling: BILINEAR (disabled)
04:55:58.592 : current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
04:55:58.592 : > endstops.enable(true)
04:55:58.594 : Raise Z (before homing) to 4.00
04:55:58.599 : >>> do_blocking_move_to(0.00, 0.00, 4.00)
04:55:59.739 : <<< do_blocking_move_to
04:55:59.742 : >>> homeaxis(X)
04:55:59.742 : Home 1 Fast:
04:55:59.745 : >>> do_homing_move(X, -360.00, [25.00])
04:55:59.751 : current_position=(0.00, 0.00, 4.00) : sync_plan_position
04:55:59.862 : <<< do_homing_move(X)
04:55:59.865 : Move Away:
04:55:59.867 : >>> do_homing_move(X, 5.00, [25.00])
04:55:59.871 : current_position=(0.00, 0.00, 4.00) : sync_plan_position
04:56:00.252 : <<< do_homing_move(X)
04:56:00.255 : Home 2 Slow:
04:56:00.257 : >>> do_homing_move(X, -10.00, 12.50)
04:56:00.261 : current_position=(0.00, 0.00, 4.00) : sync_plan_position
04:56:00.807 : <<< do_homing_move(X)
04:56:00.810 : >>> set_axis_is_at_home(X)
04:56:00.810 : For X axis:
04:56:00.813 : home_offset = 0.00
04:56:00.816 : position_shift = 0.00
04:56:00.818 : soft_endstop_min = -40.00
04:56:00.818 : soft_endstop_max = 200.00
04:56:00.821 : > home_offset[X] = 0.00
04:56:00.827 : current_position=(-40.00, 0.00, 4.00) :
04:56:00.827 : <<< set_axis_is_at_home(X)
04:56:00.832 : current_position=(-40.00, 0.00, 4.00) : sync_plan_position
04:56:00.837 : current_position=(-40.00, 0.00, 4.00) : > AFTER set_axis_is_at_home
04:56:00.840 : <<< homeaxis(X)
04:56:00.846 : current_position=(-40.00, 0.00, 4.00) : sync_plan_position
04:56:00.851 : current_position=(-40.00, 0.00, 4.00) : clean_up_after_endstop_or_probe_move
04:56:00.857 : X:-40.00 Y:0.00 Z:4.00 E:0.00 Count X:-32000 Y:0 Z:1600
04:56:00.857 : <<< G28
```

When it behaves incorrectly, it gives this:
```
05:02:22.481 : >>> G28
05:02:22.484 : Machine Type: Cartesian
05:02:22.487 : Probe: FIX_MOUNTED_PROBE
05:02:22.492 : Probe Offset X:0 Y:-26 Z:-0.55 (Middle-Front & Below Nozzle)
05:02:22.494 : Auto Bed Leveling: BILINEAR (disabled)
<Moves towards X+ and Y+, I hit Emergency Stop>
05:02:25.153 : Printer reset detected - initalizing
```

[Configuration.h and Configuration_adv.h](https://github.com/MarlinFirmware/Marlin/files/2636902/Configuration.zip)

gloomyandy, 2018-12-02:
How recent a version of 2.0.x are you using? A bug similar to this was fixed a few weeks ago. If you are not using the most recent 2.0.x build it is probably worth retesting.

Maeyanie, 2018-12-03:
Nope, tried it with the latest bugfix-2.0.x as of writing (261c6f4b96541435de786ddbafd491bc2a9d3bc6) and got it to happen on the 2nd try.

thinkyhead, 2018-12-03:
The log having no output in the case where `G28` fails is curious. It should be reporting that the endstops were turned on and that it at least attempted to home X and Y. I can't think of any code path that would lead `G28` to skip all that and initiate a move in the wrong direction.

I'd be curious to know what kind of behavior you get when `QUICK_HOME` is turned off.

Maeyanie, 2018-12-04:
I actually tried that already, and it didn't change, so I turned it back on. I didn't get a log though, will do that once my current print finishes and see if it's any different.

Maeyanie, 2018-12-04:
Never mind, found the problem. My client was set up to send a "G0 X110 Y110" as part of the homing process, left over from when I was using Repetier firmware which doesn't support safe homing.
When I switched over to Marlin, I removed most of the work-arounds I had, but apparently missed a line.

closed 2018-12-04 by Maeyanie

thinkyhead, 2018-12-07:
It may help to enable `NO_MOTION_BEFORE_HOMING` if you might be using some of that old G-code.