MarlinFirmware/Marlin #12577: [2.0.x] MKS SBASE Compile error 1: PlatformIO

martinfric1, 2018-12-03:
### Description
When using newest Marlin 2 release for my MKS Sbase
PlatformIO sends error 1, but when using release from 04. 10. 2018 it compiles OK in about 120-150 seconds compared to newest version - 280-350 seconds and than FAIL, and every time I click build(compile) another file is problem-first time i clicked build was
```
Assembler messages:
Fatal error: can't create .pioenvs\LPC1768\src\src\libs\buzzer.cpp.o: No such file or directory
*** [.pioenvs\LPC1768\src\src\libs\buzzer.cpp.o] Error 1
```
Second time was:
```
Assembler messages:
Fatal error: can't create .pioenvs\LPC1768\src\src\lcd\menu\menu_main.cpp.o: No such file or directory
*** [.pioenvs\LPC1768\src\src\lcd\menu\menu_main.cpp.o] Error 1
Assembler messages:
Fatal error: can't create .pioenvs\LPC1768\src\src\lcd\menu\menu_motion.cpp.o: No such file or directory
*** [.pioenvs\LPC1768\src\src\lcd\menu\menu_motion.cpp.o] Error 1
```
If I clicked 1 more time another file would be the problem
Configuration file:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2638698/Configuration.zip)
### Additional Information
Platformio log:

````
Executing task: C:\Users\Speedy\.platformio\penv\Scripts\platformio.exe run 

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: arduino)

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1768
HARDWARE: LPC1768 100MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(strict)
Collected 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.7
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.2.0 #5d3abfe

Unable to find destination disk.  File must be copied manually.

Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\HAL_LCD_I2C_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\HAL_LCD_pin_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\usb_serial.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\watchdog.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\I2cEeprom.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\SpiEeprom.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
[... 222 lines cut ...]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [ERROR]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
============================================================ [ERROR] Took 306.43 seconds ============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

martinfric1, 2018-12-04:
Tried deleting, but same problem persisted:

```
> Executing task: C:\Users\Speedy\.platformio\penv\Scripts\platformio.exe run <

Processing LPC1768 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/master.zip; board: nxp_lpc1768; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1768.html
PLATFORM: NXP Arduino LPC176x > NXP LPC1768
HARDWARE: LPC1768 100MHz 31.80KB RAM (464KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap) EXTERNAL(blackmagic, jlink)
Converting Marlin.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(off) COMPATIBILITY(strict)
Collected 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <Servo> 1.0.0
|-- <LiquidCrystal> 1.0.7
|-- <U8glib-HAL> 0.4
|-- <TMCStepper> 0.2.0 #5d3abfe

Unable to find destination disk.  File must be copied manually.

Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\HAL_timers.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\MarlinSerial.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\main.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\HAL_LCD_I2C_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\HAL_LCD_pin_routines.c.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_hw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_st7920_sw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\u8g\u8g_com_HAL_LPC1768_sw_spi.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\usb_serial.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\HAL_LPC1768\watchdog.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\I2cEeprom.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\SpiEeprom.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\backtrace.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarm.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarm_arm.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarm_thumb.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarmbytab.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwarmmem.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwinder.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\backtrace\unwmemaccess.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\persistent_store_api.cpp.o
Compiling .pioenvs\LPC1768\src\src\HAL\shared\servo.cpp.o
Compiling .pioenvs\LPC1768\src\src\Marlin.cpp.o
Compiling .pioenvs\LPC1768\src\src\core\serial.cpp.o
Compiling .pioenvs\LPC1768\src\src\core\utility.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\I2CPositionEncoder.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\Max7219_Debug_LEDs.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\baricuda.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\bedlevel\abl\abl.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\bedlevel\bedlevel.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\bedlevel\ubl\ubl.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\bedlevel\ubl\ubl_G29.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\bedlevel\ubl\ubl_motion.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\caselight.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\closedloop.cpp.o
Compiling .pioenvs\LPC1768\src\src\feature\controllerfan.cpp.o
[... 201 lines cut ...]
Environment megaatmega1280              [SKIP]
Environment at90usb1286_cdc             [SKIP]
Environment at90usb1286_dfu             [SKIP]
Environment DUE                         [SKIP]
Environment DUE_USB                     [SKIP]
Environment DUE_debug                   [SKIP]
Environment LPC1768                     [ERROR]
Environment LPC1769                     [SKIP]
Environment melzi                       [SKIP]
Environment melzi_optiboot              [SKIP]
Environment rambo                       [SKIP]
Environment sanguino_atmega644p         [SKIP]
Environment sanguino_atmega1284p        [SKIP]
Environment STM32F1                     [SKIP]
Environment STM32F4                     [SKIP]
Environment ARMED                       [SKIP]
Environment teensy35                    [SKIP]
Environment malyanm200                  [SKIP]
Environment esp32                       [SKIP]
Environment fysetc_f6_13                [SKIP]
============================================================ [ERROR] Took 419.60 seconds ============================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

gloomyandy, 2018-12-04:
Sorry I have no idea. I've seen platformio get into a state in the past (especially from atom), are you using atom or vscode? I would be very tempted to remove platformio and try again (and if you are using atom I'd get rid of that and switch to vscode as this seems to be the supported ide these days). Sorry if this is not much help.

DatenThielt, 2018-12-04:
Just as some aditional info,  I have had no problem compiling the latest for my MKS SBASE yesterday and today using VSCode with Platform IO

hectori4502, 2018-12-17:
I can help you with this one: (the only thing a I have found)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.piolibdeps/U8glib-HAL_ID1932/src/clib/u8g_rot.c:48:27: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
u8g_dev_t u8g_dev_rot = { u8g_dev_rot_dummy_fn, NULL, NULL };
^~~~~~~~~~~~~~~~~~~~
go to this folder: (open  the file: u8g_rot.c)
\Marlin-bugfix-2.0.x\.piolibdeps\U8glib-HAL_ID1932\src\clib

and change this part in line 37:
#include "u8g.h"

uint8_t u8g_dev_rot90_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg);
uint8_t u8g_dev_rot180_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg);
uint8_t u8g_dev_rot270_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg);

uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg)
{
  return 0;
}

BY this:
#include "u8g.h"

uint8_t u8g_dev_rot90_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg);
uint8_t u8g_dev_rot180_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg);
uint8_t u8g_dev_rot270_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg);

#ifdef __XC8
uint8_t u8g_dev_rot_dummy_fn(void *u8g, void *dev, uint8_t msg, void *arg)
{
return 0;
}
#else
uint8_t u8g_dev_rot_dummy_fn(u8g_t *u8g, u8g_dev_t *dev, uint8_t msg, void *arg)
{
return 0;
}
#endif

the other warnings I still waiting to some one be solve it.
M43 #define PINS_DEBUGGING do not work make the printer halt.
G33 //#define DELTA_AUTO_CALIBRATION shows warning.

martinfric1, 2018-12-23:
Sorry for delay answer. 
The problem with compiling was solved by deleting .pioenvs folder, and reinstalling PlatformIO.

Marlin now works, thanks everybody for your input and suggestions.

Regards 
Martin

boelle, 2019-02-21:
@martinfric1 maybe click the green close button?

boelle, 2019-03-13:
@martinfric1 ^^^^^^^^^^

closed 2019-05-16 by thinkyhead