MarlinFirmware/Marlin #12578: Z axis Stepper motors in series not working

AlanB9, 2018-12-03:
Hi Everyone,
I must say that I did get this working with the help of Bob-the-Kuhn's pins_RAMPS.zip
but for the life of me I just cannot fathom out what I,ve now done.Only when I now move the Z axis
they only move one at a time in whichever direction up or down, which would be comical
if it was'nt happening to me. 
What hair I have left I would love to keep for at least a few more years so any help would be much
appreciated.

Thank you Alan.

Have you read Marlin's Code of Conduct? By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/code_of_conduct.md

Do you want to ask a question? Are you looking for support? Please don't post here. Instead please use the Marlin Firmware forum at http://forums.reprap.org/list.php?415 or the Marlin Facebook Group https://www.facebook.com/groups/1049718498464482/.

Before filing an issue be sure to test the 1.1 and/or 2.0 "bugfix" branches to see whether the issue is already addressed.

-->

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

AlanB9, 2018-12-03:
Thank you for the reply Andy,
I do apologise for the lack of information my mistake, Im using 1.1.9 the printer is a anet a8 with a few
mods. I was having some trouble with the z ports on the ramps board so I remapped the E1 with the
pins_RAMPS.zip from Bob-the-Kuhn did the cable splicing all went great then something made me 
go back to version 1.1.8 what it was is beyond me at the moment, but I went back to 1.1.9 then the 
motors were all snakes and ladders and at that point I lost the plot.
This is the link to my config files on google drive.
https://drive.google.com/file/d/11O6C6aymaV_VoD76c55r1NouTQys0GT7/view?usp=sharing

gloomyandy, 2018-12-03:
So did you change any of the wiring or is the only change that you went from one version of Marlin to another? If your motors are connected correctly in serial it is hard to imagine how then can do anything other than both turn at the same time, they may go in different directions, but that is a wiring issue not a firmware problem.

AlanB9, 2018-12-03:
I would have to agree with you had after the wiring everything was not as expected, but seeing the motors one up 10mm then other side up 10mm and so on Im at a complete loss.
Im thinking of a clean install of windows and go from there.
Thank you once again.

pipersw, 2018-12-04:
I have a RAMPS 1.4 and now a MKS GEN 1.4, Anet A8 printer with two steppers motors in serie, Marlin bugfix 1.1.9 and all work like a charms.
But I am at 24V for the drivers of the steppers.
Are you in 12V ?

AlanB9, 2018-12-04:
Yeah 12v, my problem was caused by the : Default_Axis_Steps_Per_Unit { 100, 100, 4000, 95 } once I 
reduced the 4000 to 400 the world was again in perfect harmony, well my little world anyway.

closed 2018-12-04 by AlanB9