MarlinFirmware/Marlin #12580: Z offset after manual bed leveling with M851 unavailable

swilkens, 2018-12-03:
### Description

I assume this is something trivial, but social media channels directed me here.

As many do, I level my print bed with a feeler gage (0.1 mm thickness) and bed leveling screws. This results in the nozzle being at Z0 while the nozzle is actually 0.1 mm removed from the print bed - introducing an error. Starting a 0.2 layer height print at this point will result in the printer printing the first layer 0.3 mm removed from the bed. This is undesired, so I searched how to compensate for this Z height delta of 0.1 mm.

I assume this is something everybody must do, as every manual leveling "tool" (e.g. paper, feeler gage, dinosaur hair) will have a thickness that needs to be offset?

Situation is as follows:
[image]

My question: How do I compensate for this "error" through Marlin?

- `M851` reports as an unknown command, I have no bed leveling systems enabled or probes defined as I have no direct need for them and flash space comes at a premium.

- `M206` does not allow a compensation in the direction that I require. The printer will not move beyond the end stops which are triggered while the printer is in its Z0 state, at 0.1 mm distance from the bed.

- Mechanically moving the endstops after every leveling run is undesired

#### Additional Information

* Running recent Marlin bugfix-2.0.x
[sw-marlin-configs.zip](https://github.com/MarlinFirmware/Marlin/files/2639412/sw-marlin-configs.zip)

thinkyhead, 2018-12-03:
Try `M206 Z0.1`, `M500`.

swilkens, 2018-12-03:
@thinkyhead I found that gcode reference as well, that was the first thing that I tried. It seems that the offset set by `M206` does not allow for an offset that would drive the Z position of the printer closer to the bed while the endstop is triggered.

To clarify:

1. G28 (thus endstops are triggered during all of this)
2. Level bed using 0.1 mm gage
3. M206 Z-0.1 (note the negative)
4. G1 Z0 - nozzle moves away from bed as expected

1. G28 (thus endstops are triggered during all of this)
2. Level bed using 0.1 mm gage
3. M206 Z0.1 (note the positive)
4. G1 Z0 - z-axis does not move.

swilkens, 2018-12-03:
Courtesy of a  facebook group member, a work-around to M206 not solving this would be to do the following:

1. G28
2. G1 Z0.1 (or whatever the thickness of your instrument is)
3. Level the bed using the instrument

G1 Z0 will now result in the nozzle actually being at 0 distance from the bed.

This is workable for situations where M851 is not available when no bed leveling features are enabled. As I think the firmware is not responsible for this, I am hereby closing this issue report.

closed 2018-12-03 by swilkens