MarlinFirmware/Marlin #12581: miniViki not working anymore with Re-Arm

gunnarfreyr, 2018-12-03:
So i just downloaded a fresh copy of Marlin 2.0 bugfix, configured it and compiled .. But for some reason when i flash the firmware, the miniViki LCD just does not want to work.. Even if I try to config as little as possible for the Re-Arm to work, the LCD still just wont work... 

After looking over the commits , i noticed this one from 12 days ago by @thinkyhead , and I wonder if something got changed that should not be changed or something ? https://github.com/MarlinFirmware/Marlin/commit/b738972e690ba415961b9c164a00fb19864ee39f 

## Hardware ##
Re-Arm controller 
Ramps 1.4sb 
miniViki LCD 
TMC2130 drivers on all axis and extruder

gloomyandy, 2018-12-03:
So have you tried older versions with the same configuration? Which ones work?

gunnarfreyr, 2018-12-03:
Yes, however the last compiled version i have is pretty old, it's since April (i took a printing pause for the summer cause of other projects) and that flashes fine and everything works as expected

gloomyandy, 2018-12-03:
Well if you want to help try doing a binary chop between then and now, try a version from say August and see if that works if it does try something around October and so on. Anything you can do to narrow things down will help. If you think that a particular change may be the cause then try a version just before that change. Remember that very few devs will have access to your hardware.

gunnarfreyr, 2018-12-03:
yeah im currently cheaking too see if I have any zips with the firmware from august or sometime like that.. Is there an option to download older releases here on github ?

gloomyandy, 2018-12-03:
Yep you can easily download a zip with all of the files from a particular date. Go to the main Marlin github repo, and select the 2.0.x branch then click on the Marlin folder, once there select the history tab and browse back to a date you want to test, then click on the <> symbol at the right of the commit you want to try and it should bring up the repo at that date/time which you can then download/clone.

There is almost certainly a better way to do this!

gunnarfreyr, 2018-12-03:
thank :) i didnt find the history feature when i was looking earlyer :) But now im just gonna start going backwards untill i get a success, and then i will post an update :)

gunnarfreyr, 2018-12-06:
So looks like the issue starts at Oct 14th .. 

This is the last one that works : https://github.com/MarlinFirmware/Marlin/tree/254b68fe2c83724b2844c0b0e2aea286471261e0

So my guess is that 
"Update the sdcard usb sharing config and apply to Re-ARM"
https://github.com/MarlinFirmware/Marlin/commit/42fb7c14b09b6e484334f0a39f1e1ae3d6fa5381#diff-b1e3f1b10558445963aa5bbf4d20249c

 starts the issue or 
"Remove debug output from flash persistent store implementation"
https://github.com/MarlinFirmware/Marlin/commit/e5b4bf22d446de0781b5d0ca9faf67c472ae7a11#diff-b1e3f1b10558445963aa5bbf4d20249c

gloomyandy, 2018-12-06:
Does editing the following settings in the pins file help...
``` C
    #define DOGLCD_SCK     SCK_PIN
    #define DOGLCD_MOSI    MOSI_PIN
```
Change to...
``` C
    #define DOGLCD_SCK     P0_15
    #define DOGLCD_MOSI    P0_18
```
If that does not help you might also want to try adding the define (try with and without the above changes)...
``` C
#define LPC_SOFTWARE_SPI
```

gunnarfreyr, 2018-12-06:
Change to...

    #define DOGLCD_SCK     P0_15
    #define DOGLCD_MOSI    P0_18
##

Worked ! Downloaded the latest bugfix , changed the pins and it flashed and lcd works :) Excellent work, thank you :)

gloomyandy, 2018-12-06:
@p3p @thinkyhead can you see any problem with just making this change a standard setting in the pins file?

gloomyandy, 2018-12-07:
You have things defined in our pins file so that Marlin will use onboard SD card slot rather than the one on the LCD. This same card will also be shared with your PC via USB. This allows you to slice stuff on your PC and then copy it to the SD card via USB then print it from the printer without swapping the SD card about. To allow Marlin to access this this card yoiu will need to choose InitSD from the menu. Thsi will disconnect the card from the PC.

If you want to use the LCD SD card slot you can change the following....
``` C
//#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
#define LPC_SD_ONBOARD        // Marlin uses the SD drive on the control board
```
to
``` C
#define LPC_SD_LCD          // Marlin uses the SD drive attached to the LCD
//#define LPC_SD_ONBOARD        // Marlin uses the SD drive on the control board
```
If you do this the onboard SD card will still be shared with your PC but you will not be able to access it from Marlin. If you do not wish to share the card with your PC you can change the following...
``` C
//#define USB_SD_DISABLED
#define USB_SD_ONBOARD
```
to
``` C
#define USB_SD_DISABLED
//#define USB_SD_ONBOARD
```

I don't see any reason why eeprom should not work. What makes you think it is not working?

gunnarfreyr, 2018-12-07:
Commenting out the onboard SD worked :) No need to disable the second one :) But my suspicions about the EEPROM are cause it will not save any values , but im gonna do some more testing later today with the eeprom

closed 2018-12-08 by thinkyhead