MarlinFirmware/Marlin #12583: 'encoderPosition' not declared in scope after #12429

fiveangle, 2018-12-03:
Cleaning up my CircleCI failures and found this one that likely is affecting other builds.  I'm not too up on the LCD menu system and so much hand changed so couldn't track down the source of the problem.

Load default Cartesio example config and receive this error:

```
Marlin/src/lcd/ultralcd.cpp: In function 'void _reprapworld_keypad_move(AxisEnum, int16_t)':
Marlin/src/lcd/ultralcd.cpp:287:7: error: 'encoderPosition' was not declared in this scope
encoderPosition = dir;

```

-=dave

boelle, 2019-02-21:
@fiveangle still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

fiveangle, 2019-05-23:
I found it checking compile of the default configs. Anyone can do the same. I don't even have the hardware in question.

fiveangle, 2019-05-23:
@boelle - thanks for the dedication in trying to clear the issues. Another way too contribute is to understand a bit about the nature of the issues posted. A compile failure when the configs are posted or otherwise referenced can be checked by anyone against the latest head to see if a regression had been resolved, or remains an issue. This allows issues abandoned by their original reporter (or those that did not see the updates sure to life priories, which are a reality of open source) to not be lost even though they likely still are present. 

Plus learning to compile Marlin is a fun exercise for anyone who is an enthusiast :)