MarlinFirmware/Marlin #12585: Stopping a print should move +Z axis a little bit

PERECil, 2018-12-04:
Hello,

When stopping a print when printing the first layer, the print head will stop exactly where it is, without moving. However this can lead to damaging the bed coating, as the print head is hot, and may lead to localized melting (I had this in two different occasions, and manually moving the head by navigating through the menu was slow enough to damage the coating - I use a magnetic coating btw, it may not be the case with other coatings)

What I suggest is that, when forcefully stopping a print, the head should at least move a little bit in the Z axis to avoid damaging the coating.

Is this possible? Could you implement it?

pipersw, 2018-12-04:
In your slicer, you could add some GCODE to park the head away from the print.

G1 Z+1 F400
G1 X0 F1000
G1 Y200 F100

PERECil, 2018-12-04:
Either I'm not sure where to find this in Cura, either we didn't understood each other. 

I'm talking about shutting down the print by using the "stop" command on the printer board, not when the print is ended normally. I don't think the slicer may be able to inject GCode in this particular event (I'm using SD cards, not controlling the printer through USB).

thinkyhead, 2018-12-08:
There are a number of places where injecting additional user-configurable commands might be useful, so we're looking at implementing a more general set of such commands. For now, you can make the following change to `Marlin.cpp` => `loop()`…
```diff
void loop() {

  for (;;) {

    #if ENABLED(SDSUPPORT)
      card.checkautostart();

      if (card.flag.abort_sd_printing) {
        card.stopSDPrint(
          #if SD_RESORT
            true
          #endif
        );
        clear_command_queue();
        quickstop_stepper();
        print_job_timer.stop();
        thermalManager.disable_all_heaters();
        zero_fan_speeds();
        wait_for_heatup = false;
        #if ENABLED(POWER_LOSS_RECOVERY)
          card.removeJobRecoveryFile();
        #endif
+      enqueue_and_echo_commands_P(PSTR("G91\nG1 Z4 F200\nG90"));
      }
    #endif // SDSUPPORT

    if (commands_in_queue < BUFSIZE) get_available_commands();
    advance_command_queue();
    endstops.event_handler();
    idle();
  }
}
```

closed 2018-12-08 by thinkyhead

mauerbad, 2018-12-21:
On my Anet A6 is this code not working. I tried this on more postions in this loop. Do you have an idea why not? The difference i have is that i have to use "enqueue_and_echo_command" and not "enqueue_and_echo_commands_P" because then i get an error. Is not defiened....
It woult be great if you can help me, because every time after break i have to move manually.

mauerbad, 2018-12-21:
With not working i mean no error. It looks like the printer ignore the line.

7eggert, 2019-01-06:
¢¢: Not resolved, neither is the years-old issue being referenced. Nobody wants the nozzle to burn into the failed object (or the build plate) while the steppers prevent oneself from moving it away. Doing a pause/stop **on the LCD menu** instead is a cumbersome workaround. Also nobody will have the time to walk through the menus on an emergency, they'll switch off the printer instead. Thus the current behavior isn't sane and qualifies as a bug. At least, it should inject the release code `#if ENABLED(SD_FINISHED_STEPPERRELEASE) …`. Sorry for needing to rant, but this bugs me since forever.

@mauerbad You probably wrote `_command_p` and not `_command**s**`? `_commands_P` works on strings in the ROM, but `_command` will only work on a copy into the RAM. Therefore it will execute garbage from a random memory location or - if you are lucky - nothing at all.

... Implemented that, tested it on my printer (Anet A8), works:
https://github.com/7eggert/Marlin/tree/command_on_stop
Add this to your Configuration_adv.h:
```
#define SD_STOPPED_STEPPERRELEASE true
#define SD_STOPPED_RELEASECOMMAND "G91\nG0 Z5\nM84" // or whatever you want
```

(Not a pull request yet as the configs need to be updated, too. Q: Should the default be off or on; and if on, what should be the default gcode to execute; same as on finishing a SD print?)