MarlinFirmware/Marlin #12586: Delta on 1.1.9: On Begin and End of the print the effector seems to accelerate at maximum

dgrat, 2018-12-05:
### Description

I noticed on my Delta printer that begin and end moves happen at maximum rate. During my start gcode, the effector goes to center and is appr. 5 cm over the build plate. After a moment the print starts and the effectors moves to the build plate. In that moment the movement is very fast, which could lead to missing steps. Actually, the motors make cracking sounds. Same for the movements after prints. Setting a maximum feed rate und unit does not really help. 

**Expected behavior:** 

All movements are within maximum feed rates

**Actual behavior:**

Some movements are out of maximum feed rates which overloads motors/drivers.

dgrat, 2018-12-08:
I added a movie showing the problem at print start. At the end this behavior is more problematic.

https://www.dropbox.com/s/uqawduk9p4fi083/VID_20181208_101152.mp4?dl=0

boelle, 2019-02-21:
@dgrat problem still there in bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead