MarlinFirmware/Marlin #12587: Cannot Compile for Anycubic Delta with many features enabled.

nickaus, 2018-12-05:
I'm trying to compile 1.1.9 for an Anycubic Delta Plus using the config file from example_configurations->delta->anycubic->kossel

It compiles fine with some of my options / features enabled but add one more any it fails.
 Remove one, any one and it complies !

Features are 
Board Fan (stepper driver fan)
Level Probe (type 2)
Heated Bed (type 2)

With these enabled it won't compile.
 Remove one and it will.

Errror is 

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries\LiquidCrystal\src\LiquidCrystal.cpp: In function 'clear.constprop':

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries\LiquidCrystal\src\LiquidCrystal.cpp:180:1: internal compiler error: Segmentation fault

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/2648445/Config.Files.zip)

boelle, 2019-02-21:
@nickaus problem solved?

nickaus, 2019-02-24:
Not solved.
Download 1.1.9 again today, configured and tried to compile = same problem.

boelle, 2019-02-24:
try bugfix 2.0

nickaus, 2019-02-24:
Compiles in 2.0 unless you enable #define FILAMENT_RUNOUT_DISTANCE_MM 
 With that enables the compile fails with a segmentation fault.

boelle, 2019-02-24:
@thinkyhead configs are in first post, can you see anything wrong?

nickaus, 2019-02-24:
Actually I take that back.
With all of the options enabled (probe, heated bed & runout sensor) it won't compile in 2.0 either.

boelle, 2019-02-24:
i think we had an issue where the runout sensor enabled made it not compile, not sure

@thinkyhead could it be that we still have a gremlin in the closet?

thinkyhead, 2019-03-03:
I'd like to have @nickaus's configurations for 2.0.x to see where the compile fails.

nickaus, 2019-03-03:
Config files attached
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2923061/Configuration.zip)

thinkyhead, 2019-03-03:
It compiles fine for me. Be sure to use Arduino 1.8.8 or PlatformIO to build.
```
Sketch uses 173138 bytes (68%) of program storage space. Maximum is 253952 bytes.
Global variables use 5216 bytes (63%) of dynamic memory, leaving 2976 bytes for local variables. Maximum is 8192 bytes.
```

closed 2019-03-03 by thinkyhead