MarlinFirmware/Marlin #12588: [1.1.9] M605 S2; "move duplicate extruder into correct duplication position" handling seems flawed

emphasize, 2018-12-05:
After unparking in DXC_DUPLICATION_MODE I experience a slight movement of the duplication Extruder towards the bed (approx. 5/10mm) followed by a screeching sound, 'extruder_duplication_enabled' kicking in resulting the duplication extruder crashing the endstop. 
(no such experience with the other dual_x_carriage_modes)

Maybe a usefull sidenote. Commenting out

> if (!planner.buffer_line(
>               current_position[X_AXIS] + duplicate_extruder_x_offset,
>               current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_CART],
>               planner.max_feedrate_mm_s[X_AXIS], 1)
>              ) break;

resulting in no duplication extruder movement at all.

The positional infos from the serial communication (after M605 S2) seem accurate

> : echo:Hotend offsets: 0.00,0.00 100.00,0.00Recv: okSend: N13 G28*33
> 
>  --- G28
> 
> : Machine Type: CartesianRecv: Probe: NONE
> 
> _//  Second/Duplication Extruder_
> : For X axis:
> :  home_offset = -65.00Recv:  position_shift = 0.00Recv:  soft_endstop_min = 0.00
> :  soft_endstop_max = 217.50
> : echo:Active Extruder: 0
> :   current_position=(0.00, 0.00, 0.00) : setup_for_endstop_or_probe_move
> : > endstops.enable(true)
> : >>> homeaxis(X)
> : Home 1 Fast:
> : >>> do_homing_move(X, 387.00, [50.00])
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_position
> : <<< do_homing_move(X)
> : Move Away:
> : >>> do_homing_move(X, -7.00, [50.00])
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_position
> : <<< do_homing_move(X)
> : Home 2 Slow:
> : >>> do_homing_move(X, 14.00, 16.67)
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_position[...]
> : <<< do_homing_move(X)
> : >>> set_axis_is_at_home(X)
> : For X axis:
> :  home_offset = -65.00
> :  position_shift = 0.00
> :  soft_endstop_min = 63.50
> :  soft_endstop_max = 317.50
> :   current_position=(317.50, 0.00, 0.00) : sync_plan_position
> :   current_position=(317.50, 0.00, 0.00) : > AFTER set_axis_is_at_home
> : <<< homeaxis(X)
> 
> _//  First Extruder_
> : >>> homeaxis(X)
> : Home 1 Fast:
> : >>> do_homing_move(X, -387.00, [50.00])
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_positionRecv: <<< do_homing_move(X)
> : Move Away:
> : >>> do_homing_move(X, 7.00, [50.00])
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_positionRecv: <<< do_homing_move(X)
> : Home 2 Slow:
> : >>> do_homing_move(X, -14.00, 16.67)
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_positionRecv: <<< do_homing_move(X)
> : >>> set_axis_is_at_home(X)
> : For X axis:
> :  home_offset = -65.00
> :  position_shift = 0.00
> :  soft_endstop_min = 0.00
> :  soft_endstop_max = 217.50
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_position
> :   current_position=(0.00, 0.00, 0.00) : > AFTER set_axis_is_at_home
> : <<< homeaxis(X)
> 
> : >>> homeaxis(Y)
> .
> .
> .
> : <<< homeaxis(Y)
> 
> : >>> homeaxis(Z)
> .
> .
> .
> : <<< homeaxis(Z)
> 
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_position
> :   current_position=(0.00, 0.00, 0.00) : clean_up_after_endstop_or_probe_move
> : For X axis:
> :  home_offset = -65.00
> :  position_shift = 0.00
> :  soft_endstop_min = 0.00
> :  soft_endstop_max = 217.50
> : echo:Active Extruder: 0
> : X:-65.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> : <<< G28
> 
> Send: N14 M140 S65*99
> Send: N15 M104 T0 S215*19
> Send: N16 M109 T0 S215*29
> [...]
> Send: N17 M190 S65*109
> [...]
> Send: N18 G92 E0*126
> : X:-65.00 Y:0.00 Z:0.00 E:0.00 Count X:0 Y:0 Z:0
> : okSend: N19 M82*33
> : okSend: N20 M107*23
> : okSend: N21 M117 
> Printing...*40
> : okSend: N22 M1001*35
> : echo:Unknown command: "M1001" //Ideamaker specific
> Send: N23 M106 S0*86
> Send: N24 G1 F900 E-3.0000*20
> 
> : Set planner X317.50 ... Line to X100.00             **<--------------   small movement (approx. 10)  towards bed /screeching -> duplication -> endstop hit**     
> :   current_position=(0.00, 0.00, 0.00) : sync_plan_position    
> : Set extruder_duplication_enabled
> : Clear active_extruder_parked
> Send: N25 G0 F4800 X11.834 Y72.295*113

emphasize, 2018-12-07:
Another observation:

If I comment out SYNC_PLAN_POSITION_KINEMATIC();

`planner.set_position_mm(inactive_extruder_x_pos, current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_CART]);
            if (!planner.buffer_line(
              current_position[X_AXIS] + duplicate_extruder_x_offset,
              current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_CART],
              planner.max_feedrate_mm_s[X_AXIS], 1)
            ) break;
            //SYNC_PLAN_POSITION_KINEMATIC();
            planner.synchronize();`

the duplication extruder moves to the right position. But T0 moves as well (kissing the endstop excessively)

InsanityAutomation, 2018-12-07:
Handling of some of this needs rework, and has been in the queue between Roxy and myself. We have a sortve gameplan of what we intend to do, just neither of us have had much of any time to do so. The current menu entries need some revision as well. There are a few places with static values fixed to development machines. 

In my industry, the holidays are way overbooked (industrial controls) so it wont likely happen until afterwards.

Rough idea was an idex class with the helper functions and position calculations broken out, so the g code suite and menu's could call the positioning and mode change functions, also alleviating the need to be in S2 prior to S3 as that implementation gets finished. This would also feed back to the update softstops functions like tool change to set the limits appropriately for duplication modes.

emphasize, 2018-12-08:
Nice outlook for dual carriage, indeed.

@InsanityAutomation  Whoa, i think i got it. The screeching was the key indicator. After a couple dryruns with extended positional infos (no irregularities there; i thought it might be the planner.synchronize() ) i realized the planner.max_feedrate_mm_s in planner.buffer_line. Somehow this gap (300) to my ususal travel feedrate (80) jumped the planner.synchronize.

Sidenote: I used 300 max feedrate (X) 1.1.8 with positive results in duplication mode.

boelle, 2019-02-21:
@emphasize problem solved?

emphasize, 2019-02-22:
@boelle yeah, after tuning down the max feedrate the problem disappeared

boelle, 2019-02-22:
then maybe click close?

emphasize, 2019-02-22:
Sorry, missed that.

closed 2019-02-22 by emphasize