MarlinFirmware/Marlin #12589: Endstop Interrupts.

RudyBenner, 2018-12-05:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

RudyBenner, 2018-12-05:
Came across this file How do I utulize this.
I would like to test for interrupts in MKS_GEN_L board.
[pin_interrupt_test.zip](https://github.com/MarlinFirmware/Marlin/files/2649279/pin_interrupt_test.zip)

AnHardt, 2018-12-06:
`pin_interrupt_test.ino` is a stand alone Arduino sketch. It just prints a list of the interrupt capable pins the Arduino framework knows about. Ideally these are the pins what can interrupt the processor. A (integrated) patch for the Arduino framework was needed to make that true for the Arduino Mega.

A brief look into that file would have offered this insight.

RudyBenner, 2018-12-06:
I will have another look. Thanks for your help.

closed 2018-12-06 by RudyBenner