MarlinFirmware/Marlin #12591: Communication timeouts on LPC1769 (Azteeg X5 GT)

alexxy, 2018-12-05:
Hi! 

After recent update (2 days ago and recent commit 11c7945365bbe284ed2f863faf2eb18c7e6b92e9). I start getting random communication timeouts via usb. Octoprint log attached
[octoprint.log](https://github.com/MarlinFirmware/Marlin/files/2650008/octoprint.log)
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2650013/configs.zip)

gloomyandy, 2018-12-05:
As far as I can see none of the files changed in that commit should be being used on your setup. Could something else have changed (like Octoprint - there seem to be some strange errors in that log). Have you built the older version (that works) recently? It could be some change in the environment. If possible please rebuild the previous version and try it again leaving everything else the same.

alexxy, 2018-12-05:
I tryed both versions of Octo (release one 1.3.9 and rc 1.10rc3). Also there is nothing changed in env except Marlin version

gloomyandy, 2018-12-05:
Are you sure that there have been no updates to the environment used by platformio between the two tests? I still think it would be worth repeating the build of the older Marlin to confirm that the problem only occurs between Marlin builds.

alexxy, 2018-12-05:
Previous builds that had issues with EEPROM and serial doesnt had any timeouts...

gloomyandy, 2018-12-05:
Yes but have you rebuilt that version recently. Ideally you will build the old one it works, rebuild the new one it doesn't. At the moment we can't be sure if it is a Marlin change or something that may have changed in your build environment(which will auto update every time you run it). Once we have established that it is Marlin we can try and work out what change it was that caused the issue.

alexxy, 2018-12-06:
I changed baudrate to 250000 to see what happen

gloomyandy, 2018-12-06:
The baudrate setting has no real meaning when using serial over USB. So have we actually established if older versions of Marlin (built with the latest tools and framework) are any different to newer ones? We need to establish if the problem is in Marlin or the framework or tools. To do that you need to build the older version of Marlin using the new framework. Have you done that?

DyonisosHunter, 2018-12-06:
Hey,
I just updated to todays version of marlin and it is running fine till now. I am on OctoPrint too. Will let you know if there is anything strange happening.

gloomyandy, 2018-12-07:
OK thanks. I am using the same framework and have not seen timeouts using Repetier, but I will try a few more things. If you can reproduce this without using octoprint it would help a lot as I don't think any of the devs that work on this platform use octoprint.

@p3p any idea what may be going on here?

alexxy, 2018-12-11:
Any ideas? May it be related to timings?

gloomyandy, 2018-12-11:
No idea I've been using repetier to do various prints with no issues what so ever with an SBase board. But since I have neither the Azteeg board nor Octoprint there is not much more I can do. We have not seen any other reports of similar issues from other users on LPC176x hardware.

alexxy, 2018-12-11:
Most likely its a problem of LPC1769 which has different timings...

p3p, 2018-12-11:
Without an LPC1769 board I can't rule out a MCU specific problem, but the USB frequency is always 48Mhz (multiple of the 12MHz crystal all the boards have) so I doubt it's a timing issue in the USB peripheral, The MCU running at 120MHz should have no effect.

alexxy, 2018-12-12:
@p3p ok. but why i didnt see this issue before. however i running Marlin on this board for ~9 mounth?

alexxy, 2018-12-12:
I replaced cables... Still has issue

alexxy, 2018-12-12:
Btw may be this can help

```
Caught an exception, exiting:
Traceback (most recent call last):
  File "pronsole.py", line 54, in <module>
    interp.cmdloop()
  File "/home/orangepi/Printrun/printrun/pronsole.py", line 228, in cmdloop
    line = input(self.prompt)
UnicodeEncodeError: 'ascii' codec can't encode character '\xb0' in position 25: ordinal not in range(128)

```

This happens if i try communicate with printer when it has communication timeouts via Printrun

gloomyandy, 2018-12-12:
No idea perhaps you would be better asking this question on an octoprint forum? They may be able to explain possible causes.

boelle, 2019-02-21:
@alexxy problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-15 by p3p