MarlinFirmware/Marlin #12592: G29 P4 accumulates an error when used with active UBL

volca02, 2018-12-05:
### Description

In branch bugfix-2.0.x, when using G29 P4 to manually fix the bed leveling matrix for a particular point, repeating this command will accumulate what seems to be the set value as Z displacement - every repetition of this command will move the nozzle by seemingly the correction distance from there further, until G28 corrects it again.

### Steps to Reproduce

1. Autolevel `G28`
2. Switch UBL on with `G29 A`
3. Do a fine tuning command for specified point `G29 P4 X110 Y110`
4. Do a fine tuning command for specified point `G29 P4 X110 Y110`

**Expected behavior:**
I expect the nozzle to end up in the same place for fine tuning every time this command happens.

**Actual behavior:**
In my setup, the nozzle gets down with every repetition of the command. Step 3 has the nozzle 1mm up above surface of the heatbed. Step 4 has nozzle crashing into the surface.

The behaviour of the `G29 P4` changes, if I do `G29 D` first. Then the accumulation of Z does not happen and `G29 P4` ends up in correct height every repetition.

#### Additional Information

This is my current automatically populated UBL matrix:

```
0.578⇥	0.578⇥	0.578⇥	0.607⇥	0.648⇥	0.614⇥	0.539⇥	0.479⇥	0.386⇥	0.386␊
0.578⇥	0.578⇥	0.578⇥	0.607⇥	0.648⇥	0.614⇥	0.539⇥	0.479⇥	0.386⇥	0.386␊
0.689⇥	0.689⇥	0.689⇥	0.747⇥	0.808⇥	0.778⇥	0.714⇥	0.600⇥	0.416⇥	0.416␊
0.798⇥	0.798⇥	0.798⇥	0.941⇥	1.055⇥	0.967⇥	0.895⇥	0.817⇥	0.618⇥	0.618␊
0.809⇥	0.809⇥	0.809⇥	0.975⇥	1.042⇥	0.973⇥	0.980⇥	0.866⇥	0.684⇥	0.684␊
0.781⇥	0.781⇥	0.781⇥	0.891⇥	0.910⇥	0.869⇥	0.938⇥	0.851⇥	0.651⇥	0.651␊
0.810⇥	0.810⇥	0.810⇥	0.888⇥	0.846⇥	0.785⇥	0.888⇥	0.792⇥	0.598⇥	0.598␊
0.788⇥	0.788⇥	0.788⇥	0.800⇥	0.706⇥	0.668⇥	0.793⇥	0.734⇥	0.554⇥	0.554␊
0.728⇥	0.719⇥	0.710⇥	0.701⇥	0.632⇥	0.603⇥	0.692⇥	0.602⇥	0.479⇥	0.479␊
0.728⇥	0.719⇥	0.710⇥	0.701⇥	0.632⇥	0.603⇥	0.692⇥	0.602⇥	0.479⇥	0.479␊
```

This is my current configuration.
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2650079/configuration.zip)

gloomyandy, 2018-12-05:
That mesh seems a little odd. I would have expected the reading close to the home point to be much closer to zero and then for the other values to be +/- around it. But all of your values are +ve which seems a little strange. Have you adjusted the values at all after the G29 P1?

I notice you do not have Z_SAFE_HOMING set is there a reason for this? The comments for bed levelling using a probe recommend that you use this option.

volca02, 2018-12-05:
That mesh is what the `G92 P1` command did with UBL enabled as startup default. I tried once again with `G92 D` and it came out differently:

```
0.013⇥	0.013⇥	0.013⇥	0.049⇥	0.106⇥	0.095⇥	0.049⇥	-0.008⇥	-0.097⇥	-0.097␊
0.013⇥	0.013⇥	0.013⇥	0.049⇥	0.106⇥	0.095⇥	0.049⇥	-0.008⇥	-0.097⇥	-0.097␊
0.113⇥	0.113⇥	0.113⇥	0.173⇥	0.242⇥	0.237⇥	0.201⇥	0.116⇥	-0.068⇥	-0.068␊
0.222⇥	0.222⇥	0.222⇥	0.342⇥	0.453⇥	0.414⇥	0.390⇥	0.319⇥	0.126⇥	0.126␊
0.237⇥	0.237⇥	0.237⇥	0.386⇥	0.453⇥	0.415⇥	0.473⇥	0.368⇥	0.205⇥	0.205␊
0.212⇥	0.212⇥	0.212⇥	0.319⇥	0.351⇥	0.334⇥	0.443⇥	0.364⇥	0.167⇥	0.167␊
0.242⇥	0.242⇥	0.242⇥	0.325⇥	0.305⇥	0.260⇥	0.373⇥	0.303⇥	0.104⇥	0.104␊
0.230⇥	0.231⇥	0.230⇥	0.254⇥	0.190⇥	0.151⇥	0.284⇥	0.222⇥	0.060⇥	0.060␊
0.151⇥	0.151⇥	0.151⇥	0.159⇥	0.110⇥	0.082⇥	0.183⇥	0.110⇥	0.001⇥	0.001␊
0.150⇥	0.151⇥	0.151⇥	0.159⇥	0.110⇥	0.082⇥	0.183⇥	0.110⇥	0.001⇥	0.001␊
```

This leads me to believe that there already was a starting offset for Z axis when the UBL process began - even though I did autohome right before that. Not sure why that happened though, maybe the cause for this is related.

I do not have Z_SAFE_HOMING set because auto homing in the default XY coordinates works fine - the probe is nicely above the metalic surface (using the hall effect probe).

gloomyandy, 2018-12-05:
I think you would probably get a better mesh if you home in the centre of the bed. The edges of the bed are more likely to have issues with heat distribution and warping/movement when heated. Up to you but I suspect that the recommendation in the comments is there for a good reason.

volca02, 2018-12-05:
Thanks for the tip - I might try that, although I plan on replacing the crappy Y carriage with a aluminum one as soon as I find the time. 

This however does nothing to do with the reported bug - that enabling UBL screws with `G92 P4` making that command impossible to use.

gloomyandy, 2018-12-05:
True but it helps if you have a good mesh to begin with and your original mesh looked very wrong so I wanted to be sure that you didn't have other problems. 

But back to your original problem. If you have levelling turned off (G29 D) are you actually able to edit the mesh, or do you need it on to be able to use the editing feature? If after editing the mesh you do a G29 T does the edited value seem to be correct?

volca02, 2018-12-05:
Sure, I'm able to edit the mesh with UBL disabled - it works without accumulating the Z error. With UBL enabled, the Z error influences the outcome - the printhead is not returning to a neutral position, so further `G29 P4` accumulate that Z error and are useless - I had to auto home a lot to get a repeatable initial position when doing that command - I tried to check if it that command is repeatable, which is how I found this bug existed for me.

gloomyandy, 2018-12-05:
OK so is there any reason why you would want to have UBL enabled during the editing process or would simply disabling it when editing be a reasonable fix?

gloomyandy, 2018-12-06:
@Roxy-3D any thoughts on this?

Roxy-3D, 2018-12-06:
Did you start generation of the mesh from the center of the bed?   Or, did you start from the homing position in the lower left?       In general, it usually is best to home to the center of the bed.

volca02, 2018-12-06:
The mesh generation started automatically from the center of the bed. The second mesh I generated with `G29 P1` with disabled UBL after homing is correct (aside from the automatically completed parts on the edges that will need fine tuning) - it prints well after activating UBL.

My problem is with the fine tuning operation - `G29 P4` - which is acting wrong with activated UBL (`G29 A`), not the quality of the mesh (which seems to be correct).

Roxy-3D, 2018-12-09:
I don't have a machine available right now.   It will be several weeks before I do.   Can you try a couple of quick experiments?    First, lets do some basics:
- Do a G29 P0 to zero the mesh.   Then (with the nozzle in the center of the bed) see if you can edit several mesh points to .15mm and -.15 mm.    Then verify (with G29 T) they really did get changed.   And then with UBL Enabled (G29 A) and a Z height of .2, verify the nozzle tracks those mesh points as you move the nozzle around the center of the bed.

boelle, 2019-02-21:
@volca02 problem solved?

volca02, 2019-02-22:
Hi, sorry I totally forgot about this bug - I replaced the Y carriage on my system and was able to level the bed without resorting to manual re-probing. I'll try to find time to perform the specified test anyway.

boelle, 2019-02-22:
maybe  click close and open new issue in the future if problem is still there?

volca02, 2019-02-22:
Well the bug is most probably still there, somebody else might run into it when fine-tuning the bed leveling.

boelle, 2019-02-22:
then they can open new issue i think

boelle, 2019-03-06:
@thinkyhead i think we can close this one

volca02, 2019-03-07:
For the sake of manageability of the issue counts I'll close this one. `G29 P4` is a marginal feature not many will resort to so hopefully this trap is not something many will fall into. Can't find the time to re-test this anyway.

closed 2019-03-07 by volca02