MarlinFirmware/Marlin #12594: [bugfix-2.0.x] Build errors after recent commit

nemphys, 2018-12-05:
After commit aaf862aef22caa136f4aa63d9fcc4e71c164ae33, I get the following error when trying to  build with NUM_SERVOS 1 (for a custom blower fan I have, not for a Z-probe):

/src/module/configuration_store.cpp: In static member function 'static bool MarlinSettings::save()':
/src/module/configuration_store.cpp:640:20: error: 'servo_angles' was not declared in this scope
       EEPROM_WRITE(servo_angles);
                                             ^

This can be fixed by changing '#if !HAS_SERVOS' to '#if !ENABLED(EDITABLE_SERVO_ANGLES)' (which makes sense after the commit's changes), but even after doing so, I get EEPROM CRC errors:
'EEPROM CRC mismatch - (stored) 53170 != 58028 (calculated)!'

Trying a M502 & M500, I get:
'echo:Settings Stored (586 bytes; crc 53170)
Error:EEPROM datasize error.'

I suppose something broke with the commit.

Ludy87, 2018-12-06:
no configuration file, no help

nemphys, 2018-12-07:
Here you go:
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2657378/Configuration.zip)

thinkyhead, 2018-12-08:
The issue is now patched.

commit e8bda62d (2018-12-08): Followup to servo angles patch

Fix #12594
closed 2018-12-08 by thinkyhead