MarlinFirmware/Marlin #12596: BLTouch Not Level After Mesh Generation

amouzes, 2018-12-06:
Hi. I have an Ender 3 + BLTouch + Octopi. Using Cura 3.5.1 print settings. 

-Issue = Print initial layer isn't level after BLTouch does its mesh generation.

-What's happening = I leveled the bed using the paper trick to get it close. Then before my prints, G28 (Home All Axis), then G29 (BLTouch Mesh Generation), after G29 any modifications before hand to make the bed more level on each corner is UNDONE. For some reason the middle is good and level, but the corners go back to being slightly off.

-What I think should happen = After getting each corner in a good spot, shouldn't the BLTouch adjust to each corners dimensions (including the center) and make it level?

Like I said before, it seems like the middle is good but my corners are to far away from the nozzle. How can I go about fixing this to have each corner equal the middle so that everything is level? Any way to tell the BLTouch that the middles is to high/corners to low?

Hope that all made sense! Thanks for the help! Config H and Adv attached. Regular marlin file and word pad just in case.

[Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/2652289/Configurations.zip)

Using Cura 3.5.1 - Here is my G-code just in case.
G28 ;Home
G29; BLTOUCH mesh generation
M104 S{material_print_temperature_layer_0} ; set extruder temp
M140 S{material_bed_temperature_layer_0} ; set bed temp
M190 S{material_bed_temperature_layer_0} ; wait for bed temp
M109 S{material_print_temperature_layer_0} ; wait for extruder temp
G92 E0; Reset the Extruder's Origin
M117 Printing @ 1MM.

gloomyandy, 2018-12-06:
I can't download your configuration files right now. It would probably help if you can tell us what version of Marlin you are using and which levelling system you have configured.

GMagician, 2018-12-06:
Don't know anything about your cnf but have you tried with M420 S1 after G29?

amouzes, 2018-12-06:
@gloomyandy Hey sorry about that. I'm using Marlin 1.1.9 and for leveling system configure what do you mean? As in Bi-linear? I'm not super knowledgeable on this. 

@GMagician I have not tried that, what exactly will that do?

GMagician, 2018-12-06:
http://marlinfw.org/docs/gcode/M420.html

amouzes, 2018-12-06:
@GMagician I added it in. Doesn't seem to change anything.

amouzes, 2018-12-06:
@GMagician Where do I get the log? I'm just somewhat knowledgeable on this.

GMagician, 2018-12-06:
Don't you use repetier os something like that?

amouzes, 2018-12-06:
I'm assuming that Marlin is, but not entirely sure where to get the logs.

gloomyandy, 2018-12-06:
Yes I was asking about the levelling mode I've now been able to look at your config and see you are using bilinear. One problem you may have is that due to the relatively large probe offset you will not be able to probe that close to (at least some of) the edge of your bed. A few things you could try....
1. Increase the number of points to probe.
2. Try defining EXTRAPOLATE_BEYOND_GRID

amouzes, 2018-12-07:
@gloomyandy Neither of those worked. Defining the Extrapolate_beyond_grid doesn't seem to work either.  It just messes with my BLTouch and doesn't work properly.

amouzes, 2018-12-07:
It just seems like the Bilinear Level Grid is not being kept. It goes around, generates the mesh, and nothing is saved. Really sad and annoying to figure out..

gloomyandy, 2018-12-07:
What do you mean by "it just messes with my BLTouch" setting that define should make no difference during the probing, it should just change the way the grid is used during printing. 

Oh and one other thing you should probably be doing the G29 after the bed has been heated as the shape of the bed will change when hot.

amouzes, 2018-12-07:
Well for now I'm just going to do a bit of redoing everything. I might double check my Ender 3 build as a whole to make sure nothing got tweaked. And then redo my process of install of the BLTouch. Shall go from there. Because I've gone through a lot to figure this out.

thinkyhead, 2018-12-09:
After `G29` you can insert an `M420 V` to see the mesh that was recorded and the current state of bed leveling.

boelle, 2019-02-21:
@amouzes problem fixed?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

tomdk, 2019-03-15:
I took the time reading through this having the same issue just to find that OP abandoned this?

boelle, 2019-03-15:
@tomdk yep that happens a lot. sometimes people fix their issues and dont close issue and tell what they did

amouzes, 2019-03-15:
Sorry for the late reply. But I asked my previous question, that I needed help with, back in November/December I believe. So not to be rude but you guys took quite a bit to get back to me. I solved my problem by: rebuilding my Ender 3 and making sure everything was perfectly level, and then re-installing the auto mesh generation. After a while I just took off the BLTouch and went back to the basic Ender3 stats. My machine works fine without it now.

boelle, 2019-03-15:
you wrote on december 6th and you got a reply same day and the day after.. then everything was silent and i just stired the pot to see if something would happen

since you have solved it would you mind click the close button below? only you and admins can see it

closed 2019-05-16 by thinkyhead