MarlinFirmware/Marlin #12599: M851 not working on 1st print

Ivomola, 2018-12-07:
Hi. This is my first time writting a bug report as the code of conduct says, because it's the first time I write one from the computer lmao. Ok now in all seriousness, I have a Tevo Tarantula with the latest and gratest version of Marlin (1.1.9), and I have in my start gcode an M851 command to set the Z probe offset because in Cura because -2,70mm is too low and it won't let me put it, but a problem I noticed quite recently is that, when I print something for the first time after unplugging the printer, the offset doesn't work, I have to cancel the print and start it again for it to low the nozzle at the proper height. I tried to modify the start gcode to home Z twice, but that didn't solve anything. Below is the bug report.

Steps to reproduce the bug:
1. Unplug the printer
2. Plug it
3. Start a print

**Expected behavior:** The printer lowers the nozzle at the proper height

**Actual behavior:** The nozzle doesn't obey the offset and it stays way high until I cancel the print and start it over.

P.S.: @thinkyhead, I watched your interview at YouTube in Toms3D channel, and I must say, you're amazing. Thanks for everything.
[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/2658477/Config.files.zip)

RudyBenner, 2018-12-08:
Looking here:

// @section extras

//
// EEPROM
//
// The microcontroller can store settings in the EEPROM, e.g. max velocity...
// M500 - stores parameters in EEPROM
// M501 - reads parameters from EEPROM (if you need reset them after you changed them temporarily).
// M502 - reverts to the default "factory settings".  You still need to store them in EEPROM afterwards if you want to.
//
#define EEPROM_SETTINGS // Enable for M500 and M501 commands
#define DISABLE_M503    // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT   // Give feedback on EEPROM commands. Disable to save PROGMEM.

When you do a M501, it should echo the eeprom settings, including current M851. 
After flashing, common practice is M502 followed by M500.
Give it a try.
rudy

RudyBenner, 2018-12-08:
Baud set at 1000000?

Ivomola, 2018-12-08:
Thanks for your reply. First: I already tried that, but the point of not putting that in the start gcode is to not wear out the eeprom, as it has a limited number of writes, and to do that what I do is not save things to the eeprom, and that's where the problem starts, because if at the second print the offset works without saving it, why can't it work at the first one?
And second: Yes, I have the baud at 1000000 because I have octoprint in the first raspberry pi and printing through usb is painfully slow, so I print from the SD card, and to speed up transfers I maxed out the baud rate. I have a shielded cable so no problems there.

gloomyandy, 2018-12-08:
No idea about the M851 issue but unless you are changing your Z probe offset for each print (and if you are why are you doing that?), then why not simply configure it in your firmware, save that setting once to eeprom (if you are using eeprom) and have done with it? 

eeproms do have a limited number of write cycles, but it is a very high limit. The Arduino in your printer has a lifetime of at least 100000 cycles (and this is almost certainly a very conservative estimate). So even if with every print you write to the eeprom and do say 10 prints a day and print every day of the year you will not run out of eeprom lifetime for 27 years.

Ivomola, 2018-12-08:
Wow. That's a very, very good answer. I will still report this as a possible bug, but as for my problem with the offset, now I know what to do. Again, thanks for your answer, great to know people like you are supporting marlin.

closed 2018-12-08 by Ivomola

AnHardt, 2018-12-08:
The position of `M851` matters. It has to be before homing to be active then.