MarlinFirmware/Marlin #12600: [1.1.9] Extruder no longer works

Mattelec, 2018-12-07:
### Description

This might be a rare error, but I reinstalled Marlin 1.1.9 cause I was playing with the speeds and accelerations to improve the printer and so far I achieved some decents speeds in the X and Y axis but the extruder no longer works.

I'm using TMC2130 in SPI mode in X&Y and DRV8825 in Z&E. The steps are the same as I was using before, the current of the DRV is the same. No matter if I change the values to stock Marlin, the extruder won't move.

If I use the previous version, 1.1.8, the extruder works but I have to change the steps of the X axis (in 1.1.9 I don't have to)

**Expected behavior:** Extruder rotates so it can print

**Actual behavior:** X and Y axis moves but extruder no longer move or work

#### Additional Information

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2658581/Configuration.zip)

J-charles-C, 2018-12-08:
I,

Before move,
Have you activate hotend over low temp to PREVENT_COLD_EXTRUSION ?

best

thinkyhead, 2018-12-09:
Are you remembering to do `M502`, `M500` after flashing the firmware?

Mattelec, 2018-12-11:
Solved by installing a new driver, which solved the issue and it's running so far

closed 2018-12-11 by Mattelec