MarlinFirmware/Marlin #12601: [Question] Weird bed heating lock

GMagician, 2018-12-07:
Hi, this evening something a bit weird occured to my printer.

Note that installed firmware has been already used yesterday with no issues with same temperatures.
I'm wondering if this has been already occured to someone else and/or if someone has an idea of why this occured to.

Problem:
I turned on bed heating to 90° to print with hips I got thermal error and I thought it was due to slow heat curve. Disconnected printer (via host-repetier button and reconnected) same problem again.
After 3/4 times with the same issue, changing temperature to lower one, I checked temperature graph...temperature was increasing correctly (i also thought about bad thermistore reading).
After a few of tryes I noticed that my temp setpoint was 80° and bed was more thant 85°....What the hell!!! I checked bed led and it was fixed on (I used pid). Ok I thought broken mosfet so I tryed to turn off bed via host-repetier and.....led turned off....returned on bed heating and temperature was increasing and pid was not working again....

I solved removing power from printer and giving power again...now bed pid is working has expected...

Note: in the same time hotend was heating correctly (pid was working fine)

thinkyhead, 2018-12-09:
What version of Marlin? Do you find that the bed heats more slowly as it approaches 80-90°? Is there a lot of overshoot during heating? Does the temperature fall far below the hysteresis after the overshoot? If a heater is stuck on, even after a restart, I would begin to suspect a bad MOSFET.

GMagician, 2018-12-09:
> What version of Marlin?

I'm using latest 2.0.x 

> Do you find that the bed heats more slowly as it approaches 80-90°

yes I see different ramps after 50°, after 70 but issue is not temperature not increasing but not stopping to increase. Last test I was doing was setpoint 80° and bed already at 80°. To keep it at 80 I had to turn on and off manually from repetier otherwise, leaving control to firmware temp reached 86°

> I would begin to suspect a bad MOSFET

Mosfet is not an issue. When I turned bed off using repetier, ramps led turned off and temp begun to lowers. Turning it on again with repetier, led again on and temp raised. When I saw issued led was fixed on that mean firmare kept it active

Even if I don't understand why, after a system halt, and soft reset, issue was still present and after a full reset it was gone

Now I heated again to 90° and everything gone ok..here are 2 screenshot for full heating and zoomed when just heatet (you can see on left a little overshoot before stabilizatio
[image]
[image]

n)

closed 2018-12-14 by GMagician