MarlinFirmware/Marlin #12602: Delta stuttering on bugfix 1.1.9

wookie666, 2018-12-08:
### Description

Delta firmware on 1.1.9 bugfix stutters

### Steps to Reproduce

I have updated my anycubic delta from 1.1.0 bugfix to 1.1.9 bugfix and im now getting stuttering on fast moves.
I have tried to reduce the S value in M665 but it still happens. Its not as bad with a lower s value like 50, but it still stutters.

Im getting this behaviour on anything higher than say F5000 travel moves.

printing its fine. albeit a little slower than before.

I have bed leveling enabled.
Any ideas? or is my board too slow now with all of marlins updates.

swilkens, 2018-12-08:
Are you perhaps printing over USB? (from laptop, pc, octoprint or something similar) If so it might be that your command buffer runs empty, if are able to print from SD the issue might go away.

Have you tried bugfix-2.0.x ?

Please also provide your configuration files and the gcode file that shows the issue if you can, it will help finding your problem.

wookie666, 2018-12-08:
Nah it's Sd card. It's an issue with the buffer I think nredu ung the
segments to 30 and max travel at 5000 it's almost gone. But I wish I could
run faster than that

thinkyhead, 2018-12-08:
Planner starvation is caused by having too many segments per second, or features introducing extra delay so that the planner gets emptied out before new movements are able to be added. All you can do is try tuning things, turning off features, etc., until you find some combination of settings where planner starvation is eliminated.

closed 2018-12-08 by thinkyhead

macropin, 2019-09-05:
@wookie666 Did you figure out which features are causing the starvation?