MarlinFirmware/Marlin #12607: Microstepping are Not working Reprap

tarekmesto, 2018-12-09:
I have reprep 1.4 with drv8825, but in marlin firmware when I change the step per unit to 80 step/mm the software did not consider the mmicrostepping 1/32. when I define it as 
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 5, 5, 160, 5 } the printer works fine, without microstepping ??
how to solve this??

thinkyhead, 2018-12-09:
Did you do `M502`, `M500` after flashing the new firmware?

Roxy-3D, 2018-12-09:
> I have reprep 1.4 with drv8825, but in marlin firmware

There have been a lot of  bug fixes and improvements...    Can you load the current firmware and try your scenario out there?    We would love to discuss this problem in that context.   OK?

?

boelle, 2019-02-21:
@tarekmesto problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead