MarlinFirmware/Marlin #12613: [2.0.x] Linker error: Visual Micro for Visual Studio

j54n1n, 2018-12-09:
### Description

I am working on the firmware for the PowerWASP 3D printer that uses an ATmega2560 on its mainboard.
While using a [slightly fixed version](https://www.visualmicro.com/forums/YaBB.pl?num=1543777515/0#16) of the [Arduino IDE plugin for Visual Studio](https://www.visualmicro.com/) I get a linker error of multiple definitions if I do not rename **Marlin.cpp** from within the *src* folder.

### Steps to Reproduce

1. Download and install [Visual Studio Community 2017](https://visualstudio.microsoft.com/vs/community/) for C++ development
2. Install the [bugfix version 1812.9 of the Visual Micro plugin](https://www.visualmicro.com/forums/YaBB.pl?num=1543777515/0#16)
3. Setup from within Visual Studio following options:
  a. Menu bar: Tools > Options > Visual Micro
   - Build In Project Intermediate: False
   - Build Uses Configuration: True
  b. Menu bar: vMicro > Compiler
   - Deep Search for Libraries: Unselected
   - Verbose: Selected
4. Import Arduino Project (Marlin.ino) from the menu bar: File > Open > Arduino Project
5. Build Solution "Marlin" (F7)

**Expected behavior:**
That a fresh download of the [Marlin-bugfix-2.0.x](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) firmware compiles without renaming the Marlin.cpp file.
With a renamed Marlin.cpp I get:
```
Linking it all together ...
# Coping cached core C:\Users\Julian\AppData\Local\Temp\VMBCore\arduino16x\161e5c2bdd7f9f17435e76bd8c26e3f2\core.a to C:\Users\Julian\AppData\Local\Temp\VMBuilds\Marlin\mega_atmega2560\Release\core.a

"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560 -o "C:\Users\Julian\AppData\Local\Temp\VMBuilds\Marlin\mega_atmega2560\Release/Marlin.ino.elf" "Marlin.cpp.o" "src\core\serial.cpp.o" "src\core\utility.cpp.o" "src\feature\baricuda.cpp.o" "src\feature\caselight.cpp.o" "src\feature\closedloop.cpp.o" "src\feature\controllerfan.cpp.o" "src\feature\emergency_parser.cpp.o" "src\feature\fanmux.cpp.o" "src\feature\filwidth.cpp.o" "src\feature\fwretract.cpp.o" "src\feature\I2CPositionEncoder.cpp.o" "src\feature\Max7219_Debug_LEDs.cpp.o" "src\feature\mixing.cpp.o" "src\feature\pause.cpp.o" "src\feature\power.cpp.o" "src\feature\power_loss_recovery.cpp.o" "src\feature\runout.cpp.o" "src\feature\snmm.cpp.o" "src\feature\solenoid.cpp.o" "src\feature\tmc_util.cpp.o" "src\feature\twibus.cpp.o" "src\feature\bedlevel\abl\abl.cpp.o" "src\feature\bedlevel\bedlevel.cpp.o" "src\feature\bedlevel\mbl\mesh_bed_leveling.cpp.o" "src\feature\bedlevel\ubl\ubl.cpp.o" "src\feature\bedlevel\ubl\ubl_G29.cpp.o" "src\feature\bedlevel\ubl\ubl_motion.cpp.o" "src\feature\dac\dac_dac084s085.cpp.o" "src\feature\dac\dac_mcp4728.cpp.o" "src\feature\dac\stepper_dac.cpp.o" "src\feature\digipot\digipot_mcp4018.cpp.o" "src\feature\digipot\digipot_mcp4451.cpp.o" "src\feature\leds\blinkm.cpp.o" "src\feature\leds\leds.cpp.o" "src\feature\leds\neopixel.cpp.o" "src\feature\leds\pca9632.cpp.o" "src\feature\leds\printer_event_leds.cpp.o" "src\feature\leds\tempstat.cpp.o" "src\gcode\bedlevel\abl\G29.cpp.o" "src\gcode\bedlevel\abl\M421.cpp.o" "src\gcode\bedlevel\G26.cpp.o" "src\gcode\bedlevel\G42.cpp.o" "src\gcode\bedlevel\M420.cpp.o" "src\gcode\bedlevel\mbl\G29.cpp.o" "src\gcode\bedlevel\mbl\M421.cpp.o" "src\gcode\bedlevel\ubl\G29.cpp.o" "src\gcode\bedlevel\ubl\M421.cpp.o" "src\gcode\bedlevel\ubl\M49.cpp.o" "src\gcode\calibrate\G28.cpp.o" "src\gcode\calibrate\G33.cpp.o" "src\gcode\calibrate\G34_M422.cpp.o" "src\gcode\calibrate\M100.cpp.o" "src\gcode\calibrate\M12.cpp.o" "src\gcode\calibrate\M48.cpp.o" "src\gcode\calibrate\M665.cpp.o" "src\gcode\calibrate\M666.cpp.o" "src\gcode\calibrate\M852.cpp.o" "src\gcode\config\M200-M205.cpp.o" "src\gcode\config\M217.cpp.o" "src\gcode\config\M218.cpp.o" "src\gcode\config\M220.cpp.o" "src\gcode\config\M221.cpp.o" "src\gcode\config\M281.cpp.o" "src\gcode\config\M301.cpp.o" "src\gcode\config\M302.cpp.o" "src\gcode\config\M304.cpp.o" "src\gcode\config\M43.cpp.o" "src\gcode\config\M540.cpp.o" "src\gcode\config\M92.cpp.o" "src\gcode\control\M108_M112_M410.cpp.o" "src\gcode\control\M111.cpp.o" "src\gcode\control\M120_M121.cpp.o" "src\gcode\control\M17_M18_M84.cpp.o" "src\gcode\control\M211.cpp.o" "src\gcode\control\M226.cpp.o" "src\gcode\control\M280.cpp.o" "src\gcode\control\M350_M351.cpp.o" "src\gcode\control\M380_M381.cpp.o" "src\gcode\control\M3-M5.cpp.o" "src\gcode\control\M400.cpp.o" "src\gcode\control\M42.cpp.o" "src\gcode\control\M605.cpp.o" "src\gcode\control\M80_M81.cpp.o" "src\gcode\control\M85.cpp.o" "src\gcode\control\M999.cpp.o" "src\gcode\control\T.cpp.o" "src\gcode\eeprom\M500-M504.cpp.o" "src\gcode\feature\advance\M900.cpp.o" "src\gcode\feature\baricuda\M126-M129.cpp.o" "src\gcode\feature\camera\M240.cpp.o" "src\gcode\feature\caselight\M355.cpp.o" "src\gcode\feature\clean\G12.cpp.o" "src\gcode\feature\digipot\M907-M910.cpp.o" "src\gcode\feature\filwidth\M404-M407.cpp.o" "src\gcode\feature\fwretract\G10_G11.cpp.o" "src\gcode\feature\fwretract\M207-M209.cpp.o" "src\gcode\feature\i2c\M260_M261.cpp.o" "src\gcode\feature\leds\M150.cpp.o" "src\gcode\feature\leds\M7219.cpp.o" "src\gcode\feature\macro\M810-M819.cpp.o" "src\gcode\feature\mixing\M163-M165.cpp.o" "src\gcode\feature\pause\G27.cpp.o" "src\gcode\feature\pause\M125.cpp.o" "src\gcode\feature\pause\M600.cpp.o" "src\gcode\feature\pause\M603.cpp.o" "src\gcode\feature\pause\M701_M702.cpp.o" "src\gcode\feature\powerloss\M1000.cpp.o" "src\gcode\feature\powerloss\M413.cpp.o" "src\gcode\feature\runout\M412.cpp.o" "src\gcode\feature\trinamic\M122.cpp.o" "src\gcode\feature\trinamic\M906.cpp.o" "src\gcode\feature\trinamic\M911-M915.cpp.o" "src\gcode\gcode.cpp.o" "src\gcode\parser.cpp.o" "src\gcode\queue.cpp.o" "src\gcode\geometry\G17-G19.cpp.o" "src\gcode\geometry\G53-G59.cpp.o" "src\gcode\geometry\G92.cpp.o" "src\gcode\geometry\M206_M428.cpp.o" "src\gcode\host\M110.cpp.o" "src\gcode\host\M113.cpp.o" "src\gcode\host\M114.cpp.o" "src\gcode\host\M115.cpp.o" "src\gcode\host\M118.cpp.o" "src\gcode\host\M119.cpp.o" "src\gcode\lcd\M0_M1.cpp.o" "src\gcode\lcd\M117.cpp.o" "src\gcode\lcd\M145.cpp.o" "src\gcode\lcd\M250.cpp.o" "src\gcode\lcd\M300.cpp.o" "src\gcode\lcd\M73.cpp.o" "src\gcode\motion\G0_G1.cpp.o" "src\gcode\motion\G2_G3.cpp.o" "src\gcode\motion\G4.cpp.o" "src\gcode\motion\G5.cpp.o" "src\gcode\motion\G80.cpp.o" "src\gcode\motion\M290.cpp.o" "src\gcode\probe\G30.cpp.o" "src\gcode\probe\G31_G32.cpp.o" "src\gcode\probe\G38.cpp.o" "src\gcode\probe\M401_M402.cpp.o" "src\gcode\probe\M851.cpp.o" "src\gcode\scara\M360-M364.cpp.o" "src\gcode\sdcard\M20-M30_M32-M34_M524_M928.cpp.o" "src\gcode\stats\M31.cpp.o" "src\gcode\stats\M75-M78.cpp.o" "src\gcode\temperature\M104_M109.cpp.o" "src\gcode\temperature\M105.cpp.o" "src\gcode\temperature\M106_M107.cpp.o" "src\gcode\temperature\M140_M190.cpp.o" "src\gcode\temperature\M155.cpp.o" "src\gcode\temperature\M303.cpp.o" "src\gcode\units\G20_G21.cpp.o" "src\gcode\units\M149.cpp.o" "src\gcode\units\M82_M83.cpp.o" "src\HAL\HAL_AVR\HAL.cpp.o" "src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o" "src\HAL\HAL_AVR\MarlinSerial.cpp.o" "src\HAL\HAL_AVR\persistent_store_eeprom.cpp.o" "src\HAL\HAL_AVR\servo_AVR.cpp.o" "src\HAL\HAL_AVR\watchdog_AVR.cpp.o" "src\HAL\HAL_DUE\DebugMonitor_Due.cpp.o" "src\HAL\HAL_DUE\EepromEmulation_Due.cpp.o" "src\HAL\HAL_DUE\HAL.cpp.o" "src\HAL\HAL_DUE\HAL_spi_Due.cpp.o" "src\HAL\HAL_DUE\HAL_timers_Due.cpp.o" "src\HAL\HAL_DUE\InterruptVectors_Due.cpp.o" "src\HAL\HAL_DUE\MarlinSerial_Due.cpp.o" "src\HAL\HAL_DUE\MarlinSerialUSB_Due.cpp.o" "src\HAL\HAL_DUE\persistent_store_eeprom.cpp.o" "src\HAL\HAL_DUE\Servo_Due.cpp.o" "src\HAL\HAL_DUE\Tone.cpp.o" "src\HAL\HAL_DUE\u8g_com_HAL_DUE_shared_hw_spi.cpp.o" "src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o" "src\HAL\HAL_DUE\watchdog_Due.cpp.o" "src\HAL\HAL_DUE\usb\ctrl_access.c.o" "src\HAL\HAL_DUE\usb\sysclk.c.o" "src\HAL\HAL_DUE\usb\udc.c.o" "src\HAL\HAL_DUE\usb\udi_cdc.c.o" "src\HAL\HAL_DUE\usb\udi_cdc_desc.c.o" "src\HAL\HAL_DUE\usb\udi_composite_desc.c.o" "src\HAL\HAL_DUE\usb\udi_msc.c.o" "src\HAL\HAL_DUE\usb\uotghs_device_due.c.o" "src\HAL\HAL_DUE\usb\usb_task.c.o" "src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o" "src\HAL\HAL_ESP32\HAL.cpp.o" "src\HAL\HAL_ESP32\HAL_spi_ESP32.cpp.o" "src\HAL\HAL_ESP32\HAL_timers_ESP32.cpp.o" "src\HAL\HAL_ESP32\ota.cpp.o" "src\HAL\HAL_ESP32\watchdog_ESP32.cpp.o" "src\HAL\HAL_LPC1768\DebugMonitor_LPC1768.cpp.o" "src\HAL\HAL_LPC1768\HAL.cpp.o" "src\HAL\HAL_LPC1768\HAL_spi.cpp.o" "src\HAL\HAL_LPC [... cut ...]
main.cpp:43: undefined reference to setup
main.cpp:46: undefined reference to loop
 
collect2.exe*: error: ld returned 1 exit status
```

#### Additional Information

* This happens even without any file changed with a fresh copy from Github.
* A working solution is to *rename* the **Marlin.cpp** file to **MarlinCore.cpp** and reimport the sketch as it is suggested in this forum post: https://www.visualmicro.com/forums/YaBB.pl?num=1543777515/1#1

Could this change be included in the Marlin-bugfix-2.0.x repository?

thinkyhead, 2018-12-10:
I'd rather not rename this file, if it can be avoided. It would be better to rename the `Marlin.ino` file instead. Or, we can simply continue to not support Visual Studio Community 2017 and simply keep supporting PlatformIO (Atom or Visual Studio Code) and Arduino IDE.

j54n1n, 2018-12-10:
But is this Marlin.cpp not the successor of the Marlin_main.cpp from the bugfix-1.1.x branch? Is the new build system with PlatformIO so much different that it does not allow a rename of this file?

just-jason, 2018-12-10:
Moving away from the PlatformIO (Atom or Visual Studio Code) and Arduino IDE is a very big step, and it is not necesary at all. By making a small name change the idea is only to increase the compatibility across other platforms. I think that Atom (which also runs on Visual Studio as a plug-in) is a great platform, and up till a few days ago I had never tried out the Visual Micro plug-in, but I too am very impressd at the time saving when compiling with Visual Micro. Either way, this is not about choosing between ONLY one platform, but instead, with a tiny change FromMarin.ccp to MarlinXYZ.ccp, we can all now enjoy the use of ALL platforms. That has to be a plus, don't you think ?

boelle, 2019-02-21:
@j54n1n problem solved?

j54n1n, 2019-03-04:
> 
> 
> @j54n1n problem solved?

Not really. I asked also in the Visual Micro forum (see: https://www.visualmicro.com/forums/YaBB.pl?num=1545386155/0#1) but it seems it is one of the expected results of VMicro. Basically with the .cpp file named in the same way as the .ino you can hide the cpp file for compilation. 

 With the latest versions of Arduino IDE 1.8.8 and the Visual Micro plugin 1902.17 you can get it to compile if you do one of the following changes:
- rename `Marlin/Marlin/src/Marlin.cpp` to `Marlin/Marlin/src/Marlin_main.cpp`

or
- rename the entire sketch from `Marlin/Marlin` to `Marlin/Marlinfw` with the consequence of renaming the .ino to `Marlin/Marlinfw/Marlinfw.ino` such that Arduino IDE is happy

Note: I used full paths here to make not to much confusion about which part has to be renamed.

@thinkyhead suggested once to rename the `Marlin.ino` file. If he is happy with that, we could try to find a better name instead of `Marlinfw`. I am open for suggestions.

boelle, 2019-03-04:
Not VMicro but vs code and platform io extension

Or arduino Ide 1.9

j54n1n, 2019-03-04:
Ah ok, but in this issue we never talked about VS code and/or PlatformIO. @thinkyhead only mentioned them that he wants to support them instead of VMicro.

Since I am not using them I can give you no hints. Maybe you should try to open a new issue if you found some problem!?

boelle, 2019-03-04:
Reason I asked was to find issues that where fixed but not closed. There where many.

But for this one I think we need @thinkyhead to have a look again

j54n1n, 2019-03-04:
> 
> 
> Reason I asked was to find issues that where fixed but not closed. There where many.
> 
> But for this one I think we need @thinkyhead to have a look again

Yes, I agree

InsanityAutomation, 2019-03-04:
We should modify the readme actually, 1.8.7 and newer is fine, no need to use the 1.9.0 beta anymore on the arduino ide.

thinkyhead, 2019-03-06:
Is there a project file for VisualMicro that can be saved in a location relative-to (above) the source files, such that the project file contains the path to the source files?

j54n1n, 2019-03-06:
> 
> 
> Is there a project file for VisualMicro that can be saved in a location relative-to (above) the source files, such that the project file contains the path to the source files?

Do you mean something like a makefile?

VMicro does generate a Visual Studio solution file but to get there you first have to open it within Visual Studio by *File* -> *Open* -> *Arduino Project* which then invokes the VM plugin.
At that stage the plugin does first a deep search of the entire Arduino "Marlin" sketch. But the scan fails if you do not rename either *Marlin.cpp* or the *Marlin.ino* sketch.

The last time I looked at this issue in December the scanning part was successful but I would get a linker error because the *Marlin.cpp* would get hidden as it is intended by plugin (see: https://www.visualmicro.com/forums/YaBB.pl?num=1545386155/0#1). Now with the most recent version of VMicro the sketch can not be imported anymore because due the *Marlin.cpp/Marlin.ino* pair it thinks to find the wrong include paths for the SD library.

As mentioned before the VMicro does not expect that there exist a *.cpp* file with the same name as the *.ino* file. Till now as a workaround you have to rename one of the files.

Anyhow the folder structure is like this if I go for the Marlinfw.ino workaround:
- Marlin/Marlinfw/
  - lib/
  - src/
  - Marlinfw.ino
  - Makefile
  - Configuration.h
  - Configuration_adv.h
  - **Marlinfw.sln**
  - **Marlinfw.vcxproj**
  - **Marlinfw.vcxproj.filters**
  - **Marlinfw.vcxproj.user**

I highlighted the added files from the VMicro plugin and you can find them as an attachement: [Marlinfw.zip](https://github.com/MarlinFirmware/Marlin/files/2936370/Marlinfw.zip)

GMagician, 2019-03-06:
Just to know, I don't use your plugin but I use visual studio and I compile for mega2560 via platformIO.
I'm wondering why are you using such pluging instead of VS without it? Which benefit do you have with it?

N.B: In my case platformio generated a native visual studio project, using a plugin I'm able to see the whole project in a tree structure (folder -> subfolder - > file) and manage such project

j54n1n, 2019-03-06:
> Just to know, I don't use your plugin but I use visual studio and I compile for mega2560 via platformIO.
> I'm wondering why are you using such pluging instead of VS without it? Which benefit do you [...]

I think the main difference is between the two options is that PlatformIO uses Visual Studio Code. VMicro is instead a plugin for the native Visual Studio (not VS Code) with the avr-gcc from Arduino IDE.

I used this plugin for other projects, and also for Marlin 1.1.x, but with the new 2.0.x branch and the changed files/folder layout it causes the above mentioned problem.

One of the main features of VMicro is faster build times. With multicore build I can get down to 48 seconds on my laptop for the first build, and from the second build onwards I can get down to 4 seconds for the 2.0.x branch. Has PlatformIO similar build times?

GMagician, 2019-03-06:
> I think the main difference between the two options is that PlatformIO uses Visual Studio Code. VMicro is instead a plugin for the native Visual Studio (not VS Code) with the avr-gcc from Arduino IDE.

that's wrong...I'm using VS2017 community edition, not vs code

> One of the main features of VMicro is faster build times. 

that may a good reason.

boelle, 2019-07-21:
Lack of Activity
This issue is being closed due to lack of activity. If you have solved the
issue, please let us know how you solved it. If you haven't, please tell us
what else you've tried in the meantime, and possibly this issue will be
reopened.

closed 2019-07-21 by boelle