MarlinFirmware/Marlin #12618: Marlin 2.0.x Serial mutually exclusive with storage when connected to an RPi

kylix-rd, 2018-12-09:
### Description

As of [this commit](https://github.com/MarlinFirmware/Marlin/commit/7d15854514f8e3818722eaf87dc96e098895ed79) running on a Re-ARM board with a RAMPS 1.4 connected to a Raspberry Pi 3 running OctoPrint (OctoPi), the sd-card cannot be mounted simultaneously with the serial port connected. Once the serial port is connected, the mounted partitions disappear from /dev. Likewise, mounting the sd-card partition will cause the serial port to not connect and timeout. This is a permanent condition until the Re-Arm board is reset. I've made sure to be running the latest RPi OS and patches.

### Steps to Reproduce

1. Install Marlin following the steps published for Re-ARM on marlinfw.org.
2. Connect with USB cable to Raspberry Pi running OctoPrint (OctoPi specifically).
3. From the command-line, locate the sd-card device under /dev/ (use dmesg to help identify it).
4. Mount the partition. (for example: sudo mount -t vfat /dev/sda1 /mnt/rearm)
5. Open Octoprint web interface and connect to printer.

### Alternative steps.

1. Follow steps 1 & 2 from above.
2. Open Octoprint web interface and connect to the printer.
3. From the command-line, run "ls -l /dev"
4. Notice that /dev/sda1 (or whatever name to which it is assigned) is no longer present.
5. Attempts to mount partition will now fail (sudo mount -t vfat /dev/sda1 /mnt/rearm
5.  

**Expected behavior:**

Octoprint connects to printer and functions normally.

or with alternative steps

sd-card is successfully mounted and firmware can be updated.

**Actual behavior:**

Octoprint will timeout trying to connect to printer.

or

Mountable partition disappears from /dev/ directory.

#### Additional Information

Updating the re-ARM firmware requires two-reboots. One reboot to allow mounting of the sd-card, then another reboot to update the firmware. Upon every fresh reboot, only the serial port can be connected or the sd-card partition mounted. Doing either operation, will block out the other operation until the next reboot of the Re-ARM board.

I haven't tried this on any other 32-bit board. I have an MKS-SBASE board that I will try to see if it does the same thing.

Another thing to note here is that this does NOT seem to be the case when connecting the Re-ARM board to a Windows 10 system. From there I can use either Repetier-Host or Pronterface to connect to the Re-ARM board through the virtual serial port AND access the sdcard. So this may not specifically be an issue with the firmware, rather it may be an issue only with an RPi.

gloomyandy, 2018-12-10:
I'm pretty sure that octoprint is mounting the sd card (m021) on startup and that is causing the card to no longer be shared over USB. There are probably options to change the way that octoprint handles the card if you wish to change things, but I think there are also ways to "unmount" the card via octoprint so that you get USB access to it. But I'm not very familiar with octoprint so there are probably better places to ask those questions!

boelle, 2019-02-21:
@kylix-rd problem solved?

kylix-rd, 2019-03-05:
@boelle Yes, it seems the problem is solved. Thanks for the followup.

boelle, 2019-03-05:
Maybe click close then?

closed 2019-03-05 by kylix-rd