MarlinFirmware/Marlin #12620: Cura/Marlin combo fails when gcode comment contains utf-8 character

smartavionics, 2018-12-10:
### Description

Firstly, apologies for lack of Marlin version info but I am a Cura person and not actually the user who had the problem.
 
Cura 3.6 user who was producing gcode for Marlin based printer reported that all layers were printed at the same height (0.2mm, the initial layer height). It turns out that Cura was embedding comments that included the original model file name which contained a UTF-8 character and this was causing the printer to fail to properly execute the gcode line that followed the comment and those lines included the change in Z so the nozzle never got any higher. A typical segment of the gcode file looked like this:
````
;LAYER:1
M140 S85
G1 F2100 E30.77954
M106 S127.5
;MESH:PluggMått v1.stl
G0 F4500 X97.416 Y105.849 Z0.4
````
The å is the cause of the problem, when I replaced that file name, the file printed as normal.

### Steps to Reproduce

1. [First Step] Slice a model contained in an STL file whose name contains a UTF-8 character using Cura 3.6
2. [Second Step] Print resulting gcode on printer using Marlin firmware
3. [and so on...]

**Expected behavior:** [What you expect to happen]

Model prints as normal.

**Actual behavior:** [What actually happens]

Nozzle never gets above the first layer height because the gcode that alters Z is being trashed by the comment line before it.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute]
(https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

smartavionics, 2018-12-10:
You can find more info in the Cura issue referenced above.

boelle, 2019-02-21:
@smartavionics problem solved?

smartavionics, 2019-02-21:
Hello @boelle , I don't believe any work has been done in Cura to avoid this problem, so no.

boelle, 2019-02-21:
Since it's a cuts issue maybe close this one?

smartavionics, 2019-02-21:
Well, arguably, there is an issue with both Cura and Marlin here. Cura could avoid including UTF-8 chars in comments and Marlin could avoid barfing on them. I would say work needs to be done on both.

Ghostkeeper, 2019-03-15:
Cura 4.1 will no longer put characters with byte codes >127 in the g-code. G-codes sliced with Cura 3.5 or 3.6 will still exhibit the issue though. And perhaps other slicers, too.

closed 2019-07-11 by shitcreek