MarlinFirmware/Marlin #12621: [Question] Speed limited by JUNCTION_DEVIATION?

MasterPIC, 2018-12-10:
Hi,
while trying to tune (mainly increase) my printer speeds, I found that max observed travel speed increases if I increase feedrate for travel moves even if there was already "space and time" to keep acceleration on before the tuning.

In a few words if I send
> M205 T200
> G1 X100 F10000

I get an higher max speed than the one observed with
> M205 T100
> G1 X100 F10000

But **during the movement** I see that **max speed is kept constant for a "long" time**.

I'm using JUNCTION_DEVIATION and also S_CURVE_ACCELERATION (bugfix-2.0.x up to date).
Is it the expected behaviour?

Thanks
Armando

MasterPIC, 2018-12-11:
In both cases max speed is lower than expected one...

AnHardt, 2018-12-11:
No configs -> no comments.
Else i'm tempted to tell you about lowering segments/second.

MasterPIC, 2018-12-11:
I'm using default values for Anet SO.

AnHardt, 2018-12-11:
I don't think so. - Else you wouldn't have problems with `JUNCTION_DEVIATION`.

PLONK.

closed 2018-12-11 by MasterPIC