MarlinFirmware/Marlin #12623: [RFC, Archim2] M226 doesn't work for all pins.

marcio-ao, 2018-12-10:
On the Archim 2, certain pins, such as pin 31, don't work with M226.

The cause of this seems to be that not all pins on the Archim are readable by the Arduino `digitalRead`. There seems to be a disconnect between way Marlin reads pins (using the READ macros, that can access all the pins in the microprocessor) and the way `digitalRead` does it, which can only access legitimate Arduino pins.

Another problem is that M226 doesn't respect the _ENDSTOP_INVERTING flags, so it's a bit confusing to use even when it does work.

I don't think there is an easy way to fix this other than making a special version of `digitalRead` for M226 that is a switch statement, something like this:

```
 bool _digitalRead(const int pin_number) {
   switch(pin_number) {
    #if HAS_Z_MIN
     case Z_MIN_PIN: return READ(Z_MIN_PIN) != Z_MIN_ENDSTOP_INVERTING; break;
    #endif
    #if HAS_Z_MIN_PROBE_PIN
     case Z_MIN_PROBE_PIN: return READ(Z_MIN_PROBE_PIN) != Z_MIN_PROBE_ENDSTOP_INVERTING;  break;
    #endif
    default: return digitalRead(pin_number);
   }
 }
```

Of course, to support every pin in this way would add a lot of code to Marlin, so it's one of those things that would have to be weighed carefully. For the moment, I don't mind doing this modification just for our FW and for the pins we use in our test fixtures, but I thought I would throw this out there for consideration.

Thoughts?

boelle, 2019-06-22:
missing a label, and i assume its still relavant?

boelle, 2019-10-24:
almost an year old

@shitcreek @thinkyhead any thoughts?

boelle, 2019-11-24:
this one has gone stale and i think there is an equal chance to figure it vis discord

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle