MarlinFirmware/Marlin #12624: Add Support for Lerdge-K 32-bit Board

DroneMang, 2018-12-10:
### Description: requested feature -->

Hi,

This board seems like a viable 32-bit option.  They have their own firmware, but Marlin has better features.  Thank you.

https://www.aliexpress.com/item/LERDGE-3D-Printer-Board-ARM-32Bit-Controller-Motherboard-for-3D-Printer-Control-Mainboard-Touch-Screen-Kit/32924347991.html?spm=a2g0s.13010208.99999999.259.4daa3c003jJCQn

Ludy87, 2018-12-11:
Lerdge firmware is closed and encrypted.

https://hackaday.io/project/162592-lerdge-3d-printer-mainboard-hacking

xC0000005, 2018-12-12:
That's my project. I have a K board on the way thanks to someone, have arduino code up and running on it. Next steps will be to add a pins file for the K and X, work out which timers are available, and so on. It's going to be ugly at first, probably not something you'd immediately want to run. Oh, and there's no USB support in the ST arduino core just yet. There will be, but there's not.

DroneMang, 2018-12-12:
I want to get Marlin 2.0 running on 32-bit something. Cost not a huge
issue. Looks like RE-ARM is the reference platform?? Can I use RADDS with a
RE-ARM or is RAMPS 1.6+ the best?

Can I use https://reprap.org/wiki/RAMPSXB with RE-ARM?

How about a bltouch?

How about X3 stepper expanders?

Thanks!

Ludy87, 2018-12-12:
why no MKS sbase?

DroneMang, 2018-12-12:
That would work, but I like the removable drivers. I want to use TMC 2130
or similar...

With RE-ARM and RAMPS 1.6+ I could possibly use a RAMPSXB

https://reprap.org/wiki/RAMPSXB

Glod76, 2018-12-15:
Mks Sbase can use breakout steppers

hectori4502, 2018-12-21:
check this.
https://github.com/MarlinFirmware/Marlin/issues/12632

boelle, 2019-02-21:
@DroneMang submit an PR against bugfik 2.0

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead

evandene, 2019-08-01:
'm using the LERDGE K board as well as two S board's on my machines for quite a while now.
The good thing about these boards is: These boards are hardware wise very good designed, great stuff, high quality parts used.
The not so good thing is their firmware development. Each release comes with surprises and to communicate with LERDGE is difficult. Like many Chinese suppliers they do not communicate very well with their customers. For example the latest firmware (as I'm writing) V3.04 for the K-Board does not recognize the second extruder from our dual-material printers. To have an acceptable result you need to use v3.01 but be aware this release has also some bugs. Each response from Lerdge gives you the feeling you're the one doing things wrong.
Do not use firmware 3.03 for the K-board.

evandene, 2019-09-15:
Release V3.05 Works well
Second k board installed fried after 30 minutes in use on a new delta printer

thinkyhead, 2019-09-15:
There may be a sketch out there that lets you touch pins to ground while it prints out the pin you touched. Or, if you can get Marlin to flash onto the board, `M43` can be used to do this. Once we know all the pin numbers that are used on the board we can create a pins file for it, and that should be all that is needed to get Marlin up and running. Unless there are other blocks, such as a custom bootloader that only accepts LERDGE firmware. In that case a new bootloader or a flash bypassing the bootloader would also be needed.

xC0000005, 2019-09-15:
I have accurate pins for both X and K. I do not have them for S (though I did ask Lerdge directly).
Lerdge had encryption on the bootloader and firmware - I cracked that and have a script which allows me to encrypt and decrypt firmware for their board. @JMZ also managed to make a simpler scirpt which could be put in PlatformIO to encrypt. I’ve been focusing on nailing down some bugs in the M200, but perhaps it’s time to go ahead and add a PR for the X and K pins.

xC0000005, 2019-09-15:
If you want to encrypt firmware to run on the Lerdge [X, K, S] bootloader, you can use this code, 
https://github.com/LoialOtter/Open-Lerdge-X/issues/1#issuecomment-453948501

Or this script:
https://cdn.hackaday.io/files/1625926956336128/lerdgedecrypt2.py
usage: lerdgedecrypt2.py <inputfile> <outputfile> encrypt 
In an ideal world, once this builds in PIO, it could have a script to automatically encrypt.

You'll want to follow the force-update process to apply it.
Note that right now, SD doesn't work, LCD doesn't work. So you're pretty much stuck printing over USB with octoprint. Alternately, my setup has an ESP8266 with ESP3D on it plugged into the ESP slot, and you don't need Lerdge's custom ESP8266 code anymore, any $2 ESP8266 will work just fine.

Bald-79, 2020-05-09:
I want to install the klipper on the lerdgeK.  I use a klipper-master.  I tried all the options, but the machine did not connect to the board.  tell me, do you have a positive result?  What settings should I choose for this board?  is there a verified config? thank

evandene, 2020-05-09:
> I want to install the klipper on the lerdgeK. I use a klipper-master. I tried all the options, but the machine did not connect to the board. tell me, do you have a positive result? What settings should I choose for this board? is there a verified config? thank

On the LERDGE S, LERDGE X and the LERDGE K board, running Marlin is not possible, at least not in a normal and full option way.
J.C. Nelson did a hell of a job in trying getting the Lerdge board up to running and in certain way he did. see his work and start a conversation with him. See hackaday link below, ... Good luck
https://hackaday.io/project/162592-lerdge-3d-printer-mainboard-hacking