MarlinFirmware/Marlin #12625: Printer halted

DarkShadowsX5, 2018-12-11:
My firmware only has 1 extruder enabled (E0) but my printer randomly tries to heat E1 and gives the message: 
"E1 heating failed"
"PRINTER HALTED"
"Please reset"

This issue is presented at random and increasing in frequency to the point it happens every other print. 
replacing the heater core with an new unused one did not fix the issue. 

Extruder define is exactly: (copy pasted)
#define EXTRUDERS 1

only E0 should be the only attempted heating processes.

RudyBenner, 2018-12-11:
no files?

DarkShadowsX5, 2018-12-11:
oh right. i dont see a way to upload it but here but i made some pastebin links. expires in 1 month. 

My changes have //John at the end. 
Configuration.h [https://pastebin.com/ccnqATQ3](url) 
Configuration_adv.h [https://pastebin.com/RYfeN9BX](url)

RudyBenner, 2018-12-11:
Those links don't work.

Phr3d13, 2018-12-11:
[Configuration.h](https://pastebin.com/ccnqATQ3)
[Configuration_adv.h](https://pastebin.com/RYfeN9BX)

DarkShadowsX5, 2018-12-11:
Thanks @Phr3d13  for fixing that. 
I think i know what i did wrong making the links. (facepalm)

Phr3d13, 2018-12-11:
Not a problem, glad to be able to help.

Phr3d13, 2018-12-11:
Is this happening during printing, or while using one of the preheat menus?

RudyBenner, 2018-12-11:
#define X_PROBE_OFFSET_FROM_EXTRUDER 10  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 10  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0   // Z offset: -below +above  [the nozzle]

are you using a probe?

DarkShadowsX5, 2018-12-11:
@Phr3d13 while its heating up to start a print.

@RudyBenner No i do manual leveling. i seldom have to re-level it so a probe is nearly pointless.

Phr3d13, 2018-12-11:
I would check your slicer, maybe something in the start gcode section?

DarkShadowsX5, 2018-12-11:
Here is my Start code from cura 3.6.0

`G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 X0 Y0 ;move X/Y to min endstops
G28 Z0 ;move Z to min endstops
G1 Z15.0 F9000 ;move the platform down 15mm
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 F9000
M117 Printing...
M75 ;Start Print Timer`

DarkShadowsX5, 2018-12-11:
@Phr3d13 
no i usually make sure its set to 1 extruder when updating to new firmware.
i updated recently from 1.1.3 to 1.1.9 and made sure to go through all my settings while checking for anything new that might be helpful.

@RudyBenner I believe i did change the timeout it was a bit short. i might not have marked it as a change when i did.

Phr3d13, 2018-12-11:
I don't see anything in any of that that would cause this behavior, and I don't remember seeing any bug reports regarding this. Could you try flashing bugfix-2.0.x and see if the issue persists?

DarkShadowsX5, 2018-12-11:
I'll give it a try and make sure to mark all my changes even un-comments . I'll report back after i printed a few things.

DarkShadowsX5, 2018-12-11:
I did not have bugfix-1.1.x added so i added that one, made my changes, and it verified and uploaded.
but i cant seem to get it to verify with bugfix-2.0.x added. 
"Error Compiling for board Arduino/genuino Mega or Mega 2560"

On the very first print attempt i got the same error with the 1.1.x bugfix

i tried multiple times and without making changes.
error msg from 2.0.x bugfix:

Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

sketch\src\HAL\HAL_AVR\HAL.cpp.o (symbol from plugin): In function `freeMemory':

(.text+0x0): multiple definition of `freeMemory'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

c:/program files/windowsapps/arduinollc.arduinoide_1.8.15.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

sketch\src\HAL\HAL_AVR\MarlinSerial.cpp.o (symbol from plugin): In function `__vector_25':

(.text+0x0): multiple definition of `__vector_25'

sketch\MarlinSerial.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `Running'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `setup_killpin()'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `setup_powerhold()'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `quickstop_stepper()'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `enable_all_steppers()'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `disable_e_steppers()'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `disable_e_stepper(unsigned char)'

sketch\Marlin_main.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\src\Marlin.cpp.o (symbol from plugin): In function `Running':

(.text+0x0): multiple definition of `disable_all_steppers()'
[... 1366 lines cut ...]

(.text+0x0): multiple definition of `thermalManager'

sketch\src\module\temperature.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\utility.cpp.o (symbol from plugin): In function `safe_delay(unsigned long)':

(.text+0x0): multiple definition of `safe_delay(unsigned long)'

sketch\src\core\utility.cpp.o (symbol from plugin):(.text+0x0): first defined here

sketch\watchdog.cpp.o (symbol from plugin): In function `watchdog_init()':

(.text+0x0): multiple definition of `watchdog_init()'

sketch\src\HAL\HAL_AVR\watchdog_AVR.cpp.o (symbol from plugin):(.text+0x0): first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

DarkShadowsX5, 2018-12-11:
I did not have bugfix-1.1.x added so i added that one, made my changes, and it verified and uploaded.
but i cant seem to get it to verify with bugfix-2.0.x added.
"Error Compiling for board Arduino/genuino Mega or Mega 2560"

On the very first print attempt i got the same error with the 1.1.x bugfix

i tried multiple times and without making changes.
here is my [Bugfix 2.0.x error](https://pastebin.com/fu2dGaDJ)

DarkShadowsX5, 2018-12-12:
@AnHardt thanks for your input. 
I have an MKS base v1.5 it has the ability to use 2 extruders but i only have **one** hooked up on the machine.  
It should be only setup to only heat E0 which is where the heater is connected. it seems to be trying to heat E1 as well throwing an error halting the printer.
EFB (Extruder, Fan, Bed) should be fine I've used that board config for ~4 years now on this printer. only recently started acting up about a month ago after retrieving it from a friends house an hour away.

Occasionally i get the "Homing failed printer halted" error with no miss steps, but Heating E1 seems to be more prevalent issue right now.  
I might even try an older version as i dont really need the new features. or maybe my boards memory is failing. I dont think its old enough too but i cant tell what else could be causing it.

Ludy87, 2018-12-12:
Which PIN did you choose for #define HEATER_0_PIN?

rmoravcik, 2018-12-12:
Well, it looks like a macro TEMP_ERR_PSTR(MSG, E) is printing string "E1" in case of extruder with index 0.

Phr3d13, 2018-12-12:
So is E0 actually having temp issues, or is it just the macro freaking out?

rmoravcik, 2018-12-12:
I don't know if it was intention, but calling first extruder E0 in configuration files and print string "E1" on LCD is quite confusing.

I've also checked code of marlin-1.1.9 and the macro is behaving in the same way.

@thinkyhead any thought on this?

DarkShadowsX5, 2018-12-13:
I can confirm it is indeed miss reporting which heater failed.  I started a print and let it heat to ~130c and disconnected one of the E0 heater wires and in 30 seconds it reported E1 had failed. 
This is very confusing for trouble shooting the real problem.
If the firmware code that is printing E1 when it should be E0 that should be fixed. 

I've printed a few things after adding the 1.1.x bugfix and a full power reset and i have not gotten that Heating fault error since. I did once after uploading the firmware and before a power reset. 
so something there must have fix the heating issue/bug. 

If i dont have the Heating error again by the end of the week then its solved for now. 
Thanks for all your help.

hamster65, 2018-12-19:
This might be related: I enabled the filament runout sensor and started a print. The message on the lcd reads something like that:
<<Print paused>>
press button to heat nozzle
<<NOZZLE: E1>>

DarkShadowsX5, 2018-12-20:
@hamster65 
The code has been reporting the wrong extruder this whole time. I even tested it by unplugging my E0 heater (the only one i have installed and enabled) mid heat cycle and got E1 error message. Its confusing
hopefully they fixed it for the next update... 

That's the only thing that may be related to this issue. My problem with the heater stopped me from printing out right until i reset the printer to try again the E1 thing added a level of confusion. i had to restart the printer and hope i didn't get the error message again. i had ~80% failure to print rate. 

if you haven't already add the 1.9.x bug fix power off your printer, power on, and hope your interruption problem goes away.

boelle, 2019-02-21:
@DarkShadowsX5 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-10 by thinkyhead